kraj/meta-clang #211: `thud` branch ERROR: Multiple .bb files are due to be built which each provide 'libunwind'

sparkleholic, 2019-11-07:
In case of other metalayer has a "libunwind.bb" recipe, it has "PROVIDE" duplicate problem. To solve this, the "PREFERRED_PROVIDER_libunwind are defined in meta-clang.
```
PREFERRED_PROVIDER_libunwind = "libunwind"
PREFERRED_PROVIDER_libunwind_mipsarch = "libunwind"
PREFERRED_PROVIDER_libunwind_toolchain-clang = "libcxx"
```
But, if other recipes depend on both libunwind and libcxx both, the error occurs.

This is an actual error case.
If the other meta layer has 'libunwind' recipe and the IMAGE depends on libunwind and no-depends on libcxx, IMAGE generation is ok. 
But even in this case, the case of generating a SDK by `-c populate_sdk`, it refers to `meta-clang/packagegroup-core-standalone-sdk-target.bbappend`.
```
RRECOMMENDS_${PN} += "\
    libcxx-dev 
```
Basically, libcxx-dev recipe depends on libcxx, as a result, the target build try to include both `libcxx` and `libunwind`. In this case, an error occurs like the following.
```
ERROR: Multiple .bb files are due to be built which each provide libunwind:########################                                                                                                 
  meta-clang/recipes-devtools/clang/libcxx_git.bb
  oe-core/meta/recipes-support/libunwind/libunwind_1.2.1.bb
```

The master branch's libcxx recipes does not export libunwind beside libcxx inself, but 'thud' branch does not have this change.
I think thud bransh need to have this patch as well.

sparkleholic, 2019-11-07:
@shr-project 
I have an issue regarding `populate_sdk`

kraj, 2019-11-10:
@sparkleholic I think its best to disable llvm libunwind, you can do so by removeing it from packageconfig

https://github.com/kraj/meta-clang/blob/thud/recipes-devtools/clang/libcxx_git.bb#L18

second change would be to remove

https://github.com/kraj/meta-clang/blob/thud/classes/clang.bbclass#L50-L52

kraj, 2019-11-20:
@sparkleholic did above suggestion work for you?

jaysterling, 2020-03-08:
I also ran into this issue and just wanted to confirm that disabling llvm libunwind fixed the issue for me. I created a `libcxx_git.bbappend` file in our project's layer and added a line:

`PACKAGECONFIG_remove = "unwind"`

and no more error.

kraj, 2021-03-25:
closing based onb above

closed 2021-03-25 by kraj