kraj/meta-clang #212: zeus release branch

priv-kweihmann, 2019-11-11:
Hi, are there any plans for a zeus release branch?

kraj, 2019-11-11:
@priv-kweihmann Yes certainly, wanted to keep running with master for a while so we can fix any zeus/master related issues before branching or any incompatible change that triggers it for master. I understand that sometimes its needed for automation or tooling reasons. So do you need it to be branched out then please let me know. Right now I expect master to work equally well with zeus and master.

priv-kweihmann, 2019-11-12:
There's no urgent need from my side to have a dedicated zeus branch at the moment. Just asked out of interest.

closed 2019-11-12 by priv-kweihmann

kraj, 2019-11-12:
no worres, I am in process of creating the branch, since there has been enough time since zeus release for settling the dust