kraj/meta-clang #214: Compiler-rt fails with "linker command failed with exit code 1"

megaparanoik, 2019-11-13:
Hi!
I'm trying to build meta-electron which has a dependence on clang.
But the build process fails on compiler-rt stage with a message **"clang-5.0: error: linker command failed with exit code 1 (use -v to see invocation)"**

**Target is armv7a
Yocto - Rocko
meta-clang - Rocko branch**

Exact fail from log.do_compile:
```
[ 83%] Linking CXX shared library lib/linux/libclang_rt.ubsan_standalone-armhf.so
FAILED: lib/linux/libclang_rt.ubsan_standalone-armhf.so 
: && /home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-clang++ -fPIC -march=armv7-a -mfpu=neon -mfloat-abi=hard -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong -pie -fPIE -D_FORTIFY_SOURCE=2 -Wa,--noexecstack -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot  -O2 -Wa,--noexecstack -fexpensive-optimizations -frename-registers -fomit-frame-pointer -ftree-vectorize   -finline-functions -finline-limit=64   -Wno-error=maybe-uninitialized -Wno-error=unused-result  -fvisibility-inlines-hidden -march=armv7-a -mfpu=neon -mfloat-abi=hard -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong -pie -fPIE -D_FORTIFY_SOURCE=2 -Wa,--noexecstack -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now,-z,noexecstack  -march=armv7-a -mfloat-abi=hard -shared -Wl,-soname,libclang_rt.ubsan_standalone-armhf.so -o lib/linux/libclang_rt.ubsan_standalone-armhf.so lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_allocator.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_common.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_deadlock_detector1.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_deadlock_detector2.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_errno.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_flags.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_flag_parser.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_libc.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_libignore.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_linux.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_linux_s390.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_mac.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_persistent_allocator.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_platform_limits_linux.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_platform_limits_posix.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_posix.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_printf.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_procmaps_common.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_procmaps_freebsd.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_procmaps_linux.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_procmaps_mac.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_stackdepot.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_stacktrace.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_stacktrace_printer.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_stoptheworld_mac.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_suppressions.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_symbolizer.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_symbolizer_libbacktrace.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_symbolizer_mac.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_symbolizer_win.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_thread_registry.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_win.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_linux_x86_64.S.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_linux_mips64.S.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_termination.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_common_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sancov_flags.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_coverage_libcdep_new.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_coverage_win_sections.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_linux_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_mac_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_posix_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_stacktrace_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_stoptheworld_linux_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_symbolizer_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_symbolizer_posix_libcdep.cc.o lib/sanitizer_common/CMakeFiles/RTSanitizerCommonLibc.armhf.dir/sanitizer_unwind_linux_libcdep.cc.o lib/ubsan/CMakeFiles/RTUbsan.armhf.dir/ubsan_diag.cc.o lib/ubsan/CMakeFiles/RTUbsan.armhf.dir/ubsan_init.cc.o lib/ubsan/CMakeFiles/RTUbsan.armhf.dir/ubsan_flags.cc.o lib/ubsan/CMakeFiles/RTUbsan.armhf.dir/ubsan_handlers.cc.o lib/ubsan/CMakeFiles/RTUbsan.armhf.dir/ubsan_value.cc.o  -ldl -lrt -lpthread && :
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o(.text+0xcc): R_ARM_TLS_LE32 relocation not permitted in shared object
lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o: In function `__sanitizer::DTLS_Destroy()':
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/git/lib/sanitizer_common/sanitizer_tls_get_addr.cc:(.text+0xcc): dangerous relocation: unsupported relocation
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o(.text+0x4f0): R_ARM_TLS_LE32 relocation not permitted in shared object
lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o: In function `__sanitizer::DTLS_on_tls_get_addr(void*, void*, unsigned long, unsigned long)':
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/git/lib/sanitizer_common/sanitizer_tls_get_addr.cc:(.text+0x4f0): dangerous relocation: unsupported relocation
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld: lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o(.text+0x5c4): R_ARM_TLS_LE32 relocation not permitted in shared object
[... cut ...] ect
lib/sanitizer_common/CMakeFiles/RTSanitizerCommon.armhf.dir/sanitizer_tls_get_addr.cc.o: In function `__sanitizer::DTLS_Get()':
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/compiler-rt/5.0.1+gitAUTOINC+4b38c4038a-r0/git/lib/sanitizer_common/sanitizer_tls_get_addr.cc:(.text+0x5dc): dangerous relocation: unsupported relocation
clang-5.0: error: linker command failed with exit code 1 (use -v to see invocation)
```

Please, give some advice on how to solve this issue.

kraj, 2019-11-13:
if you dont need sanitizers just disable them

megaparanoik, 2019-11-13:
I have disabled sanitizers by 
EXTRA_OECMAKE_append = " -DCOMPILER_RT_BUILD_SANITIZERS=OFF" in compiler-rt_git.bb
and now compiler-rt is buildable.

But I ran into the problem with libcxx_git.bb
It fails on do.configure step.
the CMakeError.log contains a lot of errors as follow:

```
Run Build Command:"/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/libcxx/5.0.1+gitAUTOINC+1368f4044e_598ae4fa2e_5df6b5da0d_29d1624468-r0/recipe-sysroot-native/usr/bin/ninja" "cmTC_fe993"
[1/2] Building C object CMakeFiles/cmTC_fe993.dir/CheckIncludeFile.c.o
FAILED: CMakeFiles/cmTC_fe993.dir/CheckIncludeFile.c.o 
/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/libcxx/5.0.1+gitAUTOINC+1368f4044e_598ae4fa2e_5df6b5da0d_29d1624468-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-clang   -march=armv7-a -mfpu=neon -mfloat-abi=hard -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong -pie -fPIE -D_FORTIFY_SOURCE=2 -Wa,--noexecstack -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/libcxx/5.0.1+gitAUTOINC+1368f4044e_598ae4fa2e_5df6b5da0d_29d1624468-r0/recipe-sysroot  -O2 -Wa,--noexecstack -fexpensive-optimizations -frename-registers -fomit-frame-pointer -ftree-vectorize   -finline-functions -finline-limit=64   -Wno-error=maybe-uninitialized -Wno-error=unused-result  -march=armv7-a -mfpu=neon -mfloat-abi=hard -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong -pie -fPIE -D_FORTIFY_SOURCE=2 -Wa,--noexecstack -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/alex/temp/poky/build/tmp-glibc/work/armv7ahf-neon-oe-linux-gnueabi/libcxx/5.0.1+gitAUTOINC+1368f4044e_598ae4fa2e_5df6b5da0d_29d1624468-r0/recipe-sysroot -fPIC -o CMakeFiles/cmTC_fe993.dir/CheckIncludeFile.c.o   -c CheckIncludeFile.c
clang-5.0: warning: optimization flag '-fexpensive-optimizations' is not supported [-Wignored-optimization-argument]
clang-5.0: warning: optimization flag '-frename-registers' is not supported [-Wignored-optimization-argument]
clang-5.0: warning: optimization flag '-finline-limit=64' is not supported [-Wignored-optimization-argument]
warning: unknown warning option '-Werror=maybe-uninitialized'; did you mean '-Werror=uninitialized'? [-Wunknown-warning-option]
CheckIncludeFile.c:1:10: fatal error: 'malloc/malloc.h' file not found
#include <malloc/malloc.h>
         ^~~~~~~~~~~~~~~~~
1 warning and 1 error generated.
ninja: build stopped: subcommand failed.
```
It looks like there is some issue with paths

Could you please direct me on what I have to check to determine the root of the error with libcxx.
Thanks!

kraj, 2019-11-13:
Try adding following in libcxx recipe
```
DEPENDS_append_class-target = " compiler-rt clang-cross-${TARGET_ARCH} virtual/${MLPREFIX}libc virtual/${TARGET_PREFIX}compilerlibs"
INHIBIT_DEFAULT_DEPS = "1"
```

megaparanoik, 2019-11-14:
Unfortunately, it has no effect.

I spent some time and made an experiment.
I downloaded clean repos for poky/openembedded/clan/raspberrypi, switched to rocko branch.
And was surprised that build was completely successful.

This looks like my yocto has conflicts, but I can't determine what exactly.

Should I provide additional CMake/Yocto Error/LOG files?

kraj, 2019-11-14:
hmm interesting, if you have logs from bad and good builds for failing package then it would be helpful..

megaparanoik, 2019-11-14:
Thank you for your support!
Here is zip with logs for "bad" and "good" builds of libcxx recipe.
[libcxx_logs.zip](https://github.com/kraj/meta-clang/files/3849160/libcxx_logs.zip)

kraj, 2019-11-15:
So I see early failures of compiler checks e.g.

```diff
--- Performing Test SUPPORTS_VARIADIC_MACROS_FLAG - Failed
+-- Performing Test SUPPORTS_VARIADIC_MACROS_FLAG - Success
```
This means clang compiler was not installed correctly. Perhaps  `CMakeError.log` has more details 
as to why it failed.

kraj, 2019-11-20:
closing based on https://github.com/kraj/meta-clang/issues/214#issuecomment-554000836

closed 2019-11-20 by kraj