kraj/meta-clang #230: CMake find_package(LLVM) does not function with meta-clang

mdavis777, 2019-11-26:
Any Cmake that uses find_package(LLVM) supplied with clang fails as it tries to find llvm-tablegen in recipes-sysroot where the libraries are instead of recipes-sysroot-native which contains the binaries.

```
| CMake Error at /build/mdavis/core/tmp/work/corei7-64-poky-linux/hello-world/1.0-r1/recipe-sysroot/usr/lib/cmake/llvm/LLVMExports.cmake:683 (message):
|   The imported target "llvm-tblgen" references the file
|
|      "/build/mdavis/core/tmp/work/corei7-64-poky-linux/hello-world/1.0-r1/recipe-sysroot/usr/bin/llvm-tblgen"
|
|   but this file does not exist.  Possible reasons include:
```

Even a simple cmake such as the one below will cause the issue.
```
cmake_minimum_required(VERSION 3.12)
project(Helloworld)

# Include Clangs Cmake info file and prepare some variables with knowledge of Clang.
find_package(LLVM)

add_executable(HelloWorld helloworld.cpp)

```

**Desktop (please complete the following information):**
 - OS: Fedora 26
- Yocto 2.6 and 2.7 both have this issue

I have tried to pass various flags using OE_CMAKE but haven't found one that seems to work properly.

kraj, 2019-11-28:
does setting
```
export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"
```
in recipe help

mdavis777, 2019-12-02:
No change.
I did some more digging and it seems to be isolated to LLVMExports.cmake and LLVMExports-release.cmake.  They are not designed for binaries and libraries to not share the same prefix path.

Specifically _IMPORT_PREFIX is set to be a 3 directories above from where the .cmake files exist.

```
# Compute the installation prefix relative to this file.
get_filename_component(_IMPORT_PREFIX "${CMAKE_CURRENT_LIST_FILE}" PATH)
# Use original install prefix when loaded through a
# cross-prefix symbolic link such as /lib -> /usr/lib.
get_filename_component(_realCurr "${_IMPORT_PREFIX}" REALPATH)
get_filename_component(_realOrig "/usr/lib/cmake/llvm" REALPATH)
if(_realCurr STREQUAL _realOrig)
  set(_IMPORT_PREFIX "/usr/lib/cmake/llvm")
endif()
unset(_realOrig)
unset(_realCurr)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
if(_IMPORT_PREFIX STREQUAL "/")
  set(_IMPORT_PREFIX "")
endif()
```

That sets the location for both executables and libraries in LLVMExports-release.cmake
```
# Import target "LLVMTableGen" for configuration "Release"
set_property(TARGET LLVMTableGen APPEND PROPERTY IMPORTED_CONFIGURATIONS RELEASE)
set_target_properties(LLVMTableGen PROPERTIES
  IMPORTED_LINK_INTERFACE_LANGUAGES_RELEASE "CXX"
  IMPORTED_LOCATION_RELEASE "${_IMPORT_PREFIX}/lib/libLLVMTableGen.a"
  )

list(APPEND _IMPORT_CHECK_TARGETS LLVMTableGen )
list(APPEND _IMPORT_CHECK_FILES_FOR_LLVMTableGen "${_IMPORT_PREFIX}/lib/libLLVMTableGen.a" )

# Import target "llvm-tblgen" for configuration "Release"
set_property(TARGET llvm-tblgen APPEND PROPERTY IMPORTED_CONFIGURATIONS RELEASE)
set_target_properties(llvm-tblgen PROPERTIES
  IMPORTED_LOCATION_RELEASE "${_IMPORT_PREFIX}/bin/llvm-tblgen"
  )
```

LLVMExports.cmake includes any file named LLVMExports-*.cmake so perhaps we can create a meta-clang specific ones that can override the _IMPORT_PREFIX for binary and libraries.  Something like 
_IMPORT_PREFIX_LIB="${STAGING_EXECPREFIXDIR}"
_IMPORT_PREFIX_BIN="${STAGING_DIR_NATIVE}/usr"

kraj, 2019-12-02:
good dig. Can we setup _IMPORT_PREFIX in terms of YOCTO_ALTERNATE_EXE_PATH

mdavis777, 2019-12-02:
Possibly with some changes to the export cmakes but it should be doable.
Is the meta-clang using YOCTO_ALTERNATE_EXE_PATH working?  
"${STAGING_LIBDIR}/llvm-config" doesn't exist. ${STAGING_BINDIR_NATIVE}/llvm-config  does exist but doesn't point to the correct libdir.

I saw a script in llvm-common.bb that looks like it should wrap it, but I can't seem to get it to work properly.

mdavis777, 2019-12-02:
I did get llvm-config to actually reference the script instead of the binary by cleaning out everything rebuilding.  However llvm-config --bindir points to recipe-sysroot.  Wouldn't we need it to point to recipe-sysroot-native?

kraj, 2019-12-02:
yeah its cross llvm-config to it has to be smartly mix native package binary paths with target libs, I am not sure if llvm-config is used in such capacity thus far. So we might have to tweak it

mdavis777, 2019-12-03:
I made a rough first attempt at fixing it.
https://github.com/mdavis777/meta-clang/commit/e5a6921ec31b4a65bfe844c41be5cbf5747a4693

Currently it generates a new Export file that divides the libraries from the binaries and assigns them a path based on llvm-config if YOCTO_ALTERNATE_EXE_PATH is defined.  
Unfortunately it still reports the recipe-sysroot path for bin and lib since YOCTO_ALTERNATE_EXE_PATH prevents the return of the native bins.  I can change the llvm-config script in llvm-common, but I don't know what other issues that might cause downstream.

mdavis777, 2019-12-03:
https://github.com/mdavis777/meta-clang/commit/719680defa604fa1f0330310c2cd87ef1e53e31c
This commit fully works.  I don't know what the wide spread repercussions of it would be.  Mostly llvm-config.  The change to the cmake export should be benign.

mdavis777, 2019-12-03:
Just noticed it probably won't work and/or will break multilib. So that will be a thing too.

kraj, 2019-12-03:
> Just noticed it probably won't work with and/or will break multilib. So that will be a thing too.

right  was about to comment on this. But that seems solvable.

mdavis777, 2019-12-04:
Added a fix for multilib and rebased it on top of master.  Currently to work you have to point to:
`export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_BINDIR}/crossscripts/llvm-config"`
Not sure why the other recipes that seem to use it point to a file that doesn't actually exist.  I can change it to just the env variable being defined at all as well.

kraj, 2019-12-04:
`YOCTO_ALTERNATE_EXE_PATH` actually acts like `--sysroot` does for cross compiler. So it helps point cross llvm-config into target sysroot and then calculates variables relative to that. So its not important if the path exists or not but the depth of directory is more important since it will find the rest of dirs relative to that.

mdavis777, 2019-12-04:
Nevermind I already took that check out.  I squashed the commits and cleaned up the commit log.  I am still trying to test multilib as it appears to have broken in the last month or so on master.  
64bit clang builds fine but lib32-clang and lib32-compiler-rt fail every time in multilib with `qc: no such file or directory.`  

Seems like `recipe-sysroot-native/usr/bin/i686-pokymllib32-linux/i686-pokymllib32-linux-llvm-ar` link isn't processing right for some reason.  Trying to narrow down the commit that broke it.

kraj, 2019-12-04:
I think this needs to be fixed in llvm-ar, it is sorting `lib` above `ar` so when you have `i686-pokymllib32-linux-llvm-ar` as AR name it confuses this code. See below, we should be able to fix it, I havent had time to fix it. but while here it would be good if you can try to.

https://github.com/llvm-mirror/llvm/blob/master/tools/llvm-ar/llvm-ar.cpp#L1178

kraj, 2019-12-05:
should be fixed with #238

closed 2019-12-05 by kraj