kraj/meta-clang #231: Question: Adding gallium-llvm PACKAGECONFIG in mesa causes target clang to build

schnitzeltony, 2019-12-01:
Backgraound:
As you know I am working on gnome-desktop. Currently I get only the old (GNOME2/fallback) session to run on Raspi. During my investigations I found [1]. So I decided to add gallium-llvm to mesa's PACKAGECONFIG. Since then target clang is build. Note that I have meta-clang in my layers for meta-browser.

Now my questions:
1. Is it an expected behaviour when not setting TOOLCHAIN = "clang"?
2. If yes: Does that mean adding gallium-llvm to mesa turn meta-clang into a layer dependency?

[1] https://gitlab.gnome.org/GNOME/gnome-session/blob/master/data/hardware-compatibility#L28

kraj, 2019-12-02:
There is llvm recipe in Oe-core and if you chose to use llvm-pipe in Mesa then you will use that however if you have meta-clang then we override that with llvm coming from meta-clang for the simple reason that we Already are using clang for other things so why not for this too

If you want to use llvm from Oe-core then you can edit the preferred providers in layers.conf for meta-clang 

If you do not have use for meta-clang then probably it’s better to disable it and you will still be able to use llvm-pipe

schnitzeltony, 2019-12-02:
Thanks for explanation.

I think current behaviour is OK because TOOLCHAIN sets up native toolchain which is unrelated to target compiler.

FWIW: adding gallium-llvm to mesa did not fix nothing - but found some knobs getting me closer to gnome desktop...

closed 2019-12-02 by schnitzeltony

schnitzeltony, 2019-12-04:
Out of clang scope:

Meanwhile I got full gnome running - no need to adjust mesa. Hope my patches go out before python2-eats-my-children-so-break-world mania....