kraj/meta-clang #232: meta-clang not compatible with meta-mingw

wertig0n, 2019-12-02:
Hi!

Been going over this problem for a few weeks and I think I have finally found a workaround for now, but in the meanwhile I would like to report the following.

### Branches tested

thud, zeus

### Steps to reproduce

1. Download meta-mingw, meta-openembedded/meta-oe, meta-clang
2. set all branches to [thud, zeus]
3. `SDKMACHINE=x86_64-mingw32 bitbake meta-toolchain`

### Expected output

`$ bitbake meta-toolchain -k
Loading cache: 100% |#####################################################################################################################################################| Time: 0:00:00
Loaded 1835 entries from dependency cache.
Parsing recipes: 100% |###################################################################################################################################################| Time: 0:00:03
Parsing of 1470 .bb files complete (1272 cached, 198 parsed). 2119 targets, 76 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i586-poky-linux"
MACHINE              = "qemux86"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6.4"
TUNE_FEATURES        = "m32 i586"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "thud:8cd3ee6e1a50ad9f40466bcadb236c619c42ef19"
meta-oe              = "thud:446bd615fd7cb9bc7a159fe5c2019ed08d1a7a93"
meta-clang           = "thud:22f0b0eaad454279c604a83dba53be6e6efb1a06"
meta-mingw           = "thud:6556cde16fbdf42c7edae89bb186e5ae4982eff5"

Initialising tasks: 100% |################################################################################################################################################| Time: 0:00:01
Sstate summary: Wanted 84 Found 0 Missed 84 Current 652 (0% match, 88% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: nativesdk-openssl-1.1.1b-r0 do_configure: Function failed: do_configure (log file is located at /mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-openssl/1.1.1b-r0/temp/log.do_configure.31195)
ERROR: Logfile of failure stored in: /mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-openssl/1.1.1b-r0/temp/log.do_configure.31195
Log data follows:
| DEBUG: Executing shell function do_configure
| Usage: Configure [no-<cipher> ...] [enable-<cipher> ...] [-Dxxx] [-lxxx] [-Lxxx] [-fxxx] [-Kxxx] [no-hw-xxx|no-hw] [[no-]threads] [[no-]shared] [[no-]zlib|zlib-dynamic] [no-asm] [no-dso] [no-egd] [sctp] [386] [--prefix=DIR] [--openssldir=OPENSSLDIR] [--with-xxx[=vvv]] [--config=FILE] os/compiler[:flags]
| 
| pick os/compiler from:
| BS2000-OSD BSD-generic32 BSD-generic64 BSD-ia64 BSD-sparc64 BSD-sparcv8
| BSD-x86 BSD-x86-elf BSD-x86_64 Cygwin Cygwin-i386 Cygwin-i486 Cygwin-i586
| Cygwin-i686 Cygwin-x86 Cygwin-x86_64 DJGPP MPE/iX-gcc UEFI UWIN VC-CE VC-WIN32
| VC-WIN32-ARM VC-WIN32-ONECORE VC-WIN64-ARM VC-WIN64A VC-WIN64A-ONECORE
| VC-WIN64A-masm VC-WIN64I aix-cc aix-gcc aix64-cc aix64-gcc android-arm
| android-arm64 android-armeabi android-mips android-mips64 android-x86
| android-x86_64 android64 android64-aarch64 android64-mips64 android64-x86_64
| bsdi-elf-gcc cc darwin-i386-cc darwin-ppc-cc darwin64-ppc-cc
| darwin64-x86_64-cc gcc haiku-x86 haiku-x86_64 hpux-ia64-cc hpux-ia64-gcc
| hpux-parisc-cc hpux-parisc-gcc hpux-parisc1_1-cc hpux-parisc1_1-gcc
| hpux64-ia64-cc hpux64-ia64-gcc hpux64-parisc2-cc hpux64-parisc2-gcc hurd-x86
| ios-cross ios-xcrun ios64-cross ios64-xcrun iossimulator-xcrun iphoneos-cross
| irix-mips3-cc irix-mips3-gcc irix64-mips4-cc irix64-mips4-gcc linux-aarch64
| linux-alpha-gcc linux-aout linux-arm64ilp32 linux-armv4 linux-c64xplus
| linux-elf linux-generic32 linux-generic64 linux-ia64 linux-mips32 linux-mips64
| linux-ppc linux-ppc64 linux-ppc64le linux-sparcv8 linux-sparcv9 linux-x32
| linux-x86 linux-x86-clang linux-x86_64 linux-x86_64-clang linux32-s390x
| linux64-mips64 linux64-s390x linux64-sparcv9 mingw mingw64 nextstep
| nextstep3.3 sco5-cc sco5-gcc solaris-sparcv7-cc solaris-sparcv7-gcc
[... 1805 lines cut ...]
|   install TARGETS given no LIBRARY DESTINATION for shared library target
|   "LTO".
| Call Stack (most recent call first):
|   tools/lto/CMakeLists.txt:19 (add_llvm_library)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/build/CMakeFiles/CMakeError.log".
| WARNING: /mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/temp/run.do_configure.31197:1 exit 1 from 'cmake -G Ninja -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /mnt/extra/yocto/poky/build/tmp/work-shared/llvm-project-source-7.1.0-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/opt/poky/2.6.4/sysroots/x86_64-pokysdk-mingw32/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=bin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/opt/poky/2.6.4/sysroots/x86_64-pokysdk-mingw32/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/opt/poky/2.6.4/sysroots/x86_64-pokysdk-mingw32/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/toolchain.cmake -DCMAKE_VERBOSE_MAKEFILE=1 -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DLIBCXX_CXX_ABI=libcxxabi -DLLVM_BUILD_EXTERNAL_COMPILER_RT=ON -DCXX_SUPPORTS_CXX11=ON -DLIBCXXABI_LIBCXX_INCLUDES=/mnt/extra/yocto/poky/build/tmp/work-shared/llvm-project-source-7.1.0-r0/git/libcxx/include -DLIBCXX_CXX_ABI_INCLUDE_PATHS=/mnt/extra/yocto/poky/build/tmp/work-shared/llvm-project-source-7.1.0-r0/git/libcxxabi/include -DLIBCXX_CXX_ABI_LIBRARY_PATH=/mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/build/lib -DLLVM_ENABLE_PROJECTS='libcxx;libcxxabi;libunwind' -G Ninja /mnt/extra/yocto/poky/build/tmp/work-shared/llvm-project-source-7.1.0-r0/git/llvm -DLIBCXXABI_USE_LLVM_UNWINDER=ON -DLIBCXXABI_LIBUNWIND_INCLUDES=/mnt/extra/yocto/poky/build/tmp/work-shared/llvm-project-source-7.1.0-r0/git/projects/libunwind/include -DLIBCXX_ENABLE_ABI_LINKER_SCRIPT=OFF -Wno-dev'
| ERROR: Function failed: do_configure (log file is located at /mnt/extra/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-libcxx/7.1.0-r0/temp/log.do_configure.31197)
ERROR: Task (virtual:nativesdk:/mnt/extra/yocto/poky/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2143 tasks of which 2134 didn't need to be rerun and 9 failed.

Summary: 9 tasks failed:
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-connectivity/openssl/openssl_1.1.1b.bb:do_configure
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-support/db/db_5.3.28.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-core/ncurses/ncurses_6.1+20180630.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-support/gdbm/gdbm_1.18.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-core/libxcrypt/libxcrypt_4.1.1.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-devtools/flex/flex_2.6.0.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta/recipes-devtools/bison/bison_3.0.4.bb:do_compile
  virtual:nativesdk:/mnt/extra/yocto/poky/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure
  virtual:nativesdk:/mnt/extra/yocto/poky/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_configure
Summary: There were 15 ERROR messages shown, returning a non-zero exit code.`

rossburton, 2019-12-02:
I quickly replicated this, the problem is that clang is being pulled into the SDK.

Try setting `CLANGSDK=""` in your local.conf.

kraj, 2019-12-02:
although clang should build fine for windows, we do not yet support mingw as host with this layer. Perhaps better to disable CLANG in SDK as @rossburton suggests

wertig0n, 2019-12-03:
Yes, the CLANGSDK="" works for zeus, but not for thud, unfortunately:

`$ pwd
/mnt/extra/yocto/poky/meta-clang
$ git checkout thud
Switched to branch 'thud'
Your branch is up to date with 'origin/thud'.
$ grep CLANGSDK -R .
$ git checkout zeus
Switched to branch 'zeus'
Your branch is up to date with 'origin/zeus'.
$ grep CLANGSDK -R .
./conf/layer.conf:CLANGSDK ??= "1"
./README.md:`bitbake -cpopulate_sdk <image>` in circumstanced where clang is not expected to be part of SDK, then reset `CLANGSDK`
./README.md:CLANGSDK = ""
./recipes-core/packagegroups/packagegroup-cross-canadian.bbappend:CLANGCROSSCANADIAN = "${@bb.utils.contains('CLANGSDK', '1', 'clang-cross-canadian-${TRANSLATED_TARGET_ARCH}', '', d)}"`

Since our project has a hard dependency on meta-ti and meta-ti are gonna keep using thud until this summer, I'll just settle for manually disabling the affected layers for now.

Thank you so much for your assistance! You still might want to keep this issue open for now, I doubt I will be the only one affected by it. :)

kraj, 2019-12-03:
@wertig0n backport the needed bits to thud. Usually we only keep last two releases active but I am happy to accept patches for thud still

wertig0n, 2019-12-05:
@kraj I tried to cherry-pick the two commits that introduced CLANGSDK, but it didn't work. Seems to be more work involved, and not sure I'm allowed to fix that. :(

As thud is nearing end-of-life and we'll switch to Zeus once meta-ti switches over, we can live with the workaround to disable clang for generating the SDK. :)

wdobbe, 2020-09-22:
The NXP Yocto BSP also includes meta-browser (and thus meta-clang).
I am on the zeus branch but adding `CLANGSDK = ""` to local.conf gave me another error:

```
NOTE: Setscene tasks completed
ERROR: core-image-minimal-1.0-r0 do_populate_sdk: Could not invoke dnf. Command '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/recipe-sysroot-native/usr/bin/dnf -v --rpmverbosity=info -y -c /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/sdk/image/etc/dnf/dnf.conf --setopt=reposdir=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/sdk/image/etc/yum.repos.d --installroot=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/sdk/image --setopt=logdir=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp --repofrompath=oe-repo,/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/oe-sdk-repo --nogpgcheck install nativesdk-packagegroup-sdk-host packagegroup-cross-canadian-dynniq-flownode-acu' returned 1:
DNF version: 4.2.2
cachedir: /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/sdk/image/var/cache/dnf
Added oe-repo repo from /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/oe-sdk-repo
repo: using cache for: oe-repo
not found other for: 
not found modules for: 
not found deltainfo for: 
not found updateinfo for: 
oe-repo: using metadata from Tue 22 Sep 2020 04:17:46 PM UTC.
No module defaults found
--> Starting dependency resolution
--> Finished dependency resolution
Error: 
 Problem: conflicting requests
  - nothing provides clang-cross-canadian-arm needed by packagegroup-cross-canadian-dynniq-flownode-acu-1.0-r0.x86_64_nativesdk_mingw32
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)

ERROR: Logfile of failure stored in: /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/dynniq_flownode_acu-poky-linux-gnueabi/core-image-minimal/1.0-r0/temp/log.do_populate_sdk.50218
ERROR: Task (/home/wdobbe/yocto/flownode_zeus/sources/poky/meta/recipes-core/images/core-image-minimal.bb:do_populate_sdk) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2529 tasks of which 2528 didn't need to be rerun and 1 failed.
```

Temporarily removing the meta-browser and meta-clang layers from bblayers fixes the problem, but then I also have to disable my .bbappend files. So it would nice if there is an easier way to fix this.

Thanks in advance.

kraj, 2020-09-22:
on zeus we have `CLANGSDK ??= "1"` in `conf/layer.conf`,  and I think your setting it to '' is not taking effect. Can you change it in conf/layer.conf and see if it helps ?

wdobbe, 2020-09-22:
Changing CLANGSDK in meta-clang/conf/layer.conf gives the exact same error.

Moreover, when I don't add CLANGSDK = "" in my local.conf I get all kinds of compile errors, so I think it has (some) effect.

kraj, 2020-09-23:
@wdobbe thats seems to not good. Lets try to fix it. meta-clang is supposed to be dormant usually unless invoked with TOOLCHAIN = "clang" and CLANGSDK = "1" is set. what does `bitbake -e <your-image> | grep ^CLANGSDK=" say ?

wdobbe, 2020-09-23:
Hi Khem,

The output of the bitbake -e command:

```
> bitbake -e | grep '^CLANGSDK='
CLANGSDK=""
```
When I build without the CLANGSDK="" line in local.conf, the first compile error that occurs (several other packages fail as well) is:

```
wdobbe@dev-linux-bs03:~/yocto/flownode_zeus/build-flownode-acu> bitbake core-image-minimal -c populate_sdk
WARNING: You have included the meta-virtualization layer, but 'virtualization' has not been enabled in your DISTRO_FEATURES. Some bbappend files may not take effect. See the meta-virtualization README for details on enabling virtualization support.
WARNING: Host distribution "sles-15.2" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################################################################| Time: 0:00:00
Loaded 4235 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################################| Time: 0:00:00
Parsing of 3036 .bb files complete (3028 cached, 8 parsed). 4243 targets, 484 skipped, 2 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "dynniq-flownode-acu"
DISTRO               = "dynniq-flownode-acu-bsp"
DISTRO_VERSION       = "99.9.9"
TUNE_FEATURES        = "arm vfp cortexa9 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:f2eb22a8783f1eecf99bd4042695bab920eed00e"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      
meta-webserver       
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:8c136c4362d0f14ceb5cceb27467ad9db5e9f93d"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:63887634fab8c6305f0c825b93d1087c72ee2b14"
meta-qt5             = "HEAD:1861b68525dc8566be6bc8d70e4359a3b589e4ca"
meta-rust            = "HEAD:5cda04c7c2158fc982502ca1e7893f71e4ca1729"
meta-browser         = "HEAD:830ef438e81ba5fc915b1855e69f02b2c286b21a"
meta-virtualization  = "HEAD:48ec1b49871fef7d331ac1d0680a48d362c7482b"
meta-streamline      = "master:d5b2b9ce79e1e7126842bc3c4ea5f3ca912490d7"
meta-clang           = "zeus:711e593d5984aad3bf35c51b7ac4482982bc16c7"
meta-mingw           = "zeus:756963cc28ebc163df7d7f4b4ee004c18d3d3260"
meta-dynniq          = "master:d3d6f6fabbe1ad9abfa6e77e832b88c3802c169d"
meta-dynniq-flownode-acu = "feature/BENG-86-upgrade-yocto-zeus:203a1db168b0202a902952bcffaf349951a848cd"

Initialising tasks: 100% |###########################################################################################################| Time: 0:00:00
Sstate summary: Wanted 323 Found 232 Missed 91 Current 410 (71% match, 87% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: nativesdk-ncurses-6.1+20190803-r0 do_compile: oe_runmake failed
ERROR: nativesdk-ncurses-6.1+20190803-r0 do_compile: Execution of '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/temp/run.do_compile.69097' failed with exit code 1:
make: Entering directory '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/build/narrowc'
cd include && make DESTDIR="" RPATH_LIST="/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-mingw32/usr/lib" libs
make[1]: Entering directory '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/build/narrowc/include'
make[1]: Nothing to be done for 'libs'.
make[1]: Leaving directory '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/build/narrowc/include'
cd ncurses && make DESTDIR="" RPATH_LIST="/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-mingw32/usr/lib" libs
make[1]: Entering directory '/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/build/narrowc/ncurses'
x86_64-pokysdk-mingw32-gcc --sysroot=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0/recipe-sysroot -DHAVE_CONFIG_H -I../ncurses -I. -I../../../git/ncurses -I../include -I../../../git/ncurses/../include -D_XOPEN_SOURCE=600 -D_POSIX_C_SOURCE=199506L -D_FILE_OFFSET_BITS=64 -DNDEBUG -DWINVER=0x0501 -DWIN32_LEAN_AND_MEAN -O2 -pipe -fmacro-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0=/usr/src/debug/nativesdk-ncurses/6.1+20190803-r0                      -fdebug-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-mingw32-pokysdk-mingw32/nativesdk-ncurses/6.1+20190803-r0=/usr/src/debug/nativ [... cut ...]
ERROR: Task (virtual:nativesdk:/home/wdobbe/yocto/flownode_zeus/sources/poky/meta/recipes-core/ncurses/ncurses_6.1+20190803.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1031 tasks of which 1030 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:nativesdk:/home/wdobbe/yocto/flownode_zeus/sources/poky/meta/recipes-core/ncurses/ncurses_6.1+20190803.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
wdobbe@dev-linux-bs03:~/yocto/flownode_zeus/build-flownode-acu>
```

wdobbe, 2020-09-24:
When I add CLANGSDK = "" in my local.conf and then comment out the following line in meta-clang/recipes-core/packagegroups/packagegroup-cross-canadian.bbappend

`RDEPENDS_${PN} += "${@all_multilib_tune_values(d, 'CLANGCROSSCANADIAN')}"`

the MinGW SDK builds correctly.

kraj, 2020-09-25:
it seems the dependency is not being excluded as expected. We need to inspect a bit.

bluca, 2020-12-07:
hi, any update on this issue?

kraj, 2020-12-07:
I guess status is same if you have additional information please add it here

congntsoc, 2022-10-03:
Hello guys, as of today, I am trying to make meta-clang work with meta-mingw as well, and by disabling libtirpc and libnsl, I think I have passed the issues above. But now I got this from clang linker:
```
|     Run Build Command(s):ninja cmTC_2b543 && [1/2] Building C object CMakeFiles/cmTC_2b543.dir/testCCompiler.c.obj
|     [2/2] Linking C executable cmTC_2b543.exe
|     FAILED: cmTC_2b543.exe
|     : && /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-clang -target x86_64-w64-ming
w32   --sysroot=/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot  -O2 -pipe -fmacro-prefix-map=/home/yocto/poky/build/tmp/work/x86_64-nativesd
k-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0=/usr/src/debug/nativesdk-clang/14.0.3-r0                      -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativ
esdk-clang/14.0.3-r0=/usr/src/debug/nativesdk-clang/14.0.3-r0                      -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe
-sysroot=                      -fdebug-prefix-map=/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native= -target x86_64-w64-mingw32   --sys$
oot=/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot  -Wl,-O1                                            -Wl,-rpath-link,/home/yocto/poky/bui$
d/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-w64-mingw32/usr/lib/..
     -Wl,-rpath,/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-w64-mingw32/usr/lib/..    -Wl,-dynamic-linker,/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-w64-mingw32/lib/ld-linux-x86-64.$
o.2 -Wl,-O1                                            -Wl,-rpath-link,/home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot/usr/local/oe-sdk-hard$
oded-buildpath/sysroots/x86_64-w64-mingw32/usr/lib/..                                            -Wl,-rpath,/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-w64-mingw32/usr/lib/..    -Wl,-dynamic-l$
nker,/usr/local/oe-sdk-hardcoded-buildpath/sysroots/x86_64-w64-mingw32/lib/ld-linux-x86-64.so.2 CMakeFiles/cmTC_2b543.dir/testCCompiler.c.obj -o cmTC_2b543.exe -Wl,--out-implib,libcmTC_2b543.dll.a -Wl,-$
major-image-version,0,--minor-image-version,0  -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 && :
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find crt2.o: No such f$
le or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find crtbegin.o: No su$
h file or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find -lgcc: No such fi$
e or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find -lgcc_eh: No such
file or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find -lgcc: No such fi$
e or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find -lgcc_eh: No such
file or directory
|     /home/yocto/poky/build/tmp/work/x86_64-nativesdk-mingw32-w64-mingw32/nativesdk-clang/14.0.3-r0/recipe-sysroot-native/usr/bin/x86_64-w64-mingw32/x86_64-w64-mingw32-ld: cannot find crtend.o: No such
file or directory
|     clang-14: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.`
`
```

For configuration, I have tried to enable only CLANGSDK option in my local.conf
```
#TOOLCHAIN = "clang"
#RUNTIME = "llvm"
#LIBCPLUSPLUS = "-stdlib=libc++"
CLANGSDK = "1"
```

It would be great if you could suggest an approach for me to continue working on this.
Thanks

kraj, 2022-10-04:
@congntsoc I have not build SDK for mingw, so perhaps I never see these problems. However, problem seems to be that its looking for nativesdk gcc runtime, So can you check if these libraries are in the nativesdk-sysroot when building nativesdk-clang. Secondly, do we need to use mingw runtime instead here, I am not sure.

congntsoc, 2022-10-05:
@kraj Thank you very much for your answer. Actually, I can find all of the files/libraries needed by clang like crt2.o, crtbegin...but for some reason, the linker can not detects/recognizes those files. I will try disabling the mingw runtime to see if it works.
Btw, I am trying to build llvm SDK for windows, do you have any other suggestions to do it other than mingw?

kraj, 2022-10-05:
> @kraj Thank you very much for your answer. Actually, I can find all of the files/libraries needed by clang like crt2.o, crtbegin...but for some reason, the linker can not detects/recognizes those [...]

I think mingw is best bet so you are on right path