kraj/meta-clang #235: clang is not build in yocto for riscv arch.

cghalibkhan, 2019-12-03:
Hi guys,

**Describe the bug**
clang is not build in yocto for RISCV arch.
**To Reproduce**
I was trying to build clang ( RISCV arch ) in yocto, for that i have followed this step's,

step1: git clone git://git.openembedded.org/openembedded-core
step2: git clone https://github.com/openembedded/meta-openembedded.git
step3: cd ${PWD}/openembedded-core && git clone git://git.openembedded.org/bitbake
step4: cd ${PWD}/openembedded-core && git clone https://github.com/riscv/meta-riscv.git
step5: mv ${PWD}/openembedded-core/meta-riscv/recipes-devtools/openocd ${PWD}/openembedded-core/meta-riscv/recipes-devtools/1openocd && cp -r ${PWD}/riscv-yocto/meta-openembedded/
step6: meta-oe/recipes-devtools/openocd ${PWD}/openembedded-core/meta-riscv/recipes-devtools/
step7: cd ${PWD}/openembedded-core && git clone https://github.com/kraj/meta-clang.git

step8: source oe-init-build-env

step9: export PATH=$(PWD)/openembedded-core/bitbake/bin:$(PATH)

step10: cd build 

step11: bitbake-layers add-layer ../meta-riscv
step12: bitbake-layers add-layer ../meta-clang

step13: Added TOOLCHAIN = "clang" >> ./conf/local.conf
              MACHINE = "freedom-u540" >> ./conf/local.conf

step14: bitbake meta-toolchain

step15: bitbake core-image-full-cmdline

**Expected behavior**
output review : 

                          Here i haven't find clang of Target--riscv, But one thing i have noticed that clang for X86 has Build in following PATH but not for RISCV, I have attached log file's for your reference. 

${PWD}/openembedded-core/build/tmp-glibc/work/freedom_u540-oe-linux/core-image-full-cmdline/1.0-r0/recipe-sysroot-native/usr/bin/clang
 
${PWD}/openembedded-core/build/tmp-glibc/work/x86_64-linux/clang-cross-riscv64/9.0.1-r0/recipe-sysroot-native/usr/bin/clang

${PWD}/openembedded-core/build/tmp-glibc/work/x86_64-linux/clang-crosssdk-x86_64/9.0.1-r0/recipe-sysroot-native/usr/bin/clang

${PWD}/openembedded-core/build/tmp-glibc/work/x86_64-nativesdk-oesdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/clang

**Desktop :**
 - OS: Ubuntu 
 - Browser [chrome]
 - Version: 16

**Additional context**
 
            Can anyone please help me to build clang for RISCV arch,

Thank you,
Ghalib khan.c

closed 2019-12-03 by cghalibkhan

kraj, 2019-12-03:
@cghalibkhan please use master or zeus branch for clang/risc-v since the backend support is new in clang9 and those branches are only ones supporting clang9 at the moment.

cghalibkhan, 2019-12-04:
Here are the log files, Maybe this log file will be useful for you guys.
[Uploading 20191203051840.log…]()
[Uploading 20191203094900.log…]()
[Uploading 20191203112549.log…]()
[Uploading 20191203115709.log…]()
[Uploading 20191203120701.log…]()