kraj/meta-clang #236: Segfault in compilation of nativesdk-clang (Linking llvm-tblgen)

rchard2scout, 2019-12-04:
**Describe the bug**
A segfault occurs during the linking of bin/llvm-tblgen

When trying to build an SDK, nativesdk-clang is compiled using the just previously compiled clang-native. During the linking of bin/llvm-tblgen, a segfault occurs here.

**Additional context**
I'm using latest zeus (commit 4a8b6e8) and Poky version yocto-3.0.
The complete command invocation from log.do_compile (with some proprietary names and long paths substituted, and all arguments put on their own lines for readability):

```
ERROR: nativesdk-clang-9.0.0-r0 do_compile: Execution of 'WORKDIR/nativesdk-clang/9.0.0-r0/temp/run.do_compile.129828' failed with exit code 1:
[1/4448] Linking CXX executable bin/llvm-tblgen
FAILED: bin/llvm-tblgen 
: && WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-SYSTEMNAME-linux/x86_64-SYSTEMNAME-linux-clang++
  --sysroot=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot
  -O2 
  -pipe
  -fmacro-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0=/usr/src/debug/nativesdk-clang/9.0.0-r0
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0=/usr/src/debug/nativesdk-clang/9.0.0-r0
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot=
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot-native=
  -fPIC
  -fvisibility-inlines-hidden
  -Werror=date-time
  -Werror=unguarded-availability-new
  -std=c++11
  -Wall
  -Wextra
  -Wno-unused-parameter
  -Wwrite-strings
  -Wcast-qual
  -Wmissing-field-initializers
  -pedantic
  -Wno-long-long
  -Wimplicit-fallthrough
  -Wcovered-switch-default
  -Wno-noexcept-type
  -Wnon-virtual-dtor
  -Wdelete-non-virtual-dtor
  -Wstring-conversion
  -fdiagnostics-color
  -ffunction-sections
  -fdata-sections
  -flto=thin
  -DNDEBUG  
  --sysroot=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot 
  -O2 
  -pipe
  -fmacro-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0=/usr/src/debug/nativesdk-clang/9.0.0-r0
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0=/usr/src/debug/nativesdk-clang/9.0.0-r0
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot=
  -fdebug-prefix-map=WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot-native= 
  -Wl,-O1
  -Wl,-rpath-link,WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/usr/lib/..
  -Wl,-rpath,/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/usr/lib/.. 
  -Wl,-dynamic-linker,/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/lib/ld-linux-x86-64.so.2
  -fuse-ld=gold
  -Wl,-O1
  -Wl,-rpath-link,WORKDIR/nativesdk-clang/9.0.0-r0/recipe-sysroot/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/usr/lib/..
  -Wl,-rpath,/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/usr/lib/.. 
  -Wl,-dynamic-linker,/SDKDESTDIR/g/sysroots/x86_64-SYSTEMNAME-linux/lib/ld-linux-x86-64.so.2
  -fuse-ld=gold
  -flto=thin
  -Wl,-allow-shlib-undefined
  -Wl,-O3
  -Wl,--gc-sections
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/AsmMatcherEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/AsmWriterEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/AsmWriterInst.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/Attributes.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CallingConvEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeEmitterGen.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenDAGPatterns.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenHwModes.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenInstruction.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenMapTable.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenRegisters.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenSchedule.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CodeGenTarget.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DAGISelEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DAGISelMatcherEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DAGISelMatcherGen.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DAGISelMatcherOpt.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DAGISelMatcher.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DFAPacketizerEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/DisassemblerEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/ExegesisEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/FastISelEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/FixedLenDecoderEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/GlobalISelEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/InfoByHwMode.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/InstrInfoEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/InstrDocsEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/IntrinsicEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/OptParserEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/PredicateExpander.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/PseudoLoweringEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/RISCVCompressInstEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/RegisterBankEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/RegisterInfoEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/SDNodeProperties.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/SearchableTableEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/SubtargetEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/SubtargetFeatureInfo.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/TableGen.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/Types.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/X86DisassemblerTables.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/X86EVEX2VEXTablesEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/X86FoldTablesEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/X86ModRMFilters.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/X86RecognizableInstr.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/WebAssemblyDisassemblerEmitter.cpp.o
  utils/TableGen/CMakeFiles/llvm-tblgen.dir/CTagsEmitter.cpp.o
  -o bin/llvm-tblgen
  -Wl,-rpath,"\$ORIGIN/../lib"
  lib/libLLVMSupport.a
  lib/libLLVMTableGen.a
  -lpthread
  lib/libLLVMSupport.a
  -lz
  -lrt
  -ldl
  -lpthread
  -lm
  lib/libLLVMDemangle.a && :
clang-9: error: unable to execute command: Segmentation fault (core dumped)
clang-9: error: linker command failed due to signal (use -v to see invocation)
```

kraj, 2019-12-04:
@rchard2scout it might be that LTO is causing it for you. Perhaps you can try removing it
from `recipes-devtools/clang/clang_git.bb`
```
PACKAGECONFIG_class-nativesdk = "rtti eh libedit thin-lto"
```
remove `thin-lto` from above line

rchard2scout, 2019-12-05:
Thanks, that fixed it! Do you want to take it, or should I keep it in a local bbappend? Or do we need to figure out what the root cause is here (a bug in upstream clang perhaps)?

kraj, 2019-12-05:
it seems it happens on some hosts not all so i would like to see if its a clang bug before disabling lto

reid-p, 2019-12-06:
Thanks for the tip. 
Disabling LTO fixedthe  segfault on my machines.
I'd tried Centos 7 and Centos 8 as the build host with the same seg fault.

kraj, 2019-12-07:
should be fixed now

closed 2019-12-07 by kraj