kraj/meta-clang #241: Compile error on master & zeus with clang-native and shared-libs

priv-kweihmann, 2019-12-06:
**Describe the bug**
For another native tool I need to have "shared-libs" from the PACKAGE_CONFIG enabled.
So I made an bbappend with `PACKAGE_CONFIG_class_native += "shared_libs"`.
This works on warrior-branch pretty well. On zeus and master I get the stack trace shown below.
If I leave out the PACKAGE_CONFIG append compilation works as expected, but then I have the issues further on in the build chain.

**To Reproduce**
Steps to reproduce the behavior:
1. poky:zeus, meta-clang:zeus, my layer with default `qemu-x86-64` target
2. Create "clang_%.bbappend" in my layer with `PACKAGE_CONFIG_class_native += "shared_libs"`
3. run `bitbake clang-native`
4. See error

Reproducible on Debian-9, Ubuntu 16.04 and 19.04 host system

**Expected behavior**

- compilation works

**Additional context**

here is the stack trace
> $ORIGIN::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::" lib/libclangAST.a lib/libclangBasic.a lib/libclangDriver.a lib/libclangFrontend.a lib/libclangIndex.a lib/libclangLex.a lib/libclangSema.a lib/libclangSerialization.a lib/libclangTooling.a lib/libclangARCMigrate.a lib/libclangTidyPlugin.a lib/libclangIncludeFixerPlugin.a lib/libclangTidyAbseilModule.a lib/libclangTidyAndroidModule.a lib/libclangTidyBoostModule.a lib/libclangTidyCERTModule.a lib/libclangTidyFuchsiaModule.a lib/libclangTidyHICPPModule.a lib/libclangTidyBugproneModule.a lib/libclangTidyCppCoreGuidelinesModule.a lib/libclangTidyGoogleModule.a lib/libclangTidyLLVMModule.a lib/libclangTidyMiscModule.a lib/libclangTidyModernizeModule.a lib/libclangTidyObjCModule.a lib/libclangTidyOpenMPModule.a lib/libclangTidyPerformanceModule.a lib/libclangTidyPortabilityModule.a lib/libclangTidyReadabilityModule.a lib/libclangTidyZirconModule.a lib/libclangTidyMPIModule.a lib/libclangTidyUtils.a lib/libclangTidy.a lib/libclangStaticAnalyzerFrontend.a lib/libclangToolingRefactoring.a lib/libclangStaticAnalyzerCheckers.a lib/libclangStaticAnalyzerCore.a lib/libclangCrossTU.a lib/libclangIndex.a lib/libclangIncludeFixer.a lib/libfindAllSymbols.a lib/libclangTooling.a lib/libclangFrontend.a lib/libclangDriver.a lib/libclangParse.a lib/libclangSerialization.a lib/libclangSema.a lib/libclangEdit.a lib/libclangAnalysis.a lib/libclangFormat.a lib/libclangToolingInclusions.a lib/libclangToolingCore.a lib/libclangRewrite.a lib/libclangASTMatchers.a lib/libclangAST.a lib/libclangLex.a lib/libclangBasic.a lib/libLLVM-9.so -lpthread && :
> 2019-12-06T06:04:39.2987224Z | tools/clang/tools/libclang/CMakeFiles/libclang.dir/CIndexer.cpp.o: In function `clang::CIndexer::getClangResourcesPath[abi:cxx11]()':
> 2019-12-06T06:04:39.2987601Z | CIndexer.cpp:(.text._ZN5clang8CIndexer21getClangResourcesPathB5cxx11Ev+0x8b): undefined reference to `dladdr'
> 2019-12-06T06:04:39.2987729Z | collect2: error: ld returned 1 exit status
> 2019-12-06T06:04:39.2987851Z | [4641/4785] Building CXX object tools/lldb/tools/driver/CMakeFiles/lldb.dir/Driver.cpp.o
> 2019-12-06T06:04:39.2987965Z | ninja: build stopped: subcommand failed.
> 2019-12-06T06:04:39.2988072Z | WARNING: exit code 1 from a shell command.
> 2019-12-06T06:04:39.2988155Z | 
> 2019-12-06T06:04:39.2988453Z NOTE: recipe clang-native-9.0.1-r0: task do_compile: Failed
> 2019-12-06T06:04:41.1101780Z NOTE: Tasks Summary: Attempted 1255 tasks of which 0 didn't need to be rerun and 1 failed.

kraj, 2019-12-06:
It seems its missing linking with `-ldl` somewhere in lldb, perhaps test it by specifying in recipe ?

priv-kweihmann, 2019-12-06:
What I tried is `LDFLAGS_append_class-native = " -ldl"`, without any help.
Any suggestion where to add it inside the recipe?

kraj, 2019-12-12:
this should be fixed on master now.

closed 2019-12-12 by kraj

martin-ejdestig, 2020-05-07:
@kraj I also get this linking error on thud. Worth a cherry-pick?

kraj, 2020-05-07:
> @kraj I also get this linking error on thud. Worth a cherry-pick?

@martin-ejdestig I dont have active setup for thud but if you tested the patch feel free to send a pull

maxatka, 2020-06-05:
I have the same issue on my Gentoo x86_64, building sys-devel/clang-11.0.0.9999:
```
[1599/1602] : && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -march=native -O2 -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -pedantic -Wno-long-long  -Wl,-O1 -Wl,--as-needed -Wl,-z,defs -Wl,-z,nodelete   -Wl,-rpath-link,/var/tmp/portage/sys-devel/clang-11.0.0.9999/work/x/y/clang-abi_x86_64.amd64/./lib64  -Wl,-O3 -Wl,--gc-sections  -Wl,--version-script,"/var/tmp/portage/sys-devel/clang-11.0.0.9999/work/x/y/clang-abi_x86_64.amd64/tools/libclang/libclang.exports" -shared -Wl,-soname,libclang.so.11git -o lib64/libclang.so.11 tools/libclang/CMakeFiles/libclang.dir/ARCMigrate.cpp.o tools/libclang/CMakeFiles/libclang.dir/BuildSystem.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndex.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexCXX.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexCodeCompletion.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexHigh.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexInclusionStack.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexUSRs.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexer.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXComment.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXCursor.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXIndexDataConsumer.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXCompilationDatabase.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXLoadedDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXSourceLocation.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXStoredDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXString.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXType.cpp.o tools/libclang/CMakeFiles/libclang.dir/Indexing.cpp.o tools/libclang/CMakeFiles/libclang.dir/FatalErrorHandler.cpp.o -L/usr/lib/llvm/11/lib64 -Wl,-rpath,"\$ORIGIN/../lib64"  lib64/libclangAST.a  lib64/libclangBasic.a  lib64/libclangDriver.a  lib64/libclangFrontend.a  lib64/libclangIndex.a  lib64/libclangLex.a  lib64/libclangSema.a  lib64/libclangSerialization.a  lib64/libclangTooling.a  lib64/libclangARCMigrate.a  lib64/libclangFormat.a  lib64/libclangToolingInclusions.a  lib64/libclangToolingCore.a  lib64/libclangFrontend.a  lib64/libclangDriver.a  lib64/libclangParse.a  lib64/libclangSerialization.a  lib64/libclangSema.a  lib64/libclangEdit.a  lib64/libclangRewrite.a  lib64/libclangAnalysis.a  lib64/libclangASTMatchers.a  lib64/libclangAST.a  lib64/libclangLex.a  lib64/libclangBasic.a  /usr/lib/llvm/11/lib64/libLLVM-11git.so && :
FAILED: lib64/libclang.so.11 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -fPIC -march=native -O2 -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -Wimplicit-fallthrough -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -pedantic -Wno-long-long  -Wl,-O1 -Wl,--as-needed -Wl,-z,defs -Wl,-z,nodelete   -Wl,-rpath-link,/var/tmp/portage/sys-devel/clang-11.0.0.9999/work/x/y/clang-abi_x86_64.amd64/./lib64  -Wl,-O3 -Wl,--gc-sections  -Wl,--version-script,"/var/tmp/portage/sys-devel/clang-11.0.0.9999/work/x/y/clang-abi_x86_64.amd64/tools/libclang/libclang.exports" -shared -Wl,-soname,libclang.so.11git -o lib64/libclang.so.11 tools/libclang/CMakeFiles/libclang.dir/ARCMigrate.cpp.o tools/libclang/CMakeFiles/libclang.dir/BuildSystem.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndex.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexCXX.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexCodeCompletion.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexHigh.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexInclusionStack.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexUSRs.cpp.o tools/libclang/CMakeFiles/libclang.dir/CIndexer.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXComment.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXCursor.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXIndexDataConsumer.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXCompilationDatabase.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXLoadedDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXSourceLocation.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXStoredDiagnostic.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXString.cpp.o tools/libclang/CMakeFiles/libclang.dir/CXType.cpp.o tools/libclang/CMakeFiles/libclang.dir/Indexing.cpp.o tools/libclang/CMakeFiles/libclang.dir/FatalErrorHandler.cpp.o -L/usr/lib/llvm/11/lib64 -Wl,-rpath,"\$ORIGIN/../lib64"  lib64/libclangAST.a  lib64/libclangBasic.a  lib64/libclangDriver.a  lib64/libclangFrontend.a  lib64/libclangIndex.a  lib64/libclangLex.a  lib64/libclangSema.a  lib64/libclangSerialization.a  lib64/libclangTooling.a  lib64/libclangARCMigrate.a  lib64/libclangFormat.a  lib64/libclangToolingInclusions.a  lib64/libclangToolingCore.a  lib64/libclangFrontend.a  lib64/libclangDriver.a  lib64/libclangParse.a  lib64/libclangSerializatio [... cut ...] clangAST.a  lib64/libclangLex.a  lib64/libclangBasic.a  /usr/lib/llvm/11/lib64/libLLVM-11git.so && :
/usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: tools/libclang/CMakeFiles/libclang.dir/CIndexer.cpp.o: in function `clang::CIndexer::getClangResourcesPath[abi:cxx11]()':
CIndexer.cpp:(.text._ZN5clang8CIndexer21getClangResourcesPathB5cxx11Ev+0x89): undefined reference to `dladdr'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

```