kraj/meta-clang #247: meta-clang's llvm-config is not compatible with MULTILIBS

cosmo0920, 2019-12-11:
**Describe the bug**
One of the our target boards(RZ/G2E)'s Yocto default **conf/local.conf** specifies `MULTILIBS = "multilib:lib32"` and `DEFAULTTUNE_virtclass-multilib-lib32 = "armv7vethf-neon"` to be able to run 32bit ARMv7 binaries.
So, built binaries will be installed in `/usr/lib64/` instead of `/usr/lib`.

Because our SDK environment does not assume `/usr/lib` for library installation directory.
Instead, `/usr/lib64` is used for 64bit libraries and shared object. And `/usr/lib32` is used for 32bit objects.

ref: https://llvm.org/docs/CMake.html#frequently-used-cmake-variables

**To Reproduce**
Steps to reproduce the behavior:
1. Specify `MULTILIBS = "multilib:lib32"` and `DEFAULTTUNE_virtclass-multilib-lib32 = "armv7vethf-neon"` in local.conf
2. Add meta-clang layer
3. bitbake clang-cross-aarch64
4. See error

### Error

```
|  0:05.42 configure: error: compiler is incompatible with sanitize options
|  0:05.42 DEBUG: <truncated - see config.log for full output>
|  0:05.42 DEBUG: /home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: cannot find /home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot/usr/lib/clang/8.0.1/lib/linux/libclang_rt.builtins-aarch64.a: No such file or directory
|  0:05.42 DEBUG: clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
|  0:05.42 DEBUG: configure: failed program was:
|  0:05.42 DEBUG: #line 2836 "configure"
|  0:05.42 DEBUG: #include "confdefs.h"
|  0:05.42 DEBUG:
|  0:05.42 DEBUG: int main() {
|  0:05.42 DEBUG: XtMalloc()
|  0:05.42 DEBUG: ; return 0; }
|  0:05.42 DEBUG: configure:4123: /home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang -march=armv8-a -mtune=cortex-a53 --rtlib=compiler-rt --stdlib=libc++ -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments --sysroot=/home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot -std=gnu99 -o conftest  -Os -fsigned-char -fno-strict-aliasing -Qunused-arguments  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed  conftest.c  1>&5
|  0:05.42 DEBUG: /home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: cannot find /home/vagrant/rzg2-yoshi/build/tmp/work/aarch64-poky-linux/firefox/68.0esr-r0/recipe-sysroot/usr/lib/clang/8.0.1/lib/linux/libclang_rt.builtins-aarch64.a: No such file or directory
|  0:05.42 DEBUG: clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
|  0:05.42 DEBUG: configure: failed program was:
|  0:05.42 DEBUG: #line 4116 "configure"
|  0:05.42 DEBUG: #include "confdefs.h"
|  0:05.42 DEBUG:
|  0:05.42 DEBUG: int main() {
|  0:05.42 DEBUG:
|  0:05.42 DEBUG: ; return 0; }
|  0:05.42 DEBUG: configure: error: compiler is incompatible with sanitize options
|  0:05.42 ERROR: old-configure failed
|  0:05.46 *** Fix above errors and then restart with\
|  0:05.46                "./mach build"
|  0:05.46 client.mk:111: recipe for target 'configure' failed
|  0:05.46 make: *** [configure] Error 1
| WARNING: exit code 2 from a shell command.
```

**Expected behavior**
meta-clang's llvm-config can work with `DEFAULTTUNE_virtclass-multilib-lib32` specified environment. 

llvm-config points to ${RECIPE_SYSROOT}/usr/**lib**/clang/8.0.1/lib/linux/ but actual libclang libraries are put in ${RECIPE_SYSROOT}/usr/**lib64**/clang/8.0.1/lib/linux/

LLVM insists that using `LLVM_LIBDIR_SUFFIX` to control installation directory suffix such as lib64 or lib32. 

We should handle library directory glitch it llvm-config with `LLVM_LIBDIR_SUFFIX`.

**Desktop (please complete the following information):**
 - OS: Ubuntu 
 - Version 16.04.6 LTS

**Additional context**
Add any other context about the problem here.

kraj, 2019-12-11:
Can you cook up a patch ?

cosmo0920, 2019-12-13:
I'm cooking up patches.... :hammer:

kraj, 2019-12-15:
@cosmo0920 is master we have addressed this issue by using `baselib` for `LIBCXX_CXX_ABI_LIBRARY_PATH` and `LLVM_LIBDIR_SUFFIX` but this patch computes `LLVM_LIBDIR_SUFFIX` locally using new function `get_libdir_suffix` is this needed on master as well ?

cosmo0920, 2019-12-16:
> this patch computes `LLVM_LIBDIR_SUFFIX` locally using new function `get_libdir_suffix` is this needed on master as well ?

Yes. Current master's implementation returns empty string on multiarch environment.

kraj, 2019-12-16:
Ok then could you prepare a patch for master as well please ? Usually we do the fix on master before backport to release branches

kraj, 2019-12-16:
thanks for patch, I think this issue is now fixed.

closed 2019-12-16 by kraj

cosmo0920, 2019-12-16:
Yep, this issue is fixed on master.