kraj/meta-clang #254: nativesdk-clang prevents the SDK to be built (thud)

chlongv, 2019-12-16:
I want to generate SDKs that include clang. I have done this by extending the `TOOLCHAIN_HOST_TASK` variable:

`TOOLCHAIN_HOST_TASK_append = " nativesdk-clang"`

However, this prevents the SDK to be generated successfully. It fails with with the **libcxx** package, for both ARM and PowerPC (the 2 archs I am interested in). They however fail with 2 different effors:

ARM:

```
$ MACHINE=qemuarm bitbake score-image-minimal:do_populate_sdk
[...]
DEBUG: Executing shell function do_configure
-- The C compiler identification is Clang 7.1.0
-- The CXX compiler identification is Clang 7.1.0
-- The ASM compiler identification is unknown
-- Found assembler: /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang
-- Check for working C compiler: /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang
-- Check for working C compiler: /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang++
-- Check for working CXX compiler: /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Warning: Did not find file Compiler/-ASM
-- No build type selected, default to Debug
-- Performing Test LLVM_NO_OLD_LIBSTDCXX
-- Performing Test LLVM_NO_OLD_LIBSTDCXX - Failed
CMake Error at cmake/modules/CheckCompilerVersion.cmake:38 (message):
  Host Clang must be able to find libstdc++4.8 or newer!
Call Stack (most recent call first):
  cmake/config-ix.cmake:12 (include)
  CMakeLists.txt:600 (include)

-- Configuring incomplete, errors occurred!
See also "/home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/build/CMakeFiles/CMakeOutput.log".
See also "/home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/build/CMakeFiles/CMakeError.log".
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_configure (log file is located at /home/val/yocto/build/tmp/work/armv5e-poky-linux-gnueabi/libcxx/7.1.0-r0/temp/log.do_configure.4287)

```

PowerPC:

```
$ MACHINE=qemuppc bitbake core-image-minimal:do_populate_sdk
[...]
DEBUG: Executing shell function do_configure
-- The C compiler identification is Clang 7.1.0
-- The CXX compiler identification is Clang 7.1.0
-- The ASM compiler identification is unknown
-- Found assembler: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang
-- Check for working C compiler: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang
-- Check for working C compiler: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang -- broken
CMake Error at /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/share/cmake-3.12/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"ninja" "cmTC_081c1"
    [1/2] Building C object CMakeFiles/cmTC_081c1.dir/testCCompiler.c.o
    [2/2] Linking C executable cmTC_081c1
    FAILED: cmTC_081c1 
    : && /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang -m32  -mcpu=7400  -mlittle-endian --rtlib=compiler-rt -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong   -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types    -m32  -mcpu=7400  -mlittle-endian --rtlib=compiler-rt -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong   -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot  -m32  -mcpu=7400  -mlittle-endian --rtlib=compiler-rt -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong   -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now CMakeFiles/cmTC_081c1.dir/testCCompiler.c.o  -o cmTC_081c1   && :
    /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-ld: cannot find /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot/usr/lib/clang/7.1.0/lib/linux/libclang_rt.builtins-powerpc.a: No such file or directory
    clang-7: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:50 (project)

-- Configuring incomplete, errors occurred!
See also "/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/build/CMakeFiles/CMakeOutput.log".
See also "/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/build/CMakeFiles/CMakeError.log".
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_configure (log file is located at /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/temp/log.do_configure.13319)

```

kraj, 2019-12-17:
can you reproduce this with master ?, if you can then we might be able to handle it faster.
secondly, ARM issue seems to be due to fact that you host gcc is < 5.x ( it needs C++11 )
PPC with 7.x might need aid and I do not test it regularly myself but if you post below file then we might see why compiler is failing on simple programs.

```
"/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/build/CMakeFiles/CMakeError.log".
```

chlongv, 2019-12-17:
I will try to reproduce it with master (zeus).

Regarding the ARM issue due to the gcc version < 5.x, that's no the case:

```
$ gcc -v
[...]
gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2)
```

Here is the file you mentioned above (CMakeError.log):
```
Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang 
Build flags: ;-m32;;-mcpu=7400;;-mlittle-endian;--rtlib=compiler-rt;-Wno-error=unused-command-line-argument;-Qunused-arguments;-fstack-protector-strong;;;-D_FORTIFY_SOURCE=2;-Wformat;-Wformat-security;-Werror=format-security;;--sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot;;-O2;-pipe;-g;-feliminate-unused-debug-types;;;;-m32;;-mcpu=7400;;-mlittle-endian;--rtlib=compiler-rt;-Wno-error=unused-command-line-argument;-Qunused-arguments;-fstack-protector-strong;;;-D_FORTIFY_SOURCE=2;-Wformat;-Wformat-security;-Werror=format-security;;--sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot
Id flags:  

The output was:
1
/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-ld: cannot find /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot/usr/lib/clang/7.1.0/lib/linux/libclang_rt.builtins-powerpc.a: No such file or directory
clang-7: error: linker command failed with exit code 1 (use -v to see invocation)

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-clang++ 
Build flags: ;-m32;;-mcpu=7400;;-mlittle-endian;--rtlib=compiler-rt;-Wno-error=unused-command-line-argument;-Qunused-arguments;-fstack-protector-strong;;;-D_FORTIFY_SOURCE=2;-Wformat;-Wformat-security;-Werror=format-security;;--sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot;;-O2;-pipe;-g;-feliminate-unused-debug-types;;;-fvisibility-inlines-hidden;;-m32;;-mcpu=7400;;-mlittle-endian;--rtlib=compiler-rt;-Wno-error=unused-command-line-argument;-Qunused-arguments;-fstack-protector-strong;;;-D_FORTIFY_SOURCE=2;-Wformat;-Wformat-security;-Werror=format-security;;--sysroot=/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot
Id flags:  

The output was:
1
/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-ld: cannot find /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot/usr/lib/clang/7.1.0/lib/linux/libclang_rt.builtins-powerpc.a: No such file or directory
clang-7: error: linker command failed with exit code 1 (use -v to see invocation)

Checking whether the ASM compiler is GNU using "--version" did not match "(GNU assembler)|(GCC)|(Free Software Foundation)":
clang version 7.1.0 
Target: powerpc-poky-linux
Thread model: posix
InstalledDir: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux
Checking whether the ASM compiler is HP using "-V" did not match "HP C":
clang-7: error: argument to '-V' is missing (expected 1 value)
clang-7: error: no input files
Checking whether the ASM compiler is Intel using "--version" did not match "(ICC)":
clang version 7.1.0 
Target: powerpc-poky-linux
Thread model: posix
InstalledDir: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux
Checking whether the ASM compiler is SunPro using "-V" did not match "Sun C":
clang-7: error: argument to '-V' is missing (expected 1 value)
clang-7: error: no input files
Checking whether the ASM compiler is XL using "-qversion" did not match "XL C":
clang-7: error: unknown argument '-qversion', did you mean '--version'?
clang-7: error: no input files
Checking whether the ASM compiler is MSVC using "/?" did not match "Microsoft":
clang-7: error: no such file or directory: '/?'
clang-7: error: no input files
Checking whether the ASM compiler is TI using "-h" did not match "Texas Instruments":
clang-7: error: unknown argument: '-h'
clang-7: error: no input files
Checking whether the ASM compiler is IAR using "" did not match "IAR Assembler":
clang-7: error: no input files
Checking whether the ASM compiler is ARMCC using "" did not match "(ARM Compiler)|(ARM Assembler)":
clang-7: error: no input files
Checking whether the ASM compiler is NASM using "-v" did not match "(NASM version)":
clang version 7.1.0 
Target: powerpc-poky-linux
Thread model: posix
InstalledDir: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux
Checking whether the ASM compiler is YASM using "--version" did not match "(yasm)":
clang version 7.1.0 
Target: powerpc-poky-linux
Thread model: posix
InstalledDir: /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux
Checking whether the ASM compiler is ADSP using "-version" did not match "Analog Devices":
clang-7: error: unknown argument '-version', did you mean '--version'?
clang-7: error: no input files
Determining if the C compiler works failed with the following output:
[... cut ...]
/home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/powerpc-poky-linux/powerpc-poky-linux-ld: cannot find /home/val/yocto/build/tmp/work/ppc7400-poky-linux/libcxx/7.1.0-r0/recipe-sysroot/usr/lib/clang/7.1.0/lib/linux/libclang_rt.builtins-powerpc.a: No such file or directory
clang-7: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

The linker does not find the libclang_rt builtins for PowerPC (even though compiler_rt was built indeed).

kraj, 2019-12-17:
@chlongv using yoedistro `bitbake yoe-simple-image -c populate_sdk` succeeded for master on ppc as well as on arm.  I know there were compiler-rt issues on older releases on ppc so it might be coming your way. If you are not stuck to thud, I would recommend to use 'zeus' release since it will work like master.

chlongv, 2019-12-18:
Yes, I have just finished the build for both PowerPC and ARM with zeus and I can confirm that it succeeded.
I am not strictly stuck on thud, I could move on to zeus. However, some other groups in the organization are indeed stuck on thud and we would have preferred to stay in sync, but I can discuss that.
On the other hand, any opinion/idea on these compiler-rt issues on ppc ? And what about the ARM issue ?

kraj, 2019-12-19:
> Yes, I have just finished the build for both PowerPC and ARM with zeus and I can confirm that it succeeded.
> I am not strictly stuck on thud, I could move on to zeus. However, some other groups in [...]

I have seen ARM issue in past but dont have detailed memory on how it got fixed. you might have to dig into git logs perhaps. ppc issue is totally new, I never tried ppc with thud

kraj, 2020-02-26:
@chlongv do we need to do anything more wiith this ticket ? if not then please close it

chlongv, 2020-03-02:
No, nothing more to do. I have the confirmation that it does not work on thud and we have now moved to zeus (where it works).

closed 2020-03-02 by chlongv

lexra, 2021-05-28:
in meta-clang/recipes-devtools/clang/compiler-rt_git.bb, 

--                  -DCMAKE_AR=${STAGING_BINDIR_TOOLCHAIN}/${TARGET_PREFIX}llvm-ar \
++                  -DCMAKE_AR=${STAGING_BINDIR_TOOLCHAIN}/${TARGET_PREFIX}ar \

try if it works.