kraj/meta-clang #262: internal compiler error due to -gsplit-dwarf

saininav, 2019-12-24:
Build is failing at CentOS7, GCC version 4.8.5.

Triggered build with latest poky, meta-intel and meta-clang master branches.

Issue observed due to this patch:
https://github.com/kraj/meta-clang/commit/c885bf3a7dc210acd495be2c40147c11f4cf90e1

After setting -DLLVM_USE_SPLIT_DWARF=OFF, build is successful.

Error log:

FAILED: lib/Target/Mips/CMakeFiles/LLVMMipsCodeGen.dir/MipsAsmPrinter.cpp.o
| /home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/hosttools/g++  -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Ilib/Target/Mips -I/home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/work-shared/llvm-project-source-9.0.1-r0/git/llvm/lib/Target/Mips -Iinclude -I/home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/work-shared/llvm-project-source-9.0.1-r0/git/llvm/include -isystem/home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/work/x86_64-linux/clang-native/9.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -std=c++11 -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wno-maybe-uninitialized -Wdelete-non-virtual-dtor -Wno-comment -ffunction-sections -fdata-sections -DNDEBUG   -gsplit-dwarf -MD -MT lib/Target/Mips/CMakeFiles/LLVMMipsCodeGen.dir/MipsAsmPrinter.cpp.o -MF lib/Target/Mips/CMakeFiles/LLVMMipsCodeGen.dir/MipsAsmPrinter.cpp.o.d -o lib/Target/Mips/CMakeFiles/LLVMMipsCodeGen.dir/MipsAsmPrinter.cpp.o -c /home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/work-shared/llvm-project-source-9.0.1-r0/git/llvm/lib/Target/Mips/MipsAsmPrinter.cpp

| /home/pokybuild/yocto-autobuilder/yocto-worker/nightly-meta-intel-world/build/build/tmp/work-shared/llvm-project-source-9.0.1-r0/git/llvm/lib/Target/Mips/MipsAsmPrinter.cpp:1304:1: internal compiler error: in output_index_string, at dwarf2out.c:21891
|  }
|  ^

kraj, 2019-12-24:
yeah its pretty old compiler and clang anyway needs atleast gcc 5.x 

https://llvm.org/docs/GettingStarted.html#software

so using 4.x is proving lucky so far. However, I think we can make a patch to
convert -DLLVM_USE_SPLIT_DWARF into a packageconfig and perhaps keep it
disabled so it can build on centos7

saininav, 2019-12-26:
Issue is observed at Ubuntu 16.04/gcc 7.1.0 too.

kraj, 2019-12-26:
should be fixed in master now.

kraj, 2019-12-26:
please retest master.

saininav, 2019-12-27:
Build passed with this patch.

kraj, 2019-12-27:
fixed in master

closed 2019-12-27 by kraj