kraj/meta-clang #271: [Master] Builds fail with missing __divti3

jpuhlman, 2020-01-08:
Not sure whats going on here, but I am getting this failure. Any ideas?

| /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-montavista-linux/x86_64-montavista-linux-ld: projects/libcxx/src/CMakeFiles/cxx_shared.dir/filesystem/operations.cpp.o: in function `std::__1::chrono::__duration_cast<std::__1::chrono::duration<__int128, std::__1::ratio<1l, 1000000000l> >, std::__1::chrono::duration<__int128, std::__1::ratio<1l, 1l> >, std::__1::ratio<1l, 1000000000l>, true, false>::operator()(std::__1::chrono::duration<__int128, std::__1::ratio<1l, 1000000000l> > const&) const':
| /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-9.0.1-r0/git/libcxx/include/chrono:876: undefined reference to `__divti3'
| clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
| ninja: build stopped: subcommand failed.

jpuhlman, 2020-01-08:
Currently building on centos 8.

kraj, 2020-01-09:
@jpuhlman I have seen this error in past when GCC was used to build libcxx and another case was when using clang with --rtlib=compiler-rt, are there any customizations to such vars done ?

jpuhlman, 2020-01-09:
Hmmm. I don't see either cases of those in my builds. Ill try backing layers off in my build to see if its something coming from the distro configuration. I don't think I have any configurations that should muck things up.

Here is the line that looks like it fails on.
/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-montavista-linux/x86_64-montavista-linux-clang++   -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments --sysroot=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot -fPIC -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0=/usr/src/debug/libcxx/9.0.1-r0                      -fdebug-prefix-map=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0=/usr/src/debug/libcxx/9.0.1-r0                      -fdebug-prefix-map=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot-native=   -fvisibility-inlines-hidden  -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -std=c++11 -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers  -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -g  -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/build/lib64 -Wl,-z,defs -Wl,-z,nodelete  -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/build/lib64 -shared -Wl,-soname,libc++.so.1 -o lib64/libc++.so.1.0 projects/libcxx/src/CMakeFiles/cxx_shared.dir/algorithm.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/any.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/bind.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/charconv.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/chrono.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/condition_variable.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/condition_variable_destructor.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/debug.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/exception.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/functional.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/future.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/hash.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/ios.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/iostream.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/locale.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/memory.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/mutex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/mutex_destructor.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/new.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/optional.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/random.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/regex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/shared_mutex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/stdexcept.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/string.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/strstream.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/system_error.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/thread.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/typeinfo.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/utility.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/valarray.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/variant.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/vector.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/filesystem/operations.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/filesystem/directory_iterator.cpp.o  -Wl,-rpath,"\$ORIGIN/../lib64"  -nodefaultlibs  -lpthread  -lc  -lm  -lrt  -lc++abi

kraj, 2020-01-09:
I see `-nodefaultlibs -lpthread -lc -lm -lrt -lc++abi` which mean we are not linking in either `libgcc` or `compiler-rt` and this function should come from either of these runtimes. perhaps we need to add that to above since it now needs these for builtins during link.

kraj, 2020-01-09:
I think it seems we need to add output of `$CC --print-libgcc-file-name` to this linker cmd

jpuhlman, 2020-01-09:
> I think it seems we need to add output of $CC --print-libgcc-file-name to this linker cmd

That fixes the link.

kraj, 2020-01-09:
Cool. Can you cook a patch

jpuhlman, 2020-01-09:
clang version 9.0.1 (https://github.com/llvm/llvm-project c1a0a213378a458fbea1a5c77b315c7dce08fd05)
Target: x86_64-montavista-linux
Thread model: posix
InstalledDir: /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-montavista-linux
Found candidate GCC installation: /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0
Selected GCC installation: /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-montavista-linux/x86_64-montavista-linux-ld" --sysroot=/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot --eh-frame-hdr -m elf_x86_64 -shared -o lib64/libc++.so.1.0 /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/../../../lib64/crti.o /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/crtbeginS.o -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/build/lib64 -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/build/lib64 -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0 -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/../../x86_64-montavista-linux -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/../../../lib64 -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot//lib/../lib64 -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/../.. -L/home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot//lib -z defs -z nodelete -soname libc++.so.1 projects/libcxx/src/CMakeFiles/cxx_shared.dir/algorithm.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/any.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/bind.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/charconv.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/chrono.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/condition_variable.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/condition_variable_destructor.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/debug.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/exception.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/functional.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/future.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/hash.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/ios.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/iostream.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/locale.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/memory.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/mutex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/mutex_destructor.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/new.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/optional.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/random.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/regex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/shared_mutex.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/stdexcept.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/string.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/strstream.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/system_error.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/thread.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/typeinfo.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/utility.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/valarray.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/variant.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/vector.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/filesystem/operations.cpp.o projects/libcxx/src/CMakeFiles/cxx_shared.dir/filesystem/directory_iterator.cpp.o -rpath "\$ORIGIN/../lib64" -lpthread -lc -lm -lrt -lc++abi /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/crtendS.o /home/jpuhlman/builds/master/opencgx-x86-generic-64/project/tmp/work/corei7-64-montavista-linux/libcxx/9.0.1-r0/recipe-sysroot/usr/lib64//x86_64-montavista-linux/9.2.0/../../../lib64/crtn.o

jpuhlman, 2020-01-09:
Note adding -lgcc works as well. I am not as familiar with cmake configuration change, I am not sure where to make the change.

jpuhlman, 2020-01-10:
diff --git a/libcxx/src/CMakeLists.txt b/libcxx/src/CMakeLists.txt
index 31cd243..a194fe8 100644
--- a/libcxx/src/CMakeLists.txt
+++ b/libcxx/src/CMakeLists.txt
@@ -234,7 +234,7 @@ if (LIBCXX_ENABLE_SHARED)
     llvm_setup_rpath(cxx_shared)
   endif()
   cxx_link_system_libraries(cxx_shared)
-  target_link_libraries(cxx_shared PRIVATE ${LIBCXX_LIBRARIES})
+  target_link_libraries(cxx_shared PRIVATE ${LIBCXX_LIBRARIES} "-lgcc")
   set_target_properties(cxx_shared
     PROPERTIES
       COMPILE_FLAGS "${LIBCXX_COMPILE_FLAGS}"

This seems to do it, but like I said, my cmake-foo is weak.

jpuhlman, 2020-01-16:
Was this how we wanted to do it or is there another suggestion?

kraj, 2020-01-16:
I think hardlinking with libgcc should be replaced with automatic detection of compiler runtime something like
`$CC --print-libgcc-file-name` to find it instead

kraj, 2020-02-12:
should be fixed now

closed 2020-02-12 by kraj