kraj/meta-clang #280: compile with clang, link with gcc-toolchain?

deblauwetom, 2020-02-12:
Hello,

I am using your warrior layer and it works very well, nice job! I have now the problem that I am developing using nvidia jetpack, which provides precompiled libraries with gcc 7. So my own application comiles and links with gcc 7 and these binary libs. However, now I want to use clang because gcc is slow :p I used these options in my package:
```
TOOLCHAIN = "clang"
LIBCPLUSPLUS = ""
COMPILER_RT = ""
UNWINDLIB = ""
```
So compiling with clang is not a problem, but the linking stage fails. I get undefined references to the functions in the binaries.

In the past, I used OpenADK, and there I found a way using a toolchain file to use clang as a compiler, and then in the linking stage gcc. There I used this as the CMAKE_EXE_LINKER_FLAGS, CMAKE_MODULE_LINKER_FLAGS, CMAKE_SHARED_LINKER_FLAGS and CMAKE_STATIC_LINKER_FLAGS:
```
-fuse-ld=bfd -stdlib=libstdc++ -L "toolchain_raspberry-pi3p_glibc_cortex_a53_hard_eabihf_thumb/usr/lib/gcc/arm-openadk-linux-gnueabihf/7.3.0"  --gcc-toolchain=toolchain_raspberry-pi3p_glibc_cortex_a53_hard_eabihf_thumb/usr
```
I would like to use these linker flags too, but in the yocto context. So I need to know some way to set these flags, and also find where the equivalent gcc toolchain directories are located.

So actually the feature request is: compile with clang, but link with gcc using the --gcc-toolchain flag.
I'm going to search for it myself, but I thought, probably you know this information already :) Thanks for any pointers in the right direction!
Best regards,

kraj, 2020-02-12:
in master we already have things to choose runtimes ( gcc orr clang ) `RUNTIME ??= "gnu"` and `RUNTIME ??= "clang"` and gnu is default.

deblauwetom, 2020-02-12:
Thanks for the reply!
I am unfortunately stuck on warrior :(
The only error I get is actually the file "crtbeginS.o" and "crtendS.o" that are not found on the linker commandline. 
I managed to add this:
```
LDFLAGS_append_toolchain-clang = " -fuse-ld=bfd --gcc-toolchain=${STAGING_DIR_NATIVE}/usr -Wl,-L,${STAGING_DIR_TARGET}/lib -Wl,-rpath,${STAGING_DIR_TARGET}/usr/lib/aarch64-poky-linux/7.2.1 -v -Wl,--verbose"
```
So something is only adding the filename, and not the full path to these 2 files :(
Other .o files like "crtn.o" are also added to this commandline, but for some reason, the full path is used there...

deblauwetom, 2020-02-12:
Just found it, this works:
```
LDFLAGS_append_toolchain-clang = " -fuse-ld=lld --gcc-toolchain=${STAGING_DIR_NATIVE}/usr -Wl,-L,${STAGING_DIR_TARGET}/lib -Wl,-L,${STAGING_DIR_TARGET}/usr/lib/aarch64-poky-linux/7.2.1 -B${STAGING_DIR_TARGET}/usr/lib/aarch64-poky-linux/7.2.1"
```

closed 2020-02-12 by deblauwetom