kraj/meta-clang #287: bpftrace building fails with Clang 10.0.0

Leo-Yan, 2020-02-26:
**Describe the bug**
I also found the bpftrace building fails with Clang 10.0.0.  bpftrace added support Clang 9.0 a few days ago but has not supported Clang 10.0.0 yet.  So before bpftrace support Clang 10.0.0, please consider to add it into blacklist.

I will send a patch for OE building after bpftrace get ready for Clang 10.   

**To Reproduce**
add meta-clang into layer
bitbake bpftrace

kraj, 2020-02-26:
Thank you @Leo-Yan

kraj, 2020-02-27:
fails on musl as well
https://errors.yoctoproject.org/Errors/Details/392792/

Leo-Yan, 2020-03-04:
> fails on musl as well
> https://errors.yoctoproject.org/Errors/Details/392792/

Have sent PR to mark musl incompatible: https://github.com/kraj/meta-clang/pull/292.

Leo-Yan, 2020-03-13:
I have verified the latest meta-clang, bpftrace can build successfully with Clang 10.  So can we close this ticket?

P.s. I was committing patch to update git commit for bpftrace but recognized @ovpanait has committed patch, thanks for the patches!

kraj, 2020-03-13:
Sure go ahead

closed 2020-03-13 by Leo-Yan