kraj/meta-clang #290: I would like to introduce BOLT

Lorac, 2020-03-01:
**Describe the solution you'd like**
I would like to be able to use https://github.com/facebookincubator/BOLT, but I'm not sure that meta-clang is the right layer for it.

If poky(oe-core) is the correct layer, will they accept a recipe without a tagged commit yet? It's still in heavy development

I have started a recipe on my end(inside meta-clang), but I have a couple problems with the fact that it need the llvm source to build. I need to clone it to the llvm/tools folder. I was able to clone the bolt repository to the correct directory, but `work-shared/llvm-project-source-9.0.1-r0/git` as been replace solely by the BOLT repository and lost all of the llvm repository. I tried to patch meta-clang layer to use `rev=SHA1` instead of the SRC_REV, to be able to clone two repository while using `require clang.inc`

Inside poky I was not able to include the cmake files required to configure BOLT correctly. It needs `add_llvm_install_targets` from https://github.com/llvm/llvm-project/blob/50a6d3a6486d81d21a2c31ce8522321e19bed35e/llvm/cmake/modules/AddLLVM.cmake#L649

Lorac, 2020-03-02:
I'm rethinking my strategy and I think I need an `llvm.inc` from poky and then require it inside a  bolt recipe with having both repository using `rev` from the SRC_URI. I think you are the maintainer of the llvm recipe on the oe-core side. I could send an RFC patch on this. What do you think of this approach?

kraj, 2020-03-02:
yeah I think its good to maintain it in your own layer and use abstraction from meta-clang, if this is relatively interesting then I am open to accept it into meta-clang.

Lorac, 2020-03-02:
@kraj Thanks for the feedback, but I'll have to rethink on how to use it in yocto, because it is based off llvm 7.0...which is about 150k commits behind from master.