kraj/meta-clang #297: nativesdk-clang fails to build on top of buildtools-extended-tarball

jpuhlman, 2020-03-09:
Here is example of the compile line:

`/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work/x86_64-nativesdk-montavistasdk-linux/nativesdk-clang/10.0.0-r0/recipe-sysroot-native/usr/bin/clang++  -DGTEST_HAS_RTTI=0 -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Ilib/Support -I/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/lib/Support -Iinclude -I/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include -isystem/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work/x86_64-nativesdk-montavistasdk-linux/nativesdk-clang/10.0.0-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -DNDEBUG   -std=c++14  -fno-exceptions -fno-rtti -MD -MT lib/Support/CMakeFiles/LLVMSupport.dir/ARMTargetParser.cpp.o -MF lib/Support/CMakeFiles/LLVMSupport.dir/ARMTargetParser.cpp.o.d -o lib/Support/CMakeFiles/LLVMSupport.dir/ARMTargetParser.cpp.o -c /home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/lib/Support/ARMTargetParser.cpp`

Here are the kind of errors I am seeing:
`/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:639:28: error: no template named 'index_sequence' in namespace 'std'
  iterator begin_impl(std::index_sequence<Ns...>) const {
                      ~~~~~^
/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:642:50: error: no template named 'index_sequence' in namespace 'std'
  template <size_t... Ns> iterator end_impl(std::index_sequence<Ns...>) const {
                                            ~~~~~^
/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:650:28: error: no member named 'index_sequence_for' in namespace 'std'
    return begin_impl(std::index_sequence_for<Args...>{});
                      ~~~~~^
/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:650:47: error: 'Args' does not refer to a value
    return begin_impl(std::index_sequence_for<Args...>{});
                                              ^
/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:626:60: note: declared here
template <template <typename...> class ItType, typename... Args> class zippy {
                                                           ^
/home/jp/builds/clean/opencgx-x86-generic-64/project/tmp/work-shared/llvm-project-source-10.0.0-r0/git/llvm/include/llvm/ADT/STLExtras.h:652:47: error: no member named 'index_sequence_for' in namespace 'std'
  iterator end() const { return end_impl(std::index_sequence_for<Args...>{}); }
`

jpuhlman, 2020-03-09:
I am not entirely sure how to correct the issue.

kraj, 2020-03-10:
@jpuhlman this seems to be because the gcc in play is not passing `-std=c++14`

kraj, 2020-03-11:
Can you try using `-gcc-toolchain /path/to/your/gcc/install `--target=<tuple-with-pooky-sdk>`

jpuhlman, 2020-03-11:
Adding both resolves the individual build issue without adding the unknown links. Have not tried it in the larger build.

kraj, 2020-04-02:
@jpuhlman is this resolved ?

jpuhlman, 2020-04-02:
No. its still an issue. Using the extra options worked for building the individual files, but when I tried to put it in the recipe, it failed in another section then I got sidetracked with oe-core bugs. The initial bug was really more about me trying to figure out if it was something that needed to be fixed in oe-core or something in the nativesdk-clang build. Given the back and forth, I am pretty confident if there is a change it is going to need to be in the nativesdk clang build.

I might be able to get back to it next week. The basic jist is you install the build-extended-tarball and use that as your environment, then build nativesdk-clang.

jpuhlman, 2020-04-02:
Everything else seems to build, its just nativesdk that fails.