kraj/meta-clang #313: clang: error: the clang compiler does not support '-mtune=cortex-a57.cortex-a53'

msisov, 2020-04-22:
Hi Kraj,

I have a problem with compiling clang on Zeus branch for m3ulcb board (I didn't try other boards).

Please see the log bellow. I tried both zeus and master branches of meta-clang.

```
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is Clang 10.0.0
| -- The CXX compiler identification is Clang 10.0.0
| -- The ASM compiler identification is Clang
| -- Found assembler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
| -- Check for working C compiler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
| -- Check for working C compiler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang -- broken
| CMake Error at /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
|   The C compiler
| 
|     "/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command(s):ninja cmTC_f917f && [1/2] Building C object CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o
|     FAILED: CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o
|     /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang   -march=armv8-a -mtune=cortex-a57.cortex-a53   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0=/usr/src/debug/compiler-rt/10.0.0-r0                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0=/usr/src/debug/compiler-rt/10.0.0-r0                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native=   -march=armv8-a -mtune=cortex-a57.cortex-a53   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot -o CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o   -c testCCompiler.c
|     clang-10: error: the clang compiler does not support '-mtune=cortex-a57.cortex-a53'
|     ninja: build stopped: subcommand failed.
| 
| 
| 
| 
| 
|   CMake will not be able to correctly generate this project.
| Call Stack (most recent call first):
|   CMakeLists.txt:49 (project)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/temp/run.do_configure.10238' failed with exit code 1:
| -- The C compiler identification is Clang 10.0.0
| -- The CXX compiler identification is Clang 10.0.0
| -- The ASM compiler identification is Clang
| -- Found assembler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
| -- Check for working C compiler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
| -- Check for working C compiler: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang -- broken
| CMake Error at /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
|   The C compiler
| 
|     "/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command(s):ninja cmTC_f917f && [1/2] Building C object CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o
|     FAILED: CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o
|     /home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang   -march=armv8-a -mtune=cortex-a57.cortex-a53   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0=/usr/src/debug/compiler-rt/10.0.0-r0                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0=/usr/src/debug/compiler-rt/10.0.0-r0                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot-native=   -march=armv8-a -mtune=cortex-a57.cortex-a53   -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/msisov/code/trying/builds/ci-chromium-wayland-m3ulcb-clang-zeus/tmp/work/aarch64-poky-linux/compiler-rt/10.0.0-r0/recipe-sysroot -o CMakeFiles/cmTC_f917f.dir/testCCompiler.c.o   -c testCCompiler.c
|     clang-10: error: the clang compiler does not support '-mtune=cortex-a57.cortex-a53'
|     ninja: build stopped: subcommand failed.
| 
| 
| 
| 
| 
|   CMake will not be able to correctly generate this project.
| Call Stack (most recent call first):
|   CMakeLists.txt:49 (project)
| 
| 
| -- Configuring incomplete, errors occurred!
[... cut ...]
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/msisov/code/trying/sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 586 tasks of which 548 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/msisov/code/trying/sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure
Summary: There were 3 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

kraj, 2020-04-22:
mtune=cortex-a57.cortex-a53 is not supported by clang and I see these tunes were added lately into oe-core and seems to be wrongly using -mtune/-march they should infact be using some appropriate value for -mcpu instead.

kraj, 2020-04-23:
@msisov I would suggest to use master which is currently what is being used for dunfell ( 3.1 ) and will be branched after couple of months, just want to make sure all issues are addressed. dunfell is LTS and anything before that should become low priority.

msisov, 2020-04-27:
@kraj thanks, I will try.

msisov, 2020-07-24:
Sorry, forgot to close. It was resolved.

closed 2020-07-24 by msisov