kraj/meta-clang #332: pthread_exit() crashes on ARM

ryb-ableton, 2020-06-08:
## Description

`pthread_exit()` crashes when called from C++ code compiled with clang on ARM.

## Steps to Reproduce

1. Build an ARM system and run it in qemu:

```
# Install packages needed on Ubuntu
sudo apt-get install gawk wget git-core diffstat unzip texinfo gcc-multilib \
  build-essential chrpath socat libsdl1.2-dev xterm

git clone git://github.com/openembedded/openembedded-core.git
cd openembedded-core
git clone git://github.com/openembedded/bitbake.git
git clone git://github.com/kraj/meta-clang.git

. ./oe-init-build-env
bitbake-layers add-layer ../meta-clang

echo TOOLCHAIN = \"clang\" >> conf/local.conf
echo CLANGSDK = \"1\" >> conf/local.conf
echo IMAGE_INSTALL_append = \" clang \" >> conf/local.conf
echo EXTRA_IMAGE_FEATURES += \"tools-sdk dbg-pkgs tools-debug\" >> conf/local.conf

MACHINE=qemuarm bitbake core-image-minimal-dev
MACHINE=qemuarm runqemu nographic

# Login as root. The password is blank.
```

2. Create a file named `pthread_exit_test.cpp`:

```C++
#include <pthread.h>

void* thread(void*)
{
  pthread_exit(NULL);
  return NULL;
}

int main()
{
  pthread_t tid;
  pthread_create(&tid, NULL, &thread, NULL);
  pthread_join(tid, NULL);
  return 0;
}
```

3. Build the test cpp:

```
clang++ -lpthread -o pthread_exit_test pthread_exit_test.cpp
```

4. Run the executable in gcc:

```
gdb ./pthread_exit_test
(gdb) run
```

## Expected Behavior

The test program exits cleanly.

## Actual Behavior

A segmentation fault with the following backtrace:

```
#0  0xb6f07668 in __unw_get_proc_info (cursor=0xb6d4dc08, info=<optimized out>) at openembedded-core/build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/libunwind/src/libunwind.cpp:173
#1  0xb6f0698e in _Unwind_GetLanguageSpecificData (context=<optimized out>) at openembedded-core/build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/libunwind/src/Unwind-EHABI.cpp:731
#2  (anonymous namespace)::unwindOneFrame (state=<optimized out>, ucbp=0xb6d4e6a8, context=<optimized out>) at openembedded-core/build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/libunwind/src/Unwind-EHABI.cpp:175
#3  __aeabi_unwind_cpp_pr1 (state=<optimized out>, ucbp=0xb6d4e6a8, context=0xb6d4dc08) at openembedded-core/build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/libunwind/src/Unwind-EHABI.cpp:430
#4  0xb6d789c0 in unwind_phase2_forced (ucbp=0xb6d4e6a8, entry_vrs=<optimized out>, resuming=0) at ../../../../../../../work-shared/gcc-10.1.0-r0/gcc-10.1.0/libgcc/unwind-arm-common.inc:524
#5  0xb6d79352 in _Unwind_ForcedUnwind () at ../../../../../../../work-shared/gcc-10.1.0-r0/gcc-10.1.0/libgcc/config/arm/libunwind.S:359
#6  0xb6fbeca2 in __GI___pthread_unwind (buf=<optimized out>) at unwind.c:121
#7  0xb6fb787c in __do_cancel () at pthreadP.h:309
#8  __pthread_exit (value=<optimized out>) at pthread_exit.c:28
#9  0x004006d8 in thread(void*) ()
#10 0xb6fb6946 in start_thread (arg=0x6f85cd8b) at pthread_create.c:477
```

## Cause

`pthread_exit` calls `_Unwind_ForcedUnwind` from `libgcc_s` by [looking up the symbol with `dlsym`](https://github.com/bminor/glibc/blob/5f72f9800b250410cad3abfeeb09469ef12b2438/sysdeps/nptl/unwind-forcedunwind.c#L53...L58). This subverts the normal lookup order, which places `libc++abi` before `libgcc_s`. The crash occurs when `libgcc_s` calls `__aeabi_unwind_cpp_pr1` in LLVM's `libunwind` rather than using its own symbol.

The crash doesn't occur on x86.

## Workarounds

1. Link against `libgcc_s` explicitly so that it's placed before `libc++abi` in the search order:

```
clang++ -lgcc_s -lpthread -o pthread_exit_test pthread_exit_test.cpp
```

```
$ ldd ./pthread_exit_test

linux-vdso.so.1 (0xbeb11000)
libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6f51000)
libpthread.so.0 => /lib/libpthread.so.0 (0xb6f2c000)
libc++.so.1 => /usr/lib/libc++.so.1 (0xb6e9d000)
libc++abi.so.1 => /usr/lib/libc++abi.so.1 (0xb6e62000)
...
```

2. Build the system image without LLVM's `libunwind` and `compiler-rt`:

```
echo PACKAGECONFIG_remove_pn-libcxx=\"unwind compiler-rt\" >> conf/local.conf
echo PACKAGECONFIG_remove_pn-clang=\"compiler-rt\" >> conf/local.conf
```

kraj, 2020-06-23:
can you try using `clang++ -pthread ` instead of `clang++ -lpthread `

ryb-ableton, 2020-07-02:
@kraj I’m on vacation now but I’ll give it a try when I’m back.

kraj, 2020-07-02:
> @kraj I’m on vacation now but I’ll give it a try when I’m back.

no worries. Enjoy your time off.

ryb-ableton, 2020-07-29:
@kraj `clang++ -pthread` and `clang++ -lpthread` create binaries that crash in the same way.

I built a VM using the reproduction steps for easy experimentation: [`core-image-minimal-dev-qemuarm.tar.bz2`](https://www.dropbox.com/s/mpor6u8ea84h589/core-image-minimal-dev-qemuarm.tar.bz2?dl=0)

Run it using:

```
qemu-system-arm \
  -cpu cortex-a15 \
  -m 256 \
  -drive id=disk0,file=core-image-minimal-dev-qemuarm-20200728174122.rootfs.ext4,if=none,format=raw \
  -device virtio-blk-device,drive=disk0 \
  -machine virt,highmem=off \
  -kernel zImage--5.4.51+git0+caafbdfe38_601e67d372-r0-qemuarm-20200728151324.bin \
  -append "root=/dev/vda rootfstype=ext4 fsck.repair=yes rootwait" \
  -nographic
```

kraj, 2020-07-30:
@ryb-ableton I am able to reproduce this here on qemuarm, your analysis is correct, currently, we use gcc runtime be used during build ( -stdlib=libstdc++ -rtlib=libgcc ) where as when we use clang on target we use llvm-runtime ( -stdlib=libc++ -rtlib=compiler-rt )

threfore when building libcxx libraries during cross build we are linking in libgcc unwinder into them dynamically ( you can see that libgcc_s.so is in DT_NEEDED section of both libc++.so and libc++abi.so) and we also statically link llvm libunwind into libc++abi which then gets preferred over libgcc due to order. 

If you were to use `clang++ -pthread -stdlib=libstdc++ -rtlib=libgcc`  to compile then everything will work out fine.

So I think what we can do is that default runtime remains same for target clang as it was during cross compile, this will ensure that behavior is consistent, we will still support the other compiler runtime but it will mean using `-stdlib` and `-rtlib` and `-unwindlib` to select that explicitly on compiler cmdline.

secondly, we can also look at if we can leave libunwind out of libc++abi by not static linking it

ryb-ableton, 2020-07-31:
@kraj It'd be interesting to try to reproduce this on a stock ARM32 Linux system, independent of OpenEmbedded and meta-clang, by building clang from scratch to use compiler-rt and libunwind. If it can, it's an issue for the LLVM bug tracker. Unfortunately I can't spend more time on this at the moment.

Using the GCC runtime by default sounds like a good solution in any case, since it's more widely used and tested with clang on Linux.

kraj, 2020-07-31:
@ryb-ableton I think static linking in libunwind into libc++abi is not right and its meta-clang specific, you won't see that behavior with vanilla libcxx, perhaps I will take a dig at fixing that

kraj, 2020-08-02:
should be fixed in master now

closed 2020-08-02 by kraj

fsa-ableton, 2020-08-24:
> should be fixed in master now

Can this be backported to the `zeus` branch as well?

kraj, 2020-08-24:
@fsa-ableton test it out on zeus and report back if it works for you

fsa-ableton, 2020-08-26:
> @fsa-ableton test it out on zeus and report back if it works for you

I backported it to `zeus` successfully :-)

kraj, 2020-08-26:
OK cool. Care to propose PR as well ?

fsa-ableton, 2020-08-27:
Can you please grant me push rights to the repo?

I have a branch prepared on top of your `zeus` branch; however, I tested the fix on our `zeus` branch, which is behind on 81ba160 (`conf/nonclangable.conf: Always build mesa with gcc`). I don't think that it makes a difference, but just saying.

kraj, 2020-08-27:
you can create pull request https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request