kraj/meta-clang #335: Yocto dunfell with aarch64/armv7 multilib: clang TARGET_PREFIX$tool symlinks not present in target image nor installed SDK

rmi1974, 2020-06-16:
I've created a Yocto SDK based on Dunfell with full multilib support for target image and SDK.
Clang is enabled to be included in target image and SDK. Gcc is still the default compiler.

```
CLANGSDK = "1"
IMAGE_INSTALL_append = " clang" 
```

Target image and SDK creation works fine. Multiple multilib fixes are needed tho but this not related to meta-clang layer.

```
git clone -b dunfell git://git.yoctoproject.org/poky.git
cd poky
git clone -b dunfell git://git.openembedded.org/meta-openembedded.git
git clone -b master git://github.com/kraj/meta-clang.git
...
additonal layers
```

It seems for whatever reasons the Yocto SDK toolchain which includes clang doesn't have TARGET_PREFIX$tool symlinks in the installed SDK host sysroot nor in target image.

https://github.com/kraj/meta-clang/blob/master/recipes-core/meta/meta-environment.bbappend

```
create_sdk_files_append() {
        script=${SDK_OUTPUT}/${SDKPATH}/environment-setup-${REAL_MULTIMACH_TARGET_SYS}
        echo 'export CLANGCC="${TARGET_PREFIX}clang ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
        echo 'export CLANGCXX="${TARGET_PREFIX}clang++ ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
        echo 'export CLANGCPP="${TARGET_PREFIX}clang -E ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
        echo 'export CLANG_TIDY_EXE="${TARGET_PREFIX}clang-tidy ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
}
```

The resulting env scripts:

armv7:

```
export CC="arm-pokymllib32-linux-gnueabi-gcc  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT"
export CXX="arm-pokymllib32-linux-gnueabi-g++  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT"
export CPP="arm-pokymllib32-linux-gnueabi-gcc -E  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT"
...
export CLANGCC="arm-pokymllib32-linux-gnueabi-clang  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANGCXX="arm-pokymllib32-linux-gnueabi-clang++  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANGCPP="arm-pokymllib32-linux-gnueabi-clang -E  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANG_TIDY_EXE="arm-pokymllib32-linux-gnueabi-clang-tidy  -march=armv7-a -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT" 
```

aarch64:

```
export CC="aarch64-poky-linux-gcc  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT"
export CXX="aarch64-poky-linux-g++  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT"
export CPP="aarch64-poky-linux-gcc -E  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=$SDKTARGETSYSROOT" 
...
export CLANGCC="aarch64-poky-linux-clang  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANGCXX="aarch64-poky-linux-clang++  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANGCPP="aarch64-poky-linux-clang -E  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT"
export CLANG_TIDY_EXE="aarch64-poky-linux-clang-tidy  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=$SDKTARGETSYSROOT" 
```

From my SDK install:

```
$ pwd
/opt/winedev/3.1.1/sysroots/x86_64-pokysdk-linux/usr/bin

$ ll
lrwxrwxrwx. 1 focht focht         8 Jun 16 08:50 2to3 -> 2to3-3.8
-rwxr-xr-x. 1 focht focht       143 Jun 16 08:50 2to3-3.8
[... 73 lines cut ...]
./clang  -mcpu=cortex-a53+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=/opt/winedev/3.1.1/sysroots/aarch64-poky-linux -v
clang version 10.0.1 (https://github.com/llvm/llvm-project a634a80615b1e012f1a61aa0cd1e2e67ef77d0bd)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /opt/winedev/3.1.1/sysroots/x86_64-pokysdk-linux/usr/bin
clang-10: warning: argument unused during compilation: '-mcpu=cortex-a53+crc' [-Wunused-command-line-argument]
clang-10: warning: argument unused during compilation: '-fstack-protector-strong' [-Wunused-command-line-argument]
clang-10: warning: argument unused during compilation: '-D _FORTIFY_SOURCE=2' [-Wunused-command-line-argument]
clang-10: warning: argument unused during compilation: '-Wformat' [-Wunused-command-line-argument]
clang-10: warning: argument unused during compilation: '-Wformat-security' [-Wunused-command-line-argument]
clang-10: warning: argument unused during compilation: '-Werror=format-security' [-Wunused-command-line-argument]
...
```

I remember this worked in the past. I had a thud based multilib-Yocto working with clang from meta-clang two years ago (required lots of fixups). Revived my project now with much newer Linux host (FC32, host gcc 10) and latest Yocto release.

----

I also suffer from the SDK relocation problem, hence installed in default `/opt`.

https://github.com/kraj/meta-clang/issues/119

----

Minor: If clang is not included in the SDK (`CLANGSDK=0` default),  https://github.com/kraj/meta-clang/blob/master/recipes-core/meta/meta-environment.bbappend will unconditionally append `export CLANGxxx=yyy`. This results in variables that refer to non-existing tools. Maybe it could be made conditional.

kraj, 2020-06-23:
@rmi1974 thanks for the detailed report. You report 2 issues here, and first one seems a regression and second one about not creating append for `create_sdk_files` when `CLANGSDK` is not set is also good one to fix.

rmi1974, 2020-06-24:
Thanks. I've split the second issue into a new ticket here: https://github.com/kraj/meta-clang/issues/338
Shame on me. I'm usually a proponent of proper bug reporting.

kraj, 2020-06-25:
I can see 
```
% find /opt/yoe/3.1/sysroots/x86_64-yoesdk-linux/ -name "*-clang"
/opt/yoe/3.1/sysroots/x86_64-yoesdk-linux/usr/bin/aarch64-yoe-linux/aarch64-yoe-linux-clang
/opt/yoe/3.1/sysroots/x86_64-yoesdk-linux/usr/bin/aarch64-yoe-linux-musl/aarch64-yoe-linux-musl-clang

```

and CLANGCC and CLANGCXX vars are set correctly too.

and I can compile hello.c with CLANGCC as well.

How are you cross compiling with clang ?

rmi1974, 2020-07-01:
Hmpf ... I've updated my workspace with most recent heads of Yocto/Poky dunfell branch + meta-clang master and rebuilt the whole thing. I can't reproduce the problem .. no idea what's different now.

Cross-compiling armv7 and aarch64 Wine with multi-lib SDK's Clang 10.x works now.

```
pwd
/opt/winedev/3.1.1

find . -name "*-clang"
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-musl/arm-poky-linux-musl-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-pokymllib32-linux-musl/arm-pokymllib32-linux-musl-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-pokymllib32-linux-musleabi/arm-pokymllib32-linux-musleabi-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-musleabi/arm-poky-linux-musleabi-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-pokymllib32-linux/aarch64-pokymllib32-linux-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-pokymllib32-linux-musl/aarch64-pokymllib32-linux-musl-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-pokymllib32-linux/arm-pokymllib32-linux-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux-musl/aarch64-poky-linux-musl-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-poky-linux/arm-poky-linux-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/arm-pokymllib32-linux-gnueabi/arm-pokymllib32-linux-gnueabi-clang
./sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang
```

Background of my setup: I'm using Poky SDK for building/testing/debugging bi-arch Wine for ARM.

Without Poky, my compiler test-suite is:

Intel x86_x64 (bi-arch):

* GCC 10.1.1 from Fedora 32 host for building x86 32-bit and 64-bit Wine 
* Clang 10.0.0 from Fedora 32 host for building x86 32-bit and 64-bit Wine
* LLVM MinGW 10.0 from: https://github.com/mstorsjo/llvm-mingw for building x86 32-bit and 64-bit Wine (PE + PDB support)

ARM (bi-arch):

Yocto SDK/Poky toolchain

* GCC 9.3.0 for building ARMv7 hard-float Wine
* Clang 10.0.1 for building ARMv7 hard-float Wine
* Clang 10.0.1 for AARCH64 Wine (requirement for msabi/ varargs)

I will investigate how Martin's llvm-mingw can be used for cross-compiling ARM targets as well. I'm not sure how llvm-mingw toolchains integrate with ARM development sysroot (libs/dependencies to target userland) that Poky provides.
Either integrate llvm-mingw in a Yocto/Poky SDK as well or somehow use an existing development sysroot from Poky?

---

I propose to close the ticket for now... non-reproducible.

kraj, 2020-07-01:
thanks a bunch @rmi1974

closed 2020-07-01 by kraj