kraj/meta-clang #336: dunfell branch

priv-kweihmann, 2020-06-21:
After master and dunfell already starting to diverge (at least for poky and meta-oe) it would be nice to have a dedicated branch for dunfell here as well

kraj, 2020-06-22:
@priv-kweihmann good suggestion. I was thinking for waiting till 10.0.1 is out. Which should be only a week or so.

priv-kweihmann, 2020-07-10:
@kraj any news? In combination with the busybox append (ee1dcdfbb8d31f168b26c2e3e5be97efea6537df) the layer currently breaks a lot of builds for me - so any relief is highly appreciated.

kraj, 2020-07-10:
yeah unfortunately 10.0.1 went into rc4 and is still not done. I think I will branch now and do a merge when that happens into dunfell.

kraj, 2020-07-10:
dunfell branch is available now

closed 2020-07-10 by kraj

priv-kweihmann, 2020-07-10:
Thanks a lot - could you then also remove the busybox append in master?

kraj, 2020-07-10:
> Thanks a lot - could you then also remove the busybox append in master?

it cant be removed but needed porting which I have already proposed here
https://github.com/kraj/meta-clang/pull/345