kraj/meta-clang #346: bpftrace on thud

zinosat, 2020-07-15:
I'm trying to include bpftrace in an image based on thud: I stumbled in this repo which includes a recipe for bpftrace, not in the thud branch.

I've tried the following:
 - add meta-clang in the thud branch
 - copy the recipes for bcc and bpftrace in custom recipes-devtools

bitbake bpftrace

after a couple of hours I get the errors at the end of the following message.
I'm not too fluent with yocto, so I'd like to know if I have any chances to get bpftrace included in the image.

Thank you in advance,
Davide

ERROR: bcc-0.13.0-r0 do_configure: Function failed: do_configure (log file is located at /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/temp/log.do_configure.11107)
ERROR: Logfile of failure stored in: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/temp/log.do_configure.11107
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 8.2.0
| -- The CXX compiler identification is GNU 8.2.0
| -- Check for working C compiler: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-gcc
| -- Check for working C compiler: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-gcc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++
| -- Check for working CXX compiler: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++ -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Latest recognized Git tag is GIT-NOTFOUND
| -- Git HEAD is 942227484d3207f6a42103674001ef01fb5335a0
| -- Revision is IT-NOTFOUND-94222748
| -- Performing Test HAVE_NO_PIE_FLAG
| -- Performing Test HAVE_NO_PIE_FLAG - Success
| -- Performing Test HAVE_REALLOCARRAY_SUPPORT
| -- Performing Test HAVE_REALLOCARRAY_SUPPORT - Success
| -- Found BISON: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/bison (found version "3.0.4")
| -- Found FLEX: /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/flex (found version "2.6.0")
| CMake Error at /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/lib/cmake/llvm/LLVMExports.cmake:662 (message):
|   The imported target "llvm-tblgen" references the file
|
|      "/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/bin/llvm-tblgen"
|
|   but this file does not exist.  Possible reasons include:
|
|   * The file was deleted, renamed, or moved to another location.
|
|   * An install or uninstall procedure did not complete successfully.
|
|   * The installation package was faulty and contained
|
|      "/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/lib/cmake/llvm/LLVMExports.cmake"
|
|   but not all the files it references.
|
| Call Stack (most recent call first):
|   /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/lib/cmake/llvm/LLVMConfig.cmake:150 (include)
|   CMakeLists.txt:42 (find_package)
|
|
| -- Configuring incomplete, errors occurred!
| See also "/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/build/CMakeFiles/CMakeOutput.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/temp/log.do_configure.11107)
ERROR: Task (/home/zino/code/ghs/YOCTO/wd/poky/meta-sirti/recipes-devtools/bcc/bcc_0.13.0.bb:do_configure) failed with exit code '1'

kraj, 2020-07-15:
I think we only have this working with dunfell and newer branches

ovpanait, 2020-07-16:
Hi @zinosat 

You need to backport this clang patch from master to fix the above mentioned error:

https://github.com/kraj/meta-clang/commit/ab686950af8d7f02e0ca614d277c7c0ca36954ed

zinosat, 2020-07-16:
I've tried to [backport the patch](https://pastebin.com/vEXEyVJ1) you've suggested; now I get [a failure](https://pastebin.com/uHMX1YKv) while compiling bcc. It's a big log, where the first error is

| FAILED: src/cc/frontends/b/CMakeFiles/b_frontend.dir/loader.cc.o                                                                                                                                                                        | /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-g++   -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/include/../tools/clang/include -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src -Isrc -Isrc/cc -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc -Isrc/cc/frontends/b -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/b -I/home/zino/code/ghs/YOCTO/wd/p
oky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/clang -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot/usr/include -I/home/zino/code/ghs/YO
CTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/libbpf/include -I/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/libbpf/include/uapi -m64 -march=n
ehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/
0.13.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0=/usr/src/debug/bcc/0.13.0-r0 -fdebug-prefix-map=/home
/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot= -fdebug-prefix-map=/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot-nativ
e=  -fvisibility-inlines-hidden -m64 -march=nehalem -mtune=generic -mfpmath=sse -msse4.2 -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/zino/code/ghs/YOCTO/wd/poky/bui
ld-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/recipe-sysroot -Wall  -fno-rtti -fPIC -DBCC_PROG_TAG_DIR='"/var/tmp/bcc"' -Wno-unused-result -DLLVM_MAJOR_VERSION=7 -DNDEBUG   -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORM
AT_MACROS -D__STDC_LIMIT_MACROS -std=gnu++14 -MD -MT src/cc/frontends/b/CMakeFiles/b_frontend.dir/loader.cc.o -MF src/cc/frontends/b/CMakeFiles/b_frontend.dir/loader.cc.o.d -o src/cc/frontends/b/CMakeFiles/b_frontend.dir/loader.cc.o -
c /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/b/loader.cc                                                                                                            
| In file included from /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/b/parser.h:21,                                                                                   
|                  from /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/b/loader.cc:17:                                                                                  
| /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.13.0-r0/git/src/cc/frontends/b/lexer.h:22:10: fatal error: FlexLexer.h: No such file or directory                                                   
|  #include <FlexLexer.h>                                                                                                                                                                                                                 
|           ^~~~~~~~~~~~~                                                                                                                                                                                                                 
| compilation terminated.

Thanks alot for your help: I thoroughly appreciate it.

ovpanait, 2020-07-16:
You could add "flex" to DEPENDS in the bcc bb file in order to get rid of the "#include <FlexLexer.h>" error.

However, the bcc build will fail afterwards with:
error: 'LLVMInitializeBPFTarget' was not declared in this scope
|    LLVMInitializeBPFTarget();
|    ^~~~~~~~~~~~~~~~~~~~~~~
| /buildarea/raid0/ovidiu/Builds/Yocto/yocto-thud/poky/build/tmp/work/core2-64-poky-linux/bcc/0.13.0-r0/git/src/cc/bpf_module.cc:110:3: note: suggested alternative: 'LLVMInitializeX86Target'
|    LLVMInitializeBPFTarget();
|    ^~~~~~~~~~~~~~~~~~~~~~~
|    LLVMInitializeX86Target
| /buildarea/raid0/ovidiu/Builds/Yocto/yocto-thud/poky/build/tmp/work/core2-64-poky-linux/bcc/0.13.0-r0/git/src/cc/bpf_module.cc:111:3: error: 'LLVMInitializeBPFTargetMC' was not declared in this scope
|    LLVMInitializeBPFTargetMC();
|    ^~~~~~~~~~~~~~~~~~~~~~~~~
| /buildarea/raid0/ovidiu/Builds/Yocto/yocto-thud/poky/build/tmp/work/core2-64-poky-linux/bcc/0.13.0-r0/git/src/cc/bpf_module.cc:111:3: note: suggested alternative: 'LLVMInitializeX86TargetMC'
|    LLVMInitializeBPFTargetMC();
|    ^~~~~~~~~~~~~~~~~~~~~~~~~
|    LLVMInitializeX86TargetMC
| /buildarea/raid0/ovidiu/Builds/Yocto/yocto-thud/poky/build/tmp/work/core2-64-poky-linux/bcc/0.13.0-r0/git/src/cc/bpf_module.cc:112:3: error: 'LLVMInitializeBPFTargetInfo' was not declared in this scope
|    LLVMInitializeBPFTargetInfo();
|    ^~~~~~~~~~~~~~~~~~~~~~~~~~~

I think this happens because there is a mismatch between llvm and bcc (probably bcc is too new to be built with the old llvm). You could try to build an older version of bbc/bpftrace to check if it works with the llvm present in thud.

zinosat, 2020-07-16:
Adding "flex" to DEPENDS definitely fixes the "#include <FlexLexer.h>" error: great!
I've tried to go back in time, starting with bcc: not having a clue about a reasonable version to try, I ended up with the version tagged v0.10.0 (commit ebf0e4b2fd7190e747638f68a4421a8b653bc0a2) and what I get is [a failure](https://pastebin.com/0n26GcZc) which looks caused by some warnings.
Before trying to fix that, it'd be great to know if I'm on the right path.

thank you
Davide

ovpanait, 2020-07-16:
The real error seems to be:
[21/251] cd /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.10.0-r0/build/man/man8 && GZIP-NOTFOUND -c /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.10.0-r0/git/man/man8/llcstat.8 > /home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.10.0-r0/build/man/man8/llcstat.8.gz
FAILED: man/man8/llcstat.8.gz

It tries to archive the manpages but it can't find gzip. I think it can be solved by adding "gzip-native" to DEPENDS.

I am not sure which bcc version is compatible with thud, but I think you are on the right track. You should try to find the highest compatible version.

zinosat, 2020-07-16:
hi @ovpanait 
situation looks bad: I get the same failure for all tagged version  from v0.8.0 to v0.13.0:

/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.10.0-r0/git/src/cc/bpf_module.cc: In constructor 'ebpf::BPFModule::BPFModule(unsigned int, ebpf::TableStorage*, bool, const string&, bool)':
/home/zino/code/ghs/YOCTO/wd/poky/build-release/tmp/work/corei7-64-poky-linux/bcc/0.10.0-r0/git/src/cc/bpf_module.cc:106:3: error: 'LLVMInitializeBPFTarget' was not declared in this scope
   LLVMInitializeBPFTarget();
   ^~~~~~~~~~~~~~~~~~~~~~~

at v0.7.0 I had to exclude all patches and gave up

zinosat, 2020-07-16:
Trying a new approach: I've taken clang from newer warrior branch (from zeus it fails): it builds fine but I need to backport again the suggested  patch. It's going to take a while to complete the build. we'll see

zinosat, 2020-07-17:
Still getting the very same error while compiling bcc (no matter which version I try to build)

ovpanait, 2020-07-18:
> Trying a new approach: I've taken clang from newer warrior branch (from zeus it fails): it builds fine but I need to backport again the suggested patch. It's going to take a while to complete the build. we'll see

Are you getting the same results even with the clang from warrior branch? 

What about the clang from dunfell? Does it compile with thud?

zinosat, 2020-07-18:
Hi @ovpanait ,
yes, bcc fails with clang from warrior, the same way it fails with clang from thud.
I've tried with clang from dunfell and it fails with this dependency error:

NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'libedit-native' (but /home/zino/code/ghs/YOCTO/wd/poky/meta-sirti/recipes-devtools/clang/clang_git.bb DEPENDS on or otherwise requires it). Close matches:
  libxext-native
  libnewt-native
  libgit2-native
ERROR: Required build target 'bcc' has no buildable providers.
Missing or unbuildable dependency chain was: ['bcc', 'clang', 'libedit-native']

ovpanait, 2020-07-19:
I was able to get bpftrace to work (at least basic functionality) on thud with meta-clang master branch using the attached experimental meta-clang patch.

I used the following steps:
git clone -b thud git://git.yoctoproject.org/poky
git clone -b thud  https://github.com/openembedded/meta-openembedded.git
git clone https://github.com/kraj/meta-clang.git
cd meta-clang; git am [thud.tar.gz](https://github.com/kraj/meta-clang/files/4943830/thud.tar.gz); cd ..

local.conf:
MACHINE = "qemux86-64"
IMAGE_INSTALL_append = " bpftrace kernel-devsrc"

bitbake core-image-minimal
runqemu qemux86-64 nographic kvm slirp

Note: this is just for testing bpftrace basic functionality. It is clear  that it was not properly tested on such old environments.  I am not familiar with bpftrace internals, so I am not sure I can assist with runtime problems :)

zinosat, 2020-07-19:
Hi @ovpanait 
you did a fantastic job!

I'm trying to adapt your setup to my environment which uses a custom MACHINE:  adding the new "meta-clang" layer triggers a rebuild of the whole bunch of packages which completes correctly.
If I run "bitbake bpftrace" i get

ERROR: Nothing PROVIDES 'libedit-native' (but /home/zino/code/ghs/YOCTO/wd/poky/meta-clang/recipes-devtools/clang/clang_git.bb DEPENDS on or otherwise requires it). Close matches:
  libxext-native
  libnewt-native
  libgit2-native
ERROR: Required build target 'bcc' has no buildable providers.
Missing or unbuildable dependency chain was: ['bcc', 'clang', 'libedit-native']

zinosat, 2020-07-19:
as a test, I've tried to comment out the following line from clang_git.bb 

PACKAGECONFIG[libedit] = "-DLLVM_ENABLE_LIBEDIT=ON -DLLDB_ENABLE_LIBEDIT=ON,-DLLVM_ENABLE_LIBEDIT=OFF -DLLDB_ENABLE_LIBEDIT=OFF,libedit libedit-native"

compilation goes past the previous error: let's see if I get ti to build

ovpanait, 2020-07-20:
It seems your meta-openembedded layer is missing the following patch: 
https://github.com/openembedded/meta-openembedded/commit/934b8f6dd5f6681daed6490b229b8ba755aacc8e

You could backport this patch directly, or if you cannot patch the sources, you could create a libedit_%.bbappend in one of your custom layers and add the following to it:

BBCLASSEXTEND = "native nativesdk"

zinosat, 2020-07-20:
Compilation completed and I finally have an image containing bpftrace!

thank you for the patch: will give it a go.

zinosat, 2020-07-20:
Applied the patch to openembedded layer and it works

I own you a few virtual beers:)

Sincerely: thank you a lot for your help!

Davide

closed 2020-07-22 by zinosat

zinosat, 2020-08-07:
I'm attaching an updated version of @ovpanait patch, which fixes compilation ont thud, after the latest commit (edd7e22b0db5bf24ccbdf79557907a66cb615ee0)

[thud.patch.gz](https://github.com/kraj/meta-clang/files/5040641/thud.patch.gz)

kraj, 2020-08-07:
@zinosat we have thud branch for meta-clang, while its not actively maintained I am willing to accept well tested patches. This patch is reviving thud support on master, which we do not want. Master currently supports dunfell and gatesgarth(master) only.