kraj/meta-clang #349: fatal error: 'asm/early_ioremap.h' file not found

zinosat, 2020-07-24:
While running the tools inside /usr/share/bcc/tools provided by bcc package on my "thud" system, I get the following error:

# /usr/share/bcc/tools/cpudist 
In file included from /virtual/main.c:2:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/include/linux/sched.h:20:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/include/linux/hrtimer.h:22:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/include/linux/percpu.h:7:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/include/linux/smp.h:64:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/smp.h:13:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/apic.h:10:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/fixmap.h:29:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/acpi.h:33:
In file included from /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/realmode.h:15:
/lib/modules/4.14.79-rt29-yocto-preempt-rt/build/arch/x86/include/asm/io.h:44:10: fatal error: 'asm/early_ioremap.h' file not found
#include <asm/early_ioremap.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
Traceback (most recent call last):
  File "/usr/share/bcc/tools/cpudist", line 161, in <module>
    b = BPF(text=bpf_text, cflags=["-DMAX_PID=%d" % max_pid])
  File "/usr/lib/python3.5/site-packages/bcc/__init__.py", line 349, in __init__
    raise Exception("Failed to compile BPF module %s" % (src_file or "<text>"))
Exception: Failed to compile BPF module <text>

I've tried the following on my build host:
root@sirtix:~/code/ghs/YOCTO/wd/poky/build-release/tmp/work-shared/mymachine/kernel-source# find -name early_ioremap.h                                                                                               
./include/asm-generic/early_ioremap.h                                                                                                                                                                                    
./arch/ia64/include/asm/early_ioremap.h

thank you,
Davide

kraj, 2020-07-30:
somehow I think you need to add ${STAGING_KERNEL_BUILDDIR} to include search paths when it comes to looking for prepared kernel build tree for kernel headers. Looking into kernel-source is not right place to look for.

zinosat, 2020-08-01:
tools fail to run on the target because of missing 'asm/early_ioremap.h': you suggest to copy the missing file to the target fs taking it from ${STAGING_KERNEL_BUILDDIR}, correct? 
the fix should be part of  meta-clang/dynamic-layers/openembedded-layer/recipes-devtools/bcc/bcc_0.13.0.bb ?

thank you in advance,
Davide

kraj, 2020-08-02:
perhaps yes I think someone familiar with bcc might have better idea

ovpanait, 2020-08-02:
Hi,

You could generate "asm/early_ioremap.h" directly on target, by running "make prepare" in the kernel build directory:

- include "kernel-devsrc" to rootfs
- add "tools-sdk" to "EXTRA_IMAGE_FEATURES" (e.g. EXTRA_IMAGE_FEATURES ?= "debug-tweaks tools-sdk")
- rebuild
- on target:
  
root@qemux86-64:~# cd /lib/modules/4.18.33-yocto-standard/build/
root@qemux86-64:/lib/modules/4.18.33-yocto-standard/build# make prepare
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  YACC    scripts/kconfig/zconf.tab.c
  LEX     scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --syncconfig Kconfig
  SYSTBL  arch/x86/include/generated/asm/syscalls_32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_32_ia32.h
  SYSHDR  arch/x86/include/generated/asm/unistd_64_x32.h
  SYSTBL  arch/x86/include/generated/asm/syscalls_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_32.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_64.h
  SYSHDR  arch/x86/include/generated/uapi/asm/unistd_x32.h
  HOSTCC  arch/x86/tools/relocs_32.o
  HOSTCC  arch/x86/tools/relocs_64.o
  HOSTCC  arch/x86/tools/relocs_common.o
  HOSTLD  arch/x86/tools/relocs
  WRAP    arch/x86/include/generated/uapi/asm/bpf_perf_event.h
  WRAP    arch/x86/include/generated/uapi/asm/poll.h
  WRAP    arch/x86/include/generated/asm/dma-contiguous.h
  WRAP    arch/x86/include/generated/asm/early_ioremap.h
  WRAP    arch/x86/include/generated/asm/mcs_spinlock.h
  WRAP    arch/x86/include/generated/asm/mm-arch-hooks.h
warning: Cannot use CONFIG_STACK_VALIDATION=y, please install libelf-dev, libelf-devel or elfutils-libelf-devel
root@qemux86-64:/lib/modules/4.18.33-yocto-standard/build# 
root@qemux86-64:/lib/modules/4.18.33-yocto-standard/build# 
root@qemux86-64:/lib/modules/4.18.33-yocto-standard/build# /usr/share/bcc/tools/cpudist
Tracing on-CPU time... Hit Ctrl-C to end.
^C
     usecs               : count     distribution
         0 -> 1          : 0        |                                        |
         2 -> 3          : 0        |                                        |
         4 -> 7          : 0        |                                        |
         8 -> 15         : 0        |                                        |
        16 -> 31         : 0        |                                        |
        32 -> 63         : 1        |*                                       |
        64 -> 127        : 0        |                                        |
       128 -> 255        : 1        |*                                       |
       256 -> 511        : 1        |*                                       |
       512 -> 1023       : 1        |*                                       |
      1024 -> 2047       : 1        |*                                       |
      2048 -> 4095       : 0        |                                        |
      4096 -> 8191       : 0        |                                        |
      8192 -> 16383      : 1        |*                                       |
     16384 -> 32767      : 1        |*                                       |
     32768 -> 65535      : 1        |*                                       |
     65536 -> 131071     : 6        |*******                                 |
    131072 -> 262143     : 31       |****************************************|

zinosat, 2020-08-03:
I've managed to get this to work: excellent (and thank you)!

What would it take to get all the necessary in the rootfs without having to install gcc on the running system?

kraj, 2020-08-03:
@zinosat are you looking to compile whole rootfs with clang ? if so set `TOOLCHAIN = 'clang'` in local.conf, if you are asking something else please clarify

zinosat, 2020-08-03:
Hi @kraj , @ovpanait provided a fix which consists in installing gcc into the target filesystem and running "make prepare" always on the taget: once you do that all the bcc tools work fine. I wanted to know what would it take to get the bcc tools to work on a target machine wiithout having to run "make prepare" on the target

ovpanait, 2020-08-04:
Hi,

I am not sure it is currently possible out of the box to include the kernel headers generated at build time to the rootfs. I think the supported flow is to include kernel-devsrc and run "make prepare"/"make scripts" at runtime. You might need to create a separate recipe in your layer to do this process manually, but it may prove difficult/error-prone.

On newer kernels this is not needed because you can enable "CONFIG_IKHEADERS=m" in the kernel config and bcc will pick the kernel headers from sysfs directly. You could also try to backport this kernel module to your older kernel and see if it works.

zinosat, 2020-08-05:
@ovpanait I very much like the kheaders option: I've backported the patch and when I modprobe the module, I see /proc/kheaders.tar.xz  
I still get the original error when I run the tool, but I'm sure I'm almost there!

zinosat, 2020-08-05:
it looks like I have to backport a more recent version of the patch, since now bcc extracts the archive from /sys

zinosat, 2020-08-06:
I've backported a more recent kernel patch.
When i run (for example) _/usr/share/bcc/tools/execsnoop_ I still get the error I originally reported in this ticket.
I've manually tried the following:

    modprobe kheaders
    tar xf /sys/kernel/kheaders.tar.xz -C .
    cp -r arch/ /lib/modules/4.14.79-rt29-yocto-preempt-rt/build/

Now the tool runs fine.
Am I doing anything wrong?
I mean: manually following the 3 steps above is acceptable for debugging purposes, but if it should just work out of the box I'd like to understand why it is not.

ovpanait, 2020-08-06:
What if you remove "kernel-devsrc" from the image? It shouldn't be necessary anymore. Maybe the incomplete headers have a higher priority than kheaders module.

Instead, add "kernel-modules" to rootfs and give it a try.

zinosat, 2020-08-06:
Hi @ovpanait ,
using "kernel-modules" in place of "kernel-devsrc" fixes it: you're my hero :)

thank you,
Davide

closed 2020-08-06 by zinosat