kraj/meta-clang #360: compiler-rt does not build sanitizer libs

sandeepjain26, 2020-08-15:
Hi all,
I had build meta-clang with musl, and found that sanitizer (asam, ..etc) libs are not present in built rootfs.
I looked into the compiler-rt recipe, and looks like build for sanitizer libs are disabled.
I tried building sanitizers with below change removal in compiler-rt recipe.
-EXTRA_OECMAKE_append_libc-musl = " -DCOMPILER_RT_BUILD_SANITIZERS=OFF "

this caused below compile error.
`tmp/work-shared/llvm-project-source-10.0.1-r0/git/compiler-rt/lib/sanitizer_common/sanitizer_platform_limits_posix.cpp:61:10: fatal error: 'fstab.h' file not found
#include <fstab.h>
         ^~~~~~~~~
1 error generated.`
 
Is there a way to build compiler-rt sanitizer libs?
Any suggestion, or pointer towards including sanitizer libs.
Thanks.

Regards,
Sandeep

kraj, 2020-08-16:
@sandeepjain26 yes the reason to disable is that we were not able to build it. We now do have clang11 recipes in flight, so perhaps you can try that, if you are interested 

cherry-pick top 3 patches from https://github.com/YoeDistro/meta-clang/commits/yoe/mut

and let me know if its any better.

sandeepjain26, 2020-08-16:
@kraj , Thanks a lot for the information.
I tried build with branch yoe/mut.
Without sanitizer, build for 11.0.0 goes through.
But once I enabled Sanitizer, Build failed with same error
`tmp/work-shared/llvm-project-source-11.0.0-r0/git/compiler-rt/lib/sanitizer_common/sanitizer_platform_limits_posix.cpp:61:10: fatal error: 'fstab.h' file not found
#include <fstab.h>
         ^~~~~~~~~
1 error generated.`

if I remember correctly, with 10.0.1 there were error in 2 files for fstab.h, but with 11.0.0 now only one error noticed.
Any suggestion to go ahead further?
Thanks a lot.

Regards,
Sandeep

kraj, 2020-08-16:
you might want to cherry-pick needed bits from https://reviews.llvm.org/D63785

sandeepjain26, 2020-08-17:
@kraj , Thanks for pointer.
Patch seems to be very old (more than 1 yr ).
Not sure if its still valid or might require porting as per latest code.
I will give it a try.
Thanks.

sandeepjain26, 2020-08-18:
@kraj 
Since patch is quite old, 
- file names changed from *.cc to *.cpp.
- and hunks failed to apply.
I had worked to fix failing hunks.

At the moment trying to fix below compilation

`
llvm-project-source-11.0.0-r0/git/compiler-rt/lib/sanitizer_common/sanitizer_platform_limits_posix.cpp:1163:1: error: 'assertion_failed__1163' declared as an array with a negative size
| CHECK_TYPE_SIZE(ipc_perm);
| ^~~~~~~~~~~~~~~~~~~~~~~~~
`
Any suggestion?

Thanks.
Regards,
Sandeep

matrackif, 2021-05-18:
@kraj Hello, @sandeepjain26 

I have attempted to turn on sanitizers on the newest master branch with clang 12, it still fails to compile `sanitizer_platform_limits_posix.cpp:156`, with `error: cannot find crypt.h`. 

This is a real shame, as we would love to use sanitizers on the target aarch64 system.

I have noticed that `/usr/include/crypt.h` is located in both the target aarch64 and host sysroot, so I am not sure why it cannot find it. If someone has found a patch or a combination of options to get sanitizers to build I would much appreciate it.

Thank you,
Regards

kraj, 2021-05-19:
@matrackif are you using `musl`

kraj, 2021-05-19:
Please test https://github.com/kraj/meta-clang/pull/464 and let me know if this works for you.

matrackif, 2021-05-19:
I am building the pull request now, I will let you know in around 4-5 hours if it was successful.

matrackif, 2021-05-19:
@kraj 

Great news, your changes in PR #464 made  all of the sanitizer libraries compile. Now I face an issue that is probably due to my lack of knowledge of yocto/bitbake:

I built the SDK using newest master, pulled your PR, then built the SDK again: `bitbake <image_name> -c do_populate_sdk`. The recipe `compiler-rt-sanitizers` did not run, I had to manually run it with `bitbake  compiler-rt-sanitizers`. It finished successfully. 

After the sanitizers compiled I built the SDK one more time. However the newly built `sdk.sh` when installed did not include the  sanitizer libraries in the target aarch64 SDK sysroot. I am sure there is something small I am missing, do I need to build the SDK from scratch?

I also compiled the user image and ran the image on my embedded device, it does not contain the sanitizers as well. Moreover, it doesn't contain `libc++`, which is contained in the SDK. The latter is likely due to the fact that I didn't set `RUNTIME` variable in `local.conf`.

Thank you for your support and good work, I look forward to use the sanitizer libraries on my embedded device.

kraj, 2021-05-19:
@matrackif right now its an independent recipe so you have to explicitly add it to your image. The libraries are in `compiler-rt-sanitizers` and `compiler-rt-sanitizers-static` and `compiler-rt-sanitizers-dev` to `TOOLCHAIN_TARGET_TASK`

matrackif, 2021-05-19:
@kraj Thank you yet again, after adding to `TOOLCHAIN_TARGET_TASK` those libraries I received an error during recipe parsing:

`ERROR: Nothing RPROVIDES 'compiler-rt-sanitizers-static'`

So I removed `compiler-rt-sanitizers-static` and tried again, parsing succeeded but now there is conflict during `do_populate_sdk`:

```
Error: Transaction check error:
  file /usr/lib/clang/12.0.0/lib/linux/libclang_rt.fuzzer-aarch64.a conflicts between attempted installs of compiler-rt-sanitizers-staticdev-12.0.0-r0.aarch64 and compiler-rt-staticdev-12.0.0-r0.aarch64
  file /usr/lib/clang/12.0.0/lib/linux/libclang_rt.fuzzer_interceptors-aarch64.a conflicts between attempted installs of compiler-rt-sanitizers-staticdev-12.0.0-r0.aarch64 and compiler-rt-staticdev-12.0.0-r0.aarch64
  file /usr/lib/clang/12.0.0/lib/linux/libclang_rt.fuzzer_no_main-aarch64.a conflicts between attempted installs of compiler-rt-sanitizers-staticdev-12.0.0-r0.aarch64 and compiler-rt-staticdev-12.0.0-r0.aarch64
  file /usr/lib/clang/12.0.0/lib/linux/libclang_rt.profile-aarch64.a conflicts between attempted installs of compiler-rt-sanitizers-staticdev-12.0.0-r0.aarch64 and compiler-rt-staticdev-12.0.0-r0.aarch64
```

It seems there is a conflict between the recipes `compiler-rt` and `compiler-rt-sanitizer` since they both build the same libraries: `libclang_rt.fuzzer-aarch64.a`, `libclang_rt.fuzzer_interceptors-aarch64.a`, `libclang_rt.fuzzer_no_main-aarch64.a`, `libclang_rt.profile-aarch64.a`.

kraj, 2021-05-19:
sorry I should have said `compiler-rt-sanitizers-static-dev`

kraj, 2021-05-19:
> @kraj Thank you yet again, after adding to `TOOLCHAIN_TARGET_TASK` those libraries I received an error during recipe parsing:
>
> `ERROR: Nothing RPROVIDES 'compiler-rt-sanitizers-static'`
>
> So I [...]

yes thanks for report, we need to disable fuzzers in `compiler-rt` now.

matrackif, 2021-05-19:
@kraj 

My workaround was to use disable building fuzzers and profiler in `compiler-rt-sanitizers_git.bb`, lines 41-42:

```
-DCOMPILER_RT_BUILD_LIBFUZZER=OFF
-DCOMPILER_RT_BUILD_PROFILE=OFF
``` 

SDK install succeeded, the following files are now contained in the target sysroot `/usr/lib/clang/12.0.0/lib/linux/`:

```
libclang_rt.asan-aarch64.a
libclang_rt.asan-aarch64.so
libclang_rt.asan_cxx-aarch64.a
libclang_rt.asan-preinit-aarch64.a
libclang_rt.builtins-aarch64.a
libclang_rt.cfi-aarch64.a
libclang_rt.cfi_diag-aarch64.a
libclang_rt.dfsan-aarch64.a
libclang_rt.fuzzer-aarch64.a
libclang_rt.fuzzer_interceptors-aarch64.a
libclang_rt.fuzzer_no_main-aarch64.a
libclang_rt.hwasan-aarch64.a
libclang_rt.hwasan-aarch64.so
libclang_rt.hwasan_cxx-aarch64.a
libclang_rt.lsan-aarch64.a
libclang_rt.msan-aarch64.a
libclang_rt.msan_cxx-aarch64.a
libclang_rt.profile-aarch64.a
libclang_rt.safestack-aarch64.a
libclang_rt.scudo-aarch64.a
libclang_rt.scudo-aarch64.so
libclang_rt.scudo_cxx-aarch64.a
libclang_rt.scudo_cxx_minimal-aarch64.a
libclang_rt.scudo_minimal-aarch64.a
libclang_rt.scudo_minimal-aarch64.so
libclang_rt.stats-aarch64.a
libclang_rt.stats_client-aarch64.a
libclang_rt.ubsan_minimal-aarch64.a
libclang_rt.ubsan_minimal-aarch64.so
libclang_rt.ubsan_standalone-aarch64.a
libclang_rt.ubsan_standalone-aarch64.so
libclang_rt.ubsan_standalone_cxx-aarch64.a
libclang_rt.xray-aarch64.a
libclang_rt.xray-basic-aarch64.a
libclang_rt.xray-fdr-aarch64.a
libclang_rt.xray-profiling-aarch64.a
```

kraj, 2021-05-19:
I have pushed a fix into #464

matrackif, 2021-05-19:
@kraj I confirm your latest commit works.

As a final note I will say that I have tested clang's undefined behavior sanitizer and it works like a charm on my aarch64 system. We have a C++ CMake project that uses the yocto SDK clang 12 with a toolchain file, when we compile with the flag `-fsanitize=undefined`, then the following program:

```
#include <limits>

int main(int argc, char **argv)
{
    auto sanitizerTest = std::numeric_limits<int>::max();
    ++sanitizerTest;
    return 0;
}
```

yields as expected:

`/main.cpp:: runtime error: signed integer overflow: 2147483647 + 1 cannot be represented in type 'int'`

This is a powerful feature for all embedded linux systems and the C/C++ applications running on them. The ability to use the newest clang sanitizers on such systems is a substantial contribution. 

Regards

kraj, 2021-05-19:
@matrackif thanks for testing and reporting back. Enjoy the sanitizers !!

kraj, 2021-05-20:
this is now in master

closed 2021-05-20 by kraj