kraj/meta-clang #362: ROCKO: compiler-rt does not build

fwuehr, 2020-08-26:
Hello @kraj,
Hello @all,

we are using an SDK that is based on yocto (rocko, planning to migrate to thud) which we want to improve with clang. GCC should stay the default and we want to use GNU libstdc++ as well, but provide a cross-compiling clang as an alternative to GCC.

The project is setup as a cmake project for an ARM-Cortex-A15 chip (arm armv7a vfp thumb neon callconvention-hard)
I tried several options, you are describing in your manual but never got a working solution to be able to implement Clang into the toolchain and get it to work with cmake. 

Mostly I have problems with building compiler-rt. This never works and always crashes with the following error message:

relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC

I did add "-fPIC" , now compiler-rt builds but I am having troubles with libunwind:

|   Compiler doesn't support generation of unwind tables if exception support
|   is disabled.  Building libunwind DSO with runtime dependency on C++ ABI
|   library is not supported.
| 

Do you have any hint for me what would be the best way to do what I intend to or what I am doing wrong ?

Thanks in advance!

kraj, 2020-08-27:
see if you can use `-fno-unwind-tables ` compiler option. Btw. if you are planning to move away from rocko then move to dunfell since that's whats maintained and LTS release.

fwuehr, 2020-08-27:
Now I don´t see the previous error message anymore, but another one came up:

 CMake Error at cmake/Modules/AddCompilerRT.cmake:196 (add_library):
|   The install of the clang_rt.ubsan_standalone-dynamic-armhf target requires
|   changing an RPATH from the build tree, but this is not supported with the
|   Ninja generator unless on an ELF-based platform.  The
|   CMAKE_BUILD_WITH_INSTALL_RPATH variable may be set to avoid this relinking
|   step.
| Call Stack (most recent call first):
|   lib/ubsan/CMakeLists.txt:151 (add_compiler_rt_runtime)
| 
| 
| CMake Error at cmake/Modules/AddCompilerRT.cmake:196 (add_library):
|   The install of the clang_rt.ubsan_standalone_cxx-dynamic-armhf target
|   requires changing an RPATH from the build tree, but this is not supported
|   with the Ninja generator unless on an ELF-based platform.  The
|   CMAKE_BUILD_WITH_INSTALL_RPATH variable may be set to avoid this relinking
|   step.
| Call Stack (most recent call first):
|   lib/ubsan/CMakeLists.txt:161 (add_compiler_rt_runtime)
| 
| 
| CMake Error at cmake/Modules/AddCompilerRT.cmake:196 (add_library):
|   The install of the clang_rt.asan-dynamic-armhf target requires changing an
|   RPATH from the build tree, but this is not supported with the Ninja
|   generator unless on an ELF-based platform.  The
|   CMAKE_BUILD_WITH_INSTALL_RPATH variable may be set to avoid this relinking
|   step.
| Call Stack (most recent call first):
|   lib/asan/CMakeLists.txt:196 (add_compiler_rt_runtime)

@kraj: Do you have any idea, whats going wrong there ?
BTW: I have to stick to rocko at the moment and I am using rocko branch of meta-clang.

Thanks for your hint with dunfell, I will try to move directly to dunfell then

kraj, 2020-08-27:
You can disable sanitizers which dont build a patch like this https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/clang/0005-compiler-rt-Disable-tsan-on-OE-glibc.patch will be a good start.

fwuehr, 2020-08-31:
I was able to generate a toolchain now, but when I build a simple test application I get the following error at linking:

cannot find /opt/poky/sysroots/armv7ahf-neon-oe-linux-gnueabi/usr/lib/clang/5.0.1/lib/linux/libclang_rt.builtins-armhf.a: No such file or directory

I tried to find this file and it´s not located in the sysroots directories

kraj, 2020-08-31:
@fwuehr it perhaps means it did not generate compiler-rt builtins library. Either you need to fix that or add `--rtlib=compiler-rt` to compiler commandline

kraj, 2020-10-19:
Hi @fwuehr I will close this as the original issue was addressed. Please open new ticket if you still see problems.

closed 2020-10-19 by kraj