kraj/meta-clang #363: Wrong name busybox_1.32.%.bbappend

ska-la, 2020-08-27:
dunfell emits error on the name, as i remember symbol % must replace all version, not his part..

kraj, 2020-08-27:
> dunfell emits error on the name, as i remember symbol % must replace all
> version, not his part..

are you using dunfell branch on meta-clang

ska-la, 2020-08-28:
i checked out on the Dunfell branch, but busybox_1.31.%.bbappend is the
wrong name still..

kraj, 2020-08-28:
help me understand whats wrong ?

```
% bitbake-layers show-appends busybox
NOTE: Starting bitbake server...
NOTE: Started PRServer with DBfile: /mnt/b/yoe/dunfell/cache/prserv.sqlite3, IP: 127.0.0.1, PORT: 39205, PID: 2546675
Loading cache: 100% |##########################################################################################################################################################################################################| Time: 0:00:01
Loaded 4570 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################################################################| Time: 0:00:01
Parsing of 3267 .bb files complete (3266 cached, 1 parsed). 4571 targets, 631 skipped, 0 masked, 0 errors.
=== Matched appended recipes ===
busybox_1.31.1.bb:
  /mnt/b/yoe/dunfell/sources/meta-clang/recipes-core/busybox/busybox_1.31.%.bbappend
  /mnt/b/yoe/dunfell/sources/meta-atmel/recipes-core/busybox/busybox_%.bbappend
  /mnt/b/yoe/dunfell/sources/meta-yoe/recipes-core/busybox/busybox_%.bbappend
```

ska-la, 2020-08-31:
Sorry, i've spent your time. I'm an old donkey.

kraj, 2020-09-09:
I am closing this issue please reopen if you think its not resolved.

closed 2020-09-09 by kraj