kraj/meta-clang #366: libcxx do_compile failed occasionally during parallel build

sandy-lcq, 2020-08-31:
**Describe the bug**
libcxx do_compile failed occasionally

**To Reproduce**
hard to reproduce.  I attached 2 failed logs
[1.log.do_compile.txt](https://github.com/kraj/meta-clang/files/5148701/1.log.do_compile.txt)
[2.log.do_compile.txt](https://github.com/kraj/meta-clang/files/5148702/2.log.do_compile.txt)

**Something have done**
I analyzed the attached log 1.log.do_compile.txt,  and I think this is  a problem caused by parallel build.
and  the root cause maybe in cmake,  target_link_libraries cannot link dynamic library correctly. when 
both .a and .so will be generated parallelly.  But I don't find proper way to fix this.  

I have reported this issue to llvm and cmake,  but no response yet,  here is the link,  
https://bugs.llvm.org/show_bug.cgi?id=46617    //you can refer this link see more details
https://discourse.cmake.org/t/cannot-link-to-dynamic-library-during-parallel-build/1685

**Expected behavior**
Could you give some suggestions about how to fix this problem? Thanks

kraj, 2020-09-09:
Perhaps disable static libs ?

sandy-lcq, 2020-12-14:
problem have been fix by upstream, refer  https://github.com/llvm/llvm-project/commit/bcaa01752f5e45b9d1647aef16d74c748192f8a5,   so close this defect

closed 2020-12-14 by sandy-lcq