kraj/meta-clang #368: Zeus: populate_sdk fails on gold linker option

wdobbe, 2020-09-09:
**Describe the bug**
I added meta-clang to my Yocto Zeus project. After that command 'bitbake <myimage> -c populate_sdk failed because the linker doesn't know option '-fuse-ld=gold'.
Target is based on the NXP i.MX6q SoC.

**To Reproduce**
Download NXP Yocto framework (zeus branch).
Try to build SDK with 'bitbake core-image-full-cmdline -c populate_sdk'

**Expected behavior**
Yocto SDK is built without errors and includes clang compiler.

**Desktop (please complete the following information):**
 - OS: Suse Linux Enterprise Server 15 SP1

**Possible Fix**
The 'use-ld=gold' flag is added in recipe meta-clang/recipes-devtools/clang/clang_git.bb.
The following patch fixes the build on my system, but I am not sure if this is the correct way:

```
diff --git a/recipes-devtools/clang/clang_git.bb b/recipes-devtools/clang/clang_git.bb
index d9d36ad..c59bf98 100644
--- a/recipes-devtools/clang/clang_git.bb
+++ b/recipes-devtools/clang/clang_git.bb
@@ -15,7 +15,7 @@ BUILD_CXX_class-nativesdk = "clang++"
 BUILD_AR_class-nativesdk = "llvm-ar"
 BUILD_RANLIB_class-nativesdk = "llvm-ranlib"
 BUILD_NM_class-nativesdk = "llvm-nm"
-LDFLAGS_append_class-nativesdk = " -fuse-ld=gold"
+LDFLAGS_append_class-nativesdk = "${@bb.utils.contains('DISTRO_FEATURES', 'ld-is-gold', ' -fuse-ld=gold', '', d)}"
 
 inherit cmake cmake-native pkgconfig python3native
```

**Additional context (log)**
```
wdobbe@dev-linux-bs03:~/yocto/flownode_zeus/build-flownode-acu> bitbake dynniq-image-flownode-acu-default -c populate_sdk
WARNING: You have included the meta-virtualization layer, but 'virtualization' has not been enabled in your DISTRO_FEATURES. Some bbappend files may not take effect. See the meta-virtualization README for details on enabling virtualization support.
WARNING: Host distribution "sles-15.1" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################################################################| Time: 0:00:00
Loaded 4229 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################################| Time: 0:00:00
Parsing of 3031 .bb files complete (3023 cached, 8 parsed). 4237 targets, 476 skipped, 2 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "dynniq-flownode-acu"
DISTRO               = "dynniq-flownode-acu-bsp"
DISTRO_VERSION       = "99.9.9"
TUNE_FEATURES        = "arm vfp cortexa9 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:f2eb22a8783f1eecf99bd4042695bab920eed00e"
meta-oe              
meta-multimedia      
meta-python          
meta-networking      
meta-webserver       
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:8c136c4362d0f14ceb5cceb27467ad9db5e9f93d"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-bsp             
meta-sdk             
meta-ml              = "HEAD:63887634fab8c6305f0c825b93d1087c72ee2b14"
meta-qt5             = "HEAD:1861b68525dc8566be6bc8d70e4359a3b589e4ca"
meta-rust            = "HEAD:5cda04c7c2158fc982502ca1e7893f71e4ca1729"
[... 52 lines cut ...]
| -- Check for working C compiler: /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang
| -- Check for working C compiler: /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang -- broken
| CMake Error at /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/share/cmake-3.15/Modules/CMakeTestCCompiler.cmake:60 (message):
|   The C compiler
| 
|     "/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command(s):ninja cmTC_7c33c && [1/2] Building C object CMakeFiles/cmTC_7c33c.dir/testCCompiler.c.o
|     [2/2] Linking C executable cmTC_7c33c
|     FAILED: cmTC_7c33c
|     : && /home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang --sysroot=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot  -O2 -pipe -fmacro-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0=/usr/src/debug/nativesdk-clang/9.0.1-r0                      -fdebug-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0=/usr/src/debug/nativesdk-clang/9.0.1-r0                      -fdebug-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native=   --sysroot=/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot  -Wl,-O1                                            -Wl,-rpath-link,/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/usr/lib/..                                            -Wl,-rpath,/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/usr/lib/..  -Wl,-dynamic-linker,/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 -fuse-ld=gold -Wl,-O1                                            -Wl,-rpath-link,/home/wdobbe/yocto/flownode_zeus/build-flownode-acu/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/usr/lib/..                                            -Wl,-rpath,/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/usr/lib/..  -Wl,-dynamic-linker,/opt/dynniq-yoctosdk-host_x86_64-target_cortexa9hf-neon/3.0.0/sysroots/x86_64-pokysdk-linux/lib/ld-linux-x86-64.so.2 -fuse-ld=gold CMakeFiles/cmTC_7c33c.dir/testCCompiler.c.o  -o cmTC_7c33c   && :
|     clang-9: error: invalid linker name in argument '-fuse-ld=gold'
|     ninja: build stopped: subcommand failed.
| 
<cut similar output>
| 
ERROR: Task (virtual:nativesdk:/home/wdobbe/yocto/flownode_zeus/sources/meta-clang/recipes-devtools/clang/clang_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3333 tasks of which 3329 didn't need to be rerun and 1 failed.
```

kraj, 2020-09-09:
DISTRO_FEATURE check is mainly for target recipes not nativesdk ones, We use gold because we want to use LTO to compile second stage clang, it seems clang expects ld.gold binary in same directory as clang itself, which is the reason perhaps for this error, so maybe smlinking ld.gold would help but can you try using LLD instead of gold here

`-fuse-ld=lld` and see if that helps.

wdobbe, 2020-09-10:
Hi,

When I replace `-fuse-ld=gold` with `-fuse-ld=lld` the SDK builds successfully.

kraj, 2020-09-10:
OK can you send a patch using lld.

wdobbe, 2020-09-11:
Pull request #372 submitted.

closed 2020-09-11 by kraj