kraj/meta-clang #370: I would like "bitbake -c populate_sdk_ext ..." to include clang

cgsfv, 2020-09-10:
When I run "bitbake -c populate_sdk ..." the generated SDK nicely includes clang cross compiler.
When I run "bitbake -c populate_sdk_ext ..." the generated Extensible SDK does not include clang cross compiler.
I would like a way to also include the clang cross compiler into the Extensible SDK.

kraj, 2020-09-10:
@cgsfv are you setting `CLANGSDK = "1"` in local.conf ?

cgsfv, 2020-09-10:
Yes. I am using the same config with `CLANGSDK = "1"` in both cases.

kraj, 2020-09-10:
> Yes. I am using the same config with `CLANGSDK = "1"` in both cases.

OK, and to further clarify, I are looking to include cross compiler ( precompiled ) like standard SDK ? in other words
are you setting `SDK_INCLUDE_TOOLCHAIN = "1"`

cgsfv, 2020-09-11:
Yes, I am looking to include a precompiled clang cross compiler in the Extensible SDK, just like it gets included in the SDK. Just like the GCC cross compiler gets included in both SDK's. I have also tried with both `CLANGSDK = "1"` and `SDK_INCLUDE_TOOLCHAIN = "1"` still without any success.

kraj, 2020-09-11:
@cgsfv ok, I think we need to add a bbappend for `meta-extsdk-toolchain.bb` in meta-clang and add appropriate dependencies if CLANGSDK is set in env

cgsfv, 2020-09-11:
Great - if I can assist in anyway, do let me know

cgsfv, 2020-09-17:
Still the same result - no clang in Extensible SDK.

kraj, 2020-09-18:
I can see it when I build eSDK. Once you source env then do `echo $CLANGCC` you should see it

```
% . ./environment-setup-cortexa57-yoe-linux-musl
SDK environment now set up; additionally you may now run devtool to perform development tasks.
Run devtool --help for further details.
[kraj@apollo ~/yoe_sdk/3.2.0-alpha]
% echo $CLANGCC
aarch64-yoe-linux-musl-clang -mcpu=cortex-a57+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian --sysroot=/home/kraj/yoe_sdk/3.2.0-alpha/tmp/sysroots/qemuarm64
```

kraj, 2020-10-19:
@cgsfv I will close this since the original issue has been resolved and as seen above I can see the clang compiler in eSDK

closed 2020-10-19 by kraj