kraj/meta-clang #375: Error configuring libcxx_git

vaibhavjadhav377, 2020-09-16:
Hi all,
I am facing issue while building clang tools. I am using meta-clang dunfell branch. 
I am hit by error as follows:
/home/vaibhav/openembedded/openembedded-core/build/tmp-glibc/work/riscv64-oe-linux/compiler-rt/10.0.1-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/riscv64-oe-linux-ld: cannot find crtbeginS.o: No such file or directory
/home/vaibhav/openembedded/openembedded-core/build/tmp-glibc/work/riscv64-oe-linux/compiler-rt/10.0.1-r0/recipe-sysroot-native/usr/bin/riscv64-oe-linux/riscv64-oe-linux-ld: cannot find -lgcc
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
So I want to know why crtbeginS.o is missing and not able to find gcc ?
Or I am missing some configurations while building clang.
Also I have attached log for more information.
[CMakeError.log](https://github.com/kraj/meta-clang/files/5229944/CMakeError.log)
[CMakeOutput.log](https://github.com/kraj/meta-clang/files/5229946/CMakeOutput.log)
Thanks.

davidlt, 2020-09-16:
This might help: https://github.com/davidlt/meta-clang/commit/df75aab1b576bed3ed0cad9cc4ed994e802edc98

Haven't rebased my fork yet.

vaibhavjadhav377, 2020-09-17:
@davidlt Thanks for the patch you provided, It resolved the above errors.
But another error arise, libclang_rt.builtins.riscv64.a is missing.

kraj, 2020-09-17:
@vaibhavjadhav377 can you try master branch of meta-clang with rest of layers being dunfell ?

vaibhavjadhav377, 2020-09-17:
@kraj I will try and get back as soon as possible.

vaibhavjadhav377, 2020-09-17:
Hi @kraj 
I am getting following error after trying master branch of meta-clang :
| CMake Error at cmake/config-ix.cmake:439 (message):
|   Unknown architecture riscv
| Call Stack (most recent call first):
|   CMakeLists.txt:655 (include)
while building the recipe for clang_git.bb

kraj, 2020-09-17:
clang for target does not yet work on master perhaps @davidlt 's patches are required. but your issue of compiling compiler-rt should be gone.

vaibhavjadhav377, 2020-09-17:
yes, compiler-rt issue is resolved.

kraj, 2020-09-17:
Can you apply @davidlt patch on master and see if it works to solve target clang builds for riscv64

vaibhavjadhav377, 2020-09-19:
@kraj Sorry for late reply.
I did apply the patch, which were available in meta-clang branch-riscv @davidlt.
It did solved the target clang build problem.

kraj, 2020-09-19:
> @kraj Sorry for late reply.
> I did apply the patch, which were available in meta-clang branch-riscv @davidlt.
> It did solved the target clang build problem.

Ok nice to here do you mind posting that change here as a pull ?

davidlt, 2020-09-19:
@kraj note, that I never tested SDK builds with Clang, only native. Last time I tested was with Clang 10.X (incl. libc++, libstdc++v3, OpenMP [GNU and LLVM] and all seemed to work as expected).

Sadly I still didn't get to test with Clang 11, but it's high on TODO list.

kraj, 2020-09-19:
@davidlt no worries, I think if we have it ported over to clang11 branch then I am willing to try it out and see if it pans out we can apply it and declare  rv64 support on target for clang

vaibhavjadhav377, 2020-09-19:
> @kraj Sorry for late reply.
> I did apply the patch, which were available in meta-clang branch-riscv @davidlt.
> It did solved the target clang build problem.
@kraj, I forgot to mention that the patches I applied were for 10.x.
As per your suggestion you asked me to try davidlt patches for master i.e 11.x. to get rid of target clang build error.
I am still working on it and will update you asap.

kraj, 2020-10-19:
clang11 on master branch can now build for rv32/rv64, I have merged few commits from @davidlt with little rework. Please reopen if you still see this issue.

closed 2020-10-19 by kraj