kraj/meta-clang #394: Question: Useful for building clang tools for the target?

Jasper-Ben, 2020-10-19:
Just wondering if this layer can help me in some way on my quest to build clang-tidy for the target? :smile:

kraj, 2020-10-19:
@Jasper-Ben clang-tidy is already part of the clang recipe when built for target, all you need is `IMAGE_INSTALL_append = " clang"` in your image recipe.

Jasper-Ben, 2020-10-19:
cheers! I am relatively new to yocto and understood the description of the layer to only provide the tools needed to replace the gcc compiler during bitbake builds.

Jasper-Ben, 2020-10-20:
Followup: Building clang currently fails for me. Any ideas why this might be the case?

Used versions:
yocto: thud
meta-clang: thud

Output:
```
Initialising tasks: 100% |#####################################################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 67 Found 43 Missed 24 Current 160 (64% match, 89% complete)                                                                                                                                                           
NOTE: Executing SetScene Tasks                                                                                                                                                                                                               
NOTE: Executing RunQueue Tasks                                                                                                                                                                                                               
ERROR: clang-native-7.1.0-r0 do_configure: Function failed: do_configure (log file is located at /mnt/build/tmp/work/x86_64-linux/clang-native/7.1.0-r0/temp/log.do_configure.12643)                                                         
ERROR: Logfile of failure stored in: /mnt/build/tmp/work/x86_64-linux/clang-native/7.1.0-r0/temp/log.do_configure.12643                                                                                                                      
Log data follows:                                                                                                                                                                                                                            
| DEBUG: Executing shell function do_configure                                                                                                                                                                                               
| -- The C compiler identification is GNU 8.3.0                                                                                                                                                                                              
| -- The CXX compiler identification is GNU 8.3.0                                                                                                                                                                                            
| -- The ASM compiler identification is GNU                                                                                                                                                                                                  
| -- Found assembler: /mnt/build/tmp/hosttools/gcc                                                                                                                                                                                           
| -- Check for working C compiler: /mnt/build/tmp/hosttools/gcc                                                                                                                                                                              
| CMake Error:                                                                                                                                                                                                                               
|   The detected version of Ninja (GNU Make 4.2.1                                                                                                                                                                                            
|                                                                                                                                                                                                                                            
|   Built for x86_64-pc-linux-gnu                                                                                                                                                                                                            
|                                                                                                                                                                                                                                            
|   Copyright (C) 1988-2016 Free Software Foundation, Inc.                                                                                                                                                                                   
|                                                                                                                                                                                                                                            
|   License GPLv3+: GNU GPL version 3 or later                                                                                                                                                                                               
|   <http://gnu.org/licenses/gpl.html>                                                                                                                                                                                                       
[... cut ...]
| See also "/mnt/build/tmp/work/x86_64-linux/clang-native/7.1.0-r0/build/CMakeFiles/CMakeOutput.log".                                                                                                                                        
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /mnt/build/tmp/work/x86_64-linux/clang-native/7.1.0-r0/temp/log.do_configure.12643)
ERROR: Task (virtual:native:/mnt/meta-clang/recipes-devtools/clang/clang_git.bb:do_configure) failed with exit code '1'
```

Jasper-Ben, 2020-10-20:
Ok, I guess I have to explicitly set Ninja instead of GNU Make, as said by this butchered Error message:

```| CMake Error:                                                                                                                                                                                                                               
|   The detected version of Ninja (GNU Make 4.2.1 ...) is less than the                                                                                                                                                                  
|   version of Ninja required by CMake (1.3).            ```

Jasper-Ben, 2020-10-21:
I am still trying to figure this out. I tried using a clang_%.bbappend file to overwrite TOOLCHAIN to "clang" as I thought this might be the issue. However, it seems yocto still tries to build the clang component using gcc:
```
Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is GNU 8.3.0
| -- The CXX compiler identification is GNU 8.3.0
| -- The ASM compiler identification is GNU
| -- Found assembler: /mnt/build/tmp/hosttools/gcc
| -- Check for working C compiler: /mnt/build/tmp/hosttools/gcc
```

I am at a lost what I might be doing wrong?

kraj, 2020-10-21:
is this log from `nativesdk-clang` build ?

Jasper-Ben, 2020-10-21:
yes

Jasper-Ben, 2020-10-23:
So here is a short update on this issue. As I also experienced issues with another layer I needed when using the yocto thud branch, I spent some time making our own layer compatible with latest and greatest yocto dunfell. Now clang-native builds, however it now fails at the task meta-clang/recipes-devtools/clang/libcxx_git.bb with following error:

```
| cd /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src && /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/lib
cxx/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_link_script CMakeFiles/cxx_shared.dir/link.txt --verbose=1
| /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-clang++ -fPIC -t
arget arm-poky-linux-gnueabi  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a5 -rtlib=compiler-rt --unwindlib=libgcc  -mlittle-endian -Qunused-arguments -fstack-prot
ector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r
0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0=/usr/src/debu
g/libcxx/10.0.1-r0                      -fdebug-prefix-map=/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0=/usr/src/debug/libcxx/10.0.1-r0    
                  -fdebug-prefix-map=/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/
mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native=   -fvisibility-inlines-hidden -target arm-poky-linux-gnueabi  -mthumb -m
fpu=neon -mfloat-abi=hard -mcpu=cortex-a5 -rtlib=compiler-rt --unwindlib=libgcc  -mlittle-endian -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wform
at -Wformat-security -Werror=format-security  --sysroot=/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot -fPIC -fvisibility-inli
nes-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers  -Wno-long-
long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -g -L/mnt/build/tmp/work/cortex
a5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-z,defs -Wl,-z,n
odelete -shared -Wl,-soname,libc++.so.1 -o ../../../lib/libc++.so.1.0 CMakeFiles/cxx_shared.dir/algorithm.cpp.o CMakeFiles/cxx_shared.dir/any.cpp.o CMakeFiles/cxx_share
d.dir/bind.cpp.o CMakeFiles/cxx_shared.dir/charconv.cpp.o CMakeFiles/cxx_shared.dir/chrono.cpp.o CMakeFiles/cxx_shared.dir/condition_variable.cpp.o CMakeFiles/cxx_share
d.dir/condition_variable_destructor.cpp.o CMakeFiles/cxx_shared.dir/debug.cpp.o CMakeFiles/cxx_shared.dir/exception.cpp.o CMakeFiles/cxx_shared.dir/functional.cpp.o CMa
keFiles/cxx_shared.dir/future.cpp.o CMakeFiles/cxx_shared.dir/hash.cpp.o CMakeFiles/cxx_shared.dir/ios.cpp.o CMakeFiles/cxx_shared.dir/iostream.cpp.o CMakeFiles/cxx_sha
red.dir/locale.cpp.o CMakeFiles/cxx_shared.dir/memory.cpp.o CMakeFiles/cxx_shared.dir/mutex.cpp.o CMakeFiles/cxx_shared.dir/mutex_destructor.cpp.o CMakeFiles/cxx_shared
.dir/new.cpp.o CMakeFiles/cxx_shared.dir/optional.cpp.o CMakeFiles/cxx_shared.dir/random.cpp.o CMakeFiles/cxx_shared.dir/regex.cpp.o CMakeFiles/cxx_shared.dir/shared_mu
tex.cpp.o CMakeFiles/cxx_shared.dir/stdexcept.cpp.o CMakeFiles/cxx_shared.dir/string.cpp.o CMakeFiles/cxx_shared.dir/strstream.cpp.o CMakeFiles/cxx_shared.dir/system_er
ror.cpp.o CMakeFiles/cxx_shared.dir/thread.cpp.o CMakeFiles/cxx_shared.dir/typeinfo.cpp.o CMakeFiles/cxx_shared.dir/utility.cpp.o CMakeFiles/cxx_shared.dir/valarray.cpp
.o CMakeFiles/cxx_shared.dir/variant.cpp.o CMakeFiles/cxx_shared.dir/vector.cpp.o CMakeFiles/cxx_shared.dir/filesystem/operations.cpp.o CMakeFiles/cxx_shared.dir/filesy
stem/directory_iterator.cpp.o CMakeFiles/cxx_shared.dir/filesystem/int128_builtins.cpp.o  -Wl,-rpath,"\$ORIGIN/../lib" $$($$CC --print-libgcc-file-name) -nodefaultlibs 
-lpthread -lc -lm -lrt -lgcc_s -lc++abi
| make[3]: Entering directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| make[3]: Nothing to be done for 'projects/libcxx/src/CMakeFiles/cxx_static.dir/build'.
| make[3]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| [ 80%] Built target cxx_static
| clang-10: error: unsupported option '--print-libgcc-file-name)'; did you mean '--print-libgcc-file-name'?
| clang-10: error: no such file or directory: '$$($$CC'
| make[3]: *** [projects/libcxx/src/CMakeFiles/cxx_shared.dir/build.make:612: lib/libc++.so.1.0] Error 1
```

I'll investigate further.

Jasper-Ben, 2020-10-30:
One issue seems to be that libc++.so.1.0 is never created:

```
| [100%] Linking CXX static library ../../../lib/libc++experimental.a
| cd /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src && /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -P CMakeFiles/cxx_ex
perimental.dir/cmake_clean_target.cmake
| make[3]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| [100%] Built target install-cxx-headers
| cd /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src && /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_link_script CMakeFiles/cxx_experimental.dir/link.txt --verbose=1
| /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-llvm-ar Dqc ../../../lib/libc++experimental.a  CMakeFiles/cxx_experimental.dir/experi$
ental/memory_resource.cpp.o
| /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-llvm-ranlib -D ../../../lib/libc++experimental.a
| make[3]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| [100%] Built target cxx_experimental
| make -f projects/libcxx/src/CMakeFiles/install-cxx.dir/build.make projects/libcxx/src/CMakeFiles/install-cxx.dir/depend
| make[3]: Entering directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| cd /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build && /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E cmake_depends "Unix Makefiles" /mnt/$
uild/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm /mnt/build/tmp/work-shared/llvm-project-source-10.0.1-r0/git/libcxx/src /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build /mnt/build/tmp/work/corte$
a5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src/CMakeFiles/install-cxx.dir/DependInfo.cmake --color=
| Dependee "/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src/CMakeFiles/install-cxx.dir/DependInfo.cmake" is newer than depender "/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi$
libcxx/10.0.1-r0/build/projects/libcxx/src/CMakeFiles/install-cxx.dir/depend.internal".
| Dependee "/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/$
ibcxx/10.0.1-r0/build/projects/libcxx/src/CMakeFiles/install-cxx.dir/depend.internal".
| Scanning dependencies of target install-cxx
| make[3]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| make -f projects/libcxx/src/CMakeFiles/install-cxx.dir/build.make projects/libcxx/src/CMakeFiles/install-cxx.dir/build
| make[3]: Entering directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| cd /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/src && /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -DCMAKE_INSTALL_COM$
ONENT=cxx -P /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/cmake_install.cmake
| -- Install configuration: "Debug"
| CMake Error at cmake_install.cmake:41 (file):
|   file INSTALL cannot find
|   "/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++.so.1.0":
|   No such file or directory.**
| Call Stack (most recent call first):
|   /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/projects/libcxx/cmake_install.cmake:51 (include)
| 
| 
| make[3]: *** [projects/libcxx/src/CMakeFiles/install-cxx.dir/build.make:60: projects/libcxx/src/CMakeFiles/install-cxx] Error 1
| make[3]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| make[2]: *** [CMakeFiles/Makefile2:24765: projects/libcxx/src/CMakeFiles/install-cxx.dir/all] Error 2
| make[2]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| make[1]: *** [CMakeFiles/Makefile2:24772: projects/libcxx/src/CMakeFiles/install-cxx.dir/rule] Error 2
| make[1]: Leaving directory '/mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build'
| make: *** [Makefile:6793: install-cxx] Error 2
| WARNING: exit code 2 from a shell command.
```

```
ls /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++* -la
-rw-r--r--. 1 builder builder 7674764 Oct 30 11:14 /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++.a
-rw-r--r--. 1 builder builder 1640220 Oct 30 11:14 /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++abi.a
[... cut ...]
-rw-r--r--. 1 builder builder   46354 Oct 30 11:14 /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++experimental.a
-rw-r--r--. 1 builder builder      28 Oct 30 11:14 /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++.so
lrwxrwxrwx. 1 builder builder      13 Oct 30 11:14 /mnt/build/tmp/work/cortexa5t2hf-neon-poky-linux-gnueabi/libcxx/10.0.1-r0/build/lib/libc++.so.1 -> libc++.so.1.0
```

libc++.so.1 is a symlink to libc++.so.1.0 but it seems the latter is never created.

kraj, 2020-10-30:
hmm so its a build race

Jasper-Ben, 2020-11-03:
I finally managed to figure this out. There were incompatibilities between our own layer and this layer...

closed 2020-11-03 by Jasper-Ben