kraj/meta-clang #417: Bitbake hangs while building

vijayrakesh13, 2021-01-15:
Hi,

I had downloaded meta-clang dunfell branch and added to the bblayers.conf then ran bitbake command, but my build started to rebuild a few recipes which are already built, after some time bitbake got slow and doesn't show any progress, I had wait for 2hours and stopped the bitbake manually. When I remove meta-clang from bblayers.conf then build runs as usual without any hanging. Do I have make changes anywhere? 

Thanks,
VR

kraj, 2021-01-15:
compilimg clang can take quite a long time and therefore having good machine for build is good option. Perhaps 8core/16GB RAM might be something good start.

vijayrakesh13, 2021-01-18:
I have increased RAM and I tried once again by adding meta-clang path to bblayers.conf and ran build without adding any recipe name to the local.conf file, and bitbake started to rebuild or building a new package liberror-perl at do_package, task got failed. Earlier also I got same error, when I re-ran the build error didn't appeared again, but now it doesn't happening like that. I'm sure where is the issue is from.

> WARNING: liberror-perl-0.17029-r0 do_package: KeyError in ./package/usr/lib/perl5
>      0589:                process(os.path.join(root, f))
> File: '/home/bl-docker/rity/src/poky/meta/lib/oe/sstatesig.py', lineno: 548, function: process
>      0544:                    add_perm(stat.S_IXOTH, 'x')
>      0545:
>      0546:                if include_owners:
>      0547:                    try:
>  *** 0548:                        update_hash(" %10s" % pwd.getpwuid(s.st_uid).pw_name)
>      0549:                        update_hash(" %10s" % grp.getgrgid(s.st_gid).gr_name)
>      0550:                    except KeyError:
>      0551:                        bb.warn("KeyError in %s" % path)
>      0552:                        raise
> Exception: KeyError: 'getpwuid(): uid not found: 1001'

vijayrakesh13, 2021-01-18:
I had cleaned the liberror-perl recipe and re-ran bitbake command, this time I had added clang in local.conf file. Now liberror-perl was successful, but few other recipes got failed at do_package and  clang_git_do compile failed with following error.

> | /usr/include/c++/7/chrono:326:9: internal compiler error: Segmentation fault
> |      && (treat_as_floating_point<rep>::value
> |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> |          || !treat_as_floating_point<_Rep2>::value)>::type>
> |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> Please submit a full bug report,
> | with preprocessed source if appropriate.
> | See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
> | [467/4665] /home/bl-docker/rity/bsp/tmp/hosttools/g++  -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Ilib/Transforms/IPO -I/home/bl-docker/rity/bsp/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Transforms/IPO -Iinclude -I/home/bl-docker/rity/bsp/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/include -isystem/home/bl-docker/rity/bsp/tmp/work/x86_64-linux/clang-native/10.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -DNDEBUG     -std=c++14 -MD -MT lib/Transforms/IPO/CMakeFiles/LLVMipo.dir/Attributor.cpp.o -MF lib/Transforms/IPO/CMakeFiles/LLVMipo.dir/Attributor.cpp.o.d -o lib/Transforms/IPO/CMakeFiles/LLVMipo.dir/Attributor.cpp.o -c /home/bl-docker/rity/bsp/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Transforms/IPO/Attributor.cpp
> | ninja: build stopped: subcommand failed.
> | WARNING: /home/bl-docker/rity/bsp/tmp/work/x86_64-linux/clang-native/10.0.1-r0/temp/run.do_compile.13897:1 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} }VERBOSE=1 cmake --build '/home/bl-docker/rity/bsp/tmp/work/x86_64-linux/clang-native/10.0.1-r0/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'

kraj, 2021-01-18:
host gcc is still dying. It seems ram is still not enough. You might want to set parallelism relevant to number of CPUs you have so say if you have 8 cores on build host then then try in local.conf something like below 

```
PARALLEL_MAKE = "-j 4"
BB_NUMBER_THREADS = "4"

```

and see if that helps.

kraj, 2021-01-25:
@vijayrakesh13 did you make some progress ?

vijayrakesh13, 2021-01-26:
@kraj No, after adding meta-clang my build failed at liberror-perl. 
> 
> | *** Module name OUT: /home/rity/bsp/tmp/work/aarch64-poky-linux/liberror-perl/0.17029-r0/recipe-sysroot-native/usr/lib/perl5/5.30.1/x86_64-linux/PerlIO/encoding.pm
> | Generating a Unix-style Makefile
> | Writing Makefile for Error
> | Writing MYMETA.yml and MYMETA.json
> | ==> Your Makefile has been rebuilt. <==
> | ==> Please rerun the make command.  <==
> | false
> | Makefile:800: recipe for target 'Makefile' failed
> | make: *** [Makefile] Error 1
> | ERROR: oe_runmake failed
> | WARNING: /home/rity/bsp/tmp/work/aarch64-poky-linux/liberror-perl/0.17029-r0/temp/run.do_install.30398:1 exit 1 from 'exit 1'
> | ERROR: Execution of '/home/rity/bsp/tmp/work/aarch64-poky-linux/liberror-perl/0.17029-r0/temp/run.do_install.30398' failed with exit code 1:

vijayrakesh13, 2021-01-26:
@kraj Thank you, issue resolved when I cleaned liberror-perl. May I know why bitbake is rebuilding everything when meta-clang added to the bblayer.conf?

closed 2021-01-26 by vijayrakesh13