kraj/meta-clang #418: Cross build: clang-tidy co-compile not working when using the clangsdk with cmake

McDisaster, 2021-01-17:
First of all, thank you for the great work! Just had to add the layer and the clangsdk, works like a charm! :)

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-bikeos-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "bikeos"
DISTRO_VERSION       = "1.0.0u-git+8e234428-dbg"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:e32d854e33bc86c2a616df8708e021a098afcf73"
meta-oe              
meta-multimedia      
meta-networking      
meta-python          = "HEAD:f2d02cb71eaff8eb285a1997b30be52486c160ae"
meta-raspberrypi     = "HEAD:987993209716302eb8f314f69a2a3340555f94d8"
meta-clang           = "HEAD:e63d6f9abba5348e2183089d6ef5ea384d7ae8d8"
meta-qt5             = "HEAD:0d8eb956015acdea7e77cd6672d08dce18061510"
meta-bikeos          = "HEAD:8e2344281da7f9ffc2aa815a375329ba8be288d3"
```

But I ran into a tiny issue on the dunfell branch (haven't seen a fix in the master) that I could fix with a custom bbappend. You might want to look into it.

I'm using cmake to build a hobby project with the clang sdk. To ensure that basic coding guidelines are checked I'm using the clang-tidy provided by your layer. The issue here is the following, I'm always cross-compiling. 
```cmake
find_program(CLANG_TIDY_COMMAND NAMES clang-tidy)
set_target_properties(${TARGET} PROPERTIES CXX_CLANG_TIDY "${CLANG_TIDY_COMMAND}")
```
```
[build] /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/cmake -E __run_co_compile --tidy=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/clang-tidy --source=../application/source/Generic/Checks.cpp -- /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/arm-bikeos-linux-gnueabi/arm-bikeos-linux-gnueabi-clang++   -march=armv6 -mfpu=vfp -mfloat-abi=hard -mtune=arm1176jzf-s -mfpu=vfp -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian -mlittle-endian --sysroot=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi --sysroot=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_LOCATION_LIB -DQT_NETWORK_LIB -DQT_OPENGL_LIB -DQT_POSITIONINGQUICK_LIB -DQT_POSITIONING_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_QUICK_LIB -DQT_SQL_LIB -DQT_WIDGETS_LIB -Dbikeapp_generic_EXPORTS -Iapplication/bikeapp_generic_autogen/include -I../application/include -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtQml -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtNetwork -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtCore -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/lib/mkspecs/linux-oe-g++ -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtQuick -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtQmlModels -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtGui -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtLocation -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtPositioningQuick -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtPositioning -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtSql -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtOpenGL -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtWidgets -O2 -pipe -g -feliminate-unused-debug-types  -g -fPIC   --coverage -Wall -Wextra -Wpedantic -Wfatal-errors -Wno-gnu-zero-variadic-macro-arguments -fPIC -std=gnu++2a -MD -MT application/CMakeFiles/bikeapp_generic.dir/source/Generic/Checks.cpp.o -MF application/CMakeFiles/bikeapp_generic.dir/source/Generic/Checks.cpp.o.d -o application/CMakeFiles/bikeapp_generic.dir/source/Generic/Checks.cpp.o -c ../application/source/Generic/Checks.cpp
[build] error: unknown target CPU 'armv6' [clang-diagnostic-error]
[build] note: valid target CPU values are: nocona, core2, penryn, bonnell, atom, silvermont, slm, goldmont, goldmont-plus, tremont, nehalem, corei7, westmere, sandybridge, corei7-avx, ivybridge, core-avx-i, haswell, core-avx2, broadwell, skylake, skylake-avx512, skx, cascadelake, cooperlake, cannonlake, icelake-client, icelake-server, tigerlake, knl, knm, k8, athlon64, athlon-fx, opteron, k8-sse3, athlon64-sse3, opteron-sse3, amdfam10, barcelona, btver1, btver2, bdver1, bdver2, bdver3, bdver4, znver1, znver2, x86-64
[build] Error while processing /home/tobias/Repos/cycle-clever/bike-app/build/../application/source/Generic/Checks.cpp.
```

When applying this patch everything works like a charm, at least for a cross-compile

```patch
From 9a1b6fabe90e0f5b818da70627d5c7f73e30e7da Mon Sep 17 00:00:00 2001
From: Tobias Bayer <tobiasbayer@outlook.de>
Date: Sun, 17 Jan 2021 11:13:18 +0100
Subject: [PATCH 1/1] Fix missing target option for clang-tidy co-compile

---
 recipes-core/meta/meta-environment.bbappend | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/recipes-core/meta/meta-environment.bbappend b/recipes-core/meta/meta-environment.bbappend
index d59bfd6..a238e49 100644
--- a/recipes-core/meta/meta-environment.bbappend
+++ b/recipes-core/meta/meta-environment.bbappend
@@ -11,9 +11,9 @@ TARGET_CLANGCC_ARCH_remove_powerpc = "-mno-spe"
 create_sdk_files_append() {
         script=${SDK_OUTPUT}/${SDKPATH}/environment-setup-${REAL_MULTIMACH_TARGET_SYS}
         if ${@bb.utils.contains('CLANGSDK', '1', 'true', 'false', d)}; then
-            echo 'export CLANGCC="${TARGET_PREFIX}clang ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
-            echo 'export CLANGCXX="${TARGET_PREFIX}clang++ ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
-            echo 'export CLANGCPP="${TARGET_PREFIX}clang -E ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
+            echo 'export CLANGCC="${TARGET_PREFIX}clang --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
+            echo 'export CLANGCXX="${TARGET_PREFIX}clang++ --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
+            echo 'export CLANGCPP="${TARGET_PREFIX}clang -E --target=${TARGET_SYS} ${TARGET_CLANGCC_ARCH} --sysroot=$SDKTARGETSYSROOT"' >> $script
             echo 'export CLANG_TIDY_EXE="${TARGET_PREFIX}clang-tidy"' >> $script
         fi
 }
-- 
2.25.1

```
How it behaves with my patch file:
```
[12/19  63% :: 29.122] /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/cmake -E __run_co_compile --tidy=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/clang-tidy --source=../application/source/Sensor/GPSBackend.cpp -- /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/x86_64-bikeossdk-linux/usr/bin/arm-bikeos-linux-gnueabi/arm-bikeos-linux-gnueabi-clang++  --target=arm-bikeos-linux-gnueabi  -march=armv6 -mfpu=vfp -mfloat-abi=hard -mtune=arm1176jzf-s -mfpu=vfp -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -mlittle-endian -mlittle-endian --sysroot=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi --sysroot=/opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_LOCATION_LIB -DQT_NETWORK_LIB -DQT_OPENGL_LIB -DQT_POSITIONINGQUICK_LIB -DQT_POSITIONING_LIB -DQT_QMLMODELS_LIB -DQT_QML_LIB -DQT_QUICK_LIB -DQT_SENSORS_LIB -DQT_SQL_LIB -DQT_WIDGETS_LIB -Dbikeapp_sensor_EXPORTS -Iapplication/bikeapp_sensor_autogen/include -I../application/include -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroots/arm1176jzfshf-vfp-bikeos-linux-gnueabi/usr/include/QtQml -isystem /opt/bikeos/1.0.0u-git+8e234428-dbg/sysroot [... cut ...]
[build] /home/tobias/Repos/cycle-clever/bike-app/build/../application/source/Sensor/GPSBackend.cpp:34:5: warning: do not use 'else' after 'throw' [readability-else-after-return]
[build]   } else {
[build]     ^~~~~~
[build] In file included from ../application/source/Sensor/GPSBackend.cpp:1:
```
And it complains about my coding skills, as it is supposed to be ...

Maybe a bug, new use case or I missed a config setting by the meta-clang layer. But I want to share it as it cost me some time.

Regards Tobias

kraj, 2021-01-18:
this patch looks fine to me. Can you submit a pull against master ?

kraj, 2021-01-25:
this is now applied

closed 2021-01-25 by kraj