kraj/meta-clang #419: Loop dependency for using clang with riscv

drcut, 2021-01-18:
**Describe the bug**
When I add meta-clang into meta-riscv, it shows loop dependency

**To Reproduce**
Steps to reproduce the behavior:
(1) build meta-riscv according to the README
(2)git clone git://github.com/kraj/meta-clang.git
(3) add ```bitbake-layers add-layer ../meta-clang``` in meta-riscv/setup.sh
(4) add ```TOOLCHAIN ?= "clang"``` and ```RUNTIME = "llvm"``` in conf/local.conf
(5) MACHINE=qemuriscv32 bitbake core-image-full-cmdline
And here is the log for step(5)
```bash
Loading cache...done.
Loaded 3585 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv32-oe-linux"
MACHINE              = "qemuriscv32"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv32"
meta                 = "work:23cb39a5fa2a55681e7bc2605f435135cec9173b"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "work:a6a8fc75bad04be0842e22fb137d0e7a2dcb7279"
meta-riscv           = "work:9d10fd4f37d9f209847e79953fcc9bc018b1fe5e"
meta-clang           = "master:e2d2a029ee122b9cf25f8296c6ce610bf6bebbf6"

Initialising tasks...ERROR: 583 unbuildable tasks were found.
These are usually caused by circular dependencies and any circular dependency chains found will be printed below. Increase the debug level to see a list of unbuildable tasks.

Identifying dependency loops (this may take a short while)...

ERROR: 
Dependency loop #1 found:
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_package (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_install', 'dwarfsrcfiles.bb:do_populate_sysroot', 'glibc_2.32.bb:do_packagedata', 'libxcrypt_4.4.17.bb:do_packagedata', 'gcc-runtime_10.2.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_packagedata (dependent Tasks ['compiler-rt_git.bb:do_package'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_package (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_install', 'dwarfsrcfiles.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_packagedata', 'glibc_2.32.bb:do_packagedata', 'gcc-runtime_10.2.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_packagedata (dependent Tasks ['libcxx_git.bb:do_package'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_package (dependent Tasks ['libgcc_10.2.bb:do_packagedata', 'pseudo_git.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_install', 'libcxx_git.bb:do_packagedata', 'dwarfsrcfiles.bb:do_populate_sysroot', 'glibc_2.32.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_packagedata (dependent Tasks ['libxcrypt_4.4.17.bb:do_package'])

Dependency loop #2 found:
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_compile (dependent Tasks ['compiler-rt_git.bb:do_configure'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_compile'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_install'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_prepare_recipe_sysroot (dependent Tasks ['clang-cross_git.bb:do_populate_sysroot', 'cmake-native_3.19.2.bb:do_populate_sysroot', 'python3_3.9.1.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_fetch', 'gcc-runtime_10.2.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_populate_sysroot', 'ninja_1.10.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_configure (dependent Tasks ['libcxx_git.bb:do_generate_toolchain_file', 'libcxx_git.bb:do_prepare_recipe_sysroot', 'llvm-project-source.bb:do_patch'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_compile (dependent Tasks ['libcxx_git.bb:do_configure'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_compile'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.bb:do_populate_sysroot', 'libcxx_git.bb:do_install'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_prepare_recipe_sysroot (dependent Tasks ['clang-cross_git.bb:do_populate_sysroot', 'pkgconfig_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_fetch', 'gnu-config_git.bb:do_populate_sysroot', 'libgcc_10.2.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot', 'libtool-cross_2.4.6.bb:do_populate_sysroot', 'libtool-native_2.4.6.bb:do_populate_sysroot', 'automake_1.16.2.bb:do_populate_sysroot', 'autoconf_2.69.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_configure (dependent Tasks ['libxcrypt_4.4.17.bb:do_patch', 'libxcrypt_4.4.17.bb:do_prepare_recipe_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_compile (dependent Tasks ['libxcrypt_4.4.17.bb:do_configure'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_compile'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_install'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_prepare_recipe_sysroot (dependent Tasks ['clang-cross_git.bb:do_populate_sysroot', 'pkgconfig_git.bb:do_populate_sysroot', 'cmake-native_3.19.2.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_fetch', 'python3_3.9.1.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'gcc-runtime_10.2.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_populate_sysroot', 'ninja_1.10.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure (dependent Tasks ['compiler-rt_git.bb:do_prepare_recipe_sysroot', 'compiler-rt_git.bb:do_generate_toolchain_file', 'llvm-project-source.bb:do_patch'])

Dependency loop #3 found:
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_compile (dependent Tasks ['compiler-rt_git.bb:do_configure'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_compile'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_install'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_prepare_recipe_sysroot (dependent Tasks ['clang-cross_git.bb:do_populate_sysroot', 'cmake-native_3.19.2.bb:do_populate_sysroot', 'python3_3.9.1.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_fetch', 'gcc-runtime_10.2.bb:do_popula [... cut ...] ibc_2.32.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_populate_sysroot', 'ninja_1.10.2.bb:do_populate_sysroot'])
  Task /home/SENSETIME/hanruobing/tmp/riscv-yocto/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure (dependent Tasks ['compiler-rt_git.bb:do_prepare_recipe_sysroot', 'compiler-rt_git.bb:do_generate_toolchain_file', 'llvm-project-source.bb:do_patch'])

ERROR: Command execution failed: 1

Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

kraj, 2021-01-18:
it seems to be a bug when using `RUNTIME = "llvm" ` I would suggest to not set that and see if it works.

drcut, 2021-01-19:
> it seems to be a bug when using `RUNTIME = "llvm" ` I would suggest to not set that and see if it works.

Thanks, after removing it, it can run. However, I can not find clang/llvm after login. Do I have to something else?

kraj, 2021-01-19:
> > it seems to be a bug when using `RUNTIME = "llvm" ` I would suggest to not set that and see if it works.
> 
> Thanks, after removing it, it can run. However, I can not find clang/llvm after login. Do I have to something else?

you need to add

```
EXTRA_IMAGE_FEATURES_append = " tools-sdk dev-pkgs"
```
in `local.conf`

drcut, 2021-01-19:
It works! Thanks

closed 2021-01-19 by drcut