kraj/meta-clang #425: RUNTIME='llvm' seems to fail due to dependency loops?

MastaG, 2021-01-27:
**Describe the bug**
When I set RUNTIME='llvm' it seems to fail.
```
Build Configuration:
BB_VERSION           = "1.49.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "fedora-33"
TARGET_SYS           = "arm-oe-linux-gnueabi"
MACHINE              = "odroid-xu3"
DISTRO               = "odroid-os"
DISTRO_VERSION       = "release"
TUNE_FEATURES        = "arm vfp cortexa15 neon vfpv4 callconvention-hard"
TARGET_FPU           = "hard"
meta-oe              
meta-filesystems     
meta-multimedia      
meta-networking      
meta-python          
meta-webserver       
meta-perl            
meta-initramfs       = "master:4877dd19c05fc972ce51ebb39c28ec460ca0c3aa"
meta                 = "master:1140ca0090eb235cad3bc1427703dea43429d9de"
meta-rust            = "master:c72b2dda3a4f70ed257c7de9bedb4b04732970a4"
odroid-os            = "master:ddba66db96cff248d7d86533d0df8dc854c4eb39"
meta-odroid          = "master:61b4b767a8be87913ad28cd491f1dad40ab29b61"
meta-browser         = "master:d9b0cbeaf4280ff325a15441c701ea1f4a140b17"
meta-local           = "master:ddba66db96cff248d7d86533d0df8dc854c4eb39"
meta-qt5             = "master:7520e38d187cba6685848e12b3203f805c8f31a3"
meta-selinux         = "master:73de14d8650c8f8d6b6009c04d131cfd0d873826"
meta-python2         = "master:c43c29e57f16af4e77441b201855321fbd546661"
meta-clang           = "master:67a7cad4bb48ef569b4eeb4b3c923133d2d422a6"

DEBUG: Added 12632 recursive dependencies in this loop                                                                                                                                                                                                                                                                                                                                                                                                                         | ETA:  0:00:03
DEBUG: Added 34 recursive dependencies in this loop
DEBUG: Added 0 recursive dependencies in this loop
NOTE: Marking Active Tasks#########################################################################################################################################################################################                                                                                                                                                                                                                                                            | ETA:  0:00:00
NOTE: Pruned 5758 inactive tasks, 2990 left#########################################################################################################################################################################                                                                                                                                                                                                                                                           | ETA:  0:00:00
NOTE: Assign Weightings
NOTE: Compute totals (have 1 endpoint(s))############################################################################################################################################################################                                                                                                                                                                                                                                                          | ETA:  0:00:00
ERROR: 583 unbuildable tasks were found.##############################################################################################################################################################################                                                                                                                                                                                                                                                         | ETA:  0:00:00
These are usually caused by circular dependencies and any circular dependency chains found will be printed below. Increase the debug level to see a list of unbuildable tasks.

Identifying dependency loops (this may take a short while)...

ERROR: 
Dependency loop #1 found:
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_package (dependent Tasks ['glibc_2.32.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot', 'gcc-runtime_10.2.bb:do_packagedata', 'pseudo_git.bb:do_populate_sysroot', 'libxcrypt_4.4.17.bb:do_packagedata', 'dwarfsrcfiles.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_install'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_packagedata (dependent Tasks ['compiler-rt_git.bb:do_package'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_package (dependent Tasks ['glibc_2.32.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot', 'dwarfsrcfiles.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_packagedata', 'gcc-runtime_10.2.bb:do_packagedata', 'pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_install'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_packagedata (dependent Tasks ['libcxx_git.bb:do_package'])
  Task /home/mastag/src/odroid-oe-core/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_package (dependent Tasks ['glibc_2.32.bb:do_packagedata', 'rpm_4.16.1.2.bb:do_populate_sysroot', 'libgcc_10.2.bb:do_packagedata', 'pseudo_git.bb:do_populate_sysroot', 'dwarfsrcfiles.bb:do_populate_sysroot', 'libcxx_git.bb:do_packagedata', 'libxcrypt_4.4.17.bb:do_install'])
  Task /home/mastag/src/odroid-oe-core/openembedded-core/meta/recipes-core/libxcrypt/libxcrypt_4.4.17.bb:do_packagedata (dependent Tasks ['libxcrypt_4.4.17.bb:do_package'])

Dependency loop #2 found:
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_compile (dependent Tasks ['compiler-rt_git.bb:do_configure'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_compile'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.1.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_install'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_prepare_recipe_sysroot (dependent Tasks ['python3_3.9.1.bb:do_populate_sysroot', 'compiler-rt_git.bb:do_populate_sysroot', 'clang-cross_git.bb:do_populate_sysroot', 'gcc-runtime_10.2.bb:do_populate_sysroot', 'cmake-native_3.19.2.bb:do_populate_sysroot', 'ninja_1.10.2.bb:do_populate_sysroot', 'libcxx_git.bb:do_fetch', 'pseudo_git.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_configure (dependent Tasks ['llvm-project-source.bb:do_patch', 'libcxx_git.bb:do_prepare_recipe_sysroot', 'libcxx_git.bb:do_generate_toolchain_file'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_compile (dependent Tasks ['libcxx_git.bb:do_configure'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'libcxx_git.bb:do_compile'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_populate_sysroot (dependent Tasks ['binutils-cross_2.35.1.bb:do_populate_sysroot', 'libcxx_git.bb:do_install'])
  Task /home/mastag/src/odroid-oe-core/openembedded-core/meta/recipes-devtools/libtool/libtool-cross_2.4.6.bb:do_prepare_recipe_sysroot (dependent Tasks ['clang-cross_git.bb:do_populate_sysroot', 'libtool-cross_2.4.6.bb:do_fetch', 'autoconf_2.69.bb:do_populate_sysroot', 'gnu-config_git.bb:do_populate_sysroot', 'automake_1.16.2.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot', 'libcxx_git.bb:do_populate_sysroot', 'libgcc_10.2.bb:do_populate_sysroot', 'libtool-native_2.4.6.bb:do_populate_sysroot', 'texinfo-dummy-native.bb:do_populate_sysroot'])
  Task /home/mastag/src/odroid-oe-core/openembedded-core/meta/recipes-devtools/libtool/libtool-cross_2.4.6.bb:do_configure (dependent Tasks ['libtool-cross_2.4.6.bb:do_prepare_recipe_sysroot', 'libtool-cross_2.4.6.bb:do_patch'])
[... cut ...] ', 'ninja_1.10.2.bb:do_populate_sysroot', 'pkgconfig_git.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'glibc_2.32.bb:do_populate_sysroot'])
  Task /home/mastag/src/odroid-oe-core/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure (dependent Tasks ['compiler-rt_git.bb:do_generate_toolchain_file', 'compiler-rt_git.bb:do_prepare_recipe_sysroot', 'llvm-project-source.bb:do_patch'])

ERROR: Command execution failed: 1
```

Any ideas?
RUNTIME='gnu' seems to work though.

kraj, 2021-01-28:
its because of `virtual/crypt` in compiler-rt DEPENDS,  this was earlier provided by glibc but has now moved to be a independent package called libxcrypt providing it. So I think we perhaps need to disable sanitizers in compiler-rt or separate it out into own  recipe.

MastaG, 2021-01-29:
Ah I see, I'll try your commit then to disable sanitizer builds then.

commit 04a1194c (2021-02-01): compiler-rt: Disable sanitizer builds

Sanitizers include crypt.h from libcrypt to build, so when using llvm runtime this
becomes important to break the cyclic dependency

compiler-rt->libxcrypt->compiler-rt

Fixes #425
closed 2021-02-01 by kraj