kraj/meta-clang #428: cmake_do_configure error with clang

dagmcr, 2021-02-11:
Hi, 
I'm trying to compile AMD ROCm with meta-clang as it is llvm required for the project to compile.

I first started with this: https://github.com/RadeonOpenCompute/ROCm-Device-Libs but when I add meta-clang layer and `TOOLCHAIN = "clang"` to the recipe it fails with a lot of these:

```
| CMake Error at cmake/OCL.cmake:119 (add_custom_command):                                                                                                                                                                                                                        
|   Error evaluating generator expression:                                                                                                                                                                                                                                        
|                                                                                                                                                                                                                                                                                 
|     $<TARGET_FILE:llvm-link>  
| CMake Error at cmake/OCL.cmake:119 (add_custom_command):
|   Error evaluating generator expression:
| 
|     $<TARGET_FILE:opt>
| 
|   No target "opt"
| Call Stack (most recent call first):
|   hc/CMakeLists.txt:21 (opencl_bc_lib)
| 
| 
| CMake Error at cmake/OCL.cmake:89 (add_custom_command):
|   Error evaluating generator expression:
| 
|     $<TARGET_FILE:clang>
| 
|   No target "clang"
```

For reference, recipe now does not do anything special:

```
SUMMARY = "Radeon Open Compute - ROCm Device Libraries"                                                                                                                                                                                                                           
DESCRIPTION = "ROCm Device Libraries for AMD ROCm"                                                                                                                                                                                                                                
HOMEPAGE = "https://github.com/RadeonOpenCompute/ROCm-Device-Libs"                                                                                                                                                                                                                
BUGTRACKER = "https://github.com/RadeonOpenCompute/ROCm-Device-Libs/issues"                                                                                                                                                                                                       
UPSTREAM_CHECK_URI = "https://github.com/RadeonOpenCompute/ROCm-Device-Libs/releases"                                                                                                                                                                                             
                                                                                                                                                                                                                                                                                  
SECTION = "graphics"                                                                                                                                                                                                                                                              
LICENSE = "NCSA"                                                                                                                                                                                                                                                                  
                                                                                                                                                                                                                                                                                  
LIC_FILES_CHKSUM = "file://LICENSE;md5=356efe91bb49e4daf4ce7e734b4b93a2"                                                                                                                                                                                                          
                                                                                                                                                                                                                                                                                  
SRC_URI = "git://github.com/RadeonOpenCompute/ROCm-Device-Libs;protocol=https;branch=amd-stg-open"                                                                                                                                                                                
SRCREV = "e778e213fc0e8aba156eb66ae84b63bee3b10d7b"                                                                                                                                                                                                                               
SRC_URI[sha256sum] = "1536a1a9773dcd74489ff7ada58757757bd2a9470602661e59ed36be164f80fe"                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                  
inherit cmake                                                                                                                                                                                                                                              

PV = "rocm-4.0.0"                                                                                                                                                                                                                
                                                                                                                                                                                                                                                                                  
S = "${WORKDIR}/git"                                                                                                                                                                                                                                                              
                                                                                                                                                                                                                                                                                  
TOOLCHAIN = "clang"
```

Right now, I'm using yocto lts 3.1.4 and `dunfell` branch in meta-clang. I've also tried to add `export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"` but it didn't help. 

Any clue what I'm doing wrong?

Thanks!

BTW, I had to cherry-pick https://github.com/kraj/meta-clang/commit/3427029851cd2654b1359ccbe7f111ff76655a81 because the expansion for gallium drivers in mesa was failing.

dagmcr, 2021-02-11:
I've now realized the `llvm-config` is installed here: `./rocm-4.0.0-r0/recipe-sysroot-native/usr/bin/llvm-config` so I've updated `YOCTO_ALTERNATE_EXE_PATH` to `export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR_NATIVE}/llvm-config"` but the result is the same.

Here the full log file:
[log.do_configure.982348.log](https://github.com/kraj/meta-clang/files/5968242/log.do_configure.982348.log)

kraj, 2021-03-25:
@dagmcr can this issue be closed for master ? or is there anything more needed

dagmcr, 2021-03-25:
@kraj let me try it today. I couldn’t fix it with master las time.

dagmcr, 2021-03-26:
Hi @kraj, I've tried with poky & meta-clang master branches but the result is the same:

```
Build Configuration:
BB_VERSION           = "1.49.2"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.2+snapshot-ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
meta-clang           = "master:404e8129c26ea5387a51248eebff1820db6f5015"
```

The final output is this:

```
| CMake Generate step failed.  Build files cannot be regenerated correctly.
| WARNING: /workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/temp/run.do_configure.3511820:170 exit 1 from 'cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/git -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -Wno-dev'
| WARNING: Backtrace (BB generated script):
|       #1: cmake_do_configure, /workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/temp/run.do_configure.3511820, line 170
|       #2: do_configure, /workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/temp/run.do_configure.3511820, line 145
|       #3: main, /workdir/build/tmp/work/core2-64-poky-linux/rocm-device-libs/rocm-4.0.0-r0/temp/run.do_configure.3511820, line 205
| 
| Backtrace (metadata-relative locations):
|       #1: cmake_do_configure, /workdir/layers/poky/meta/classes/cmake.bbclass, line 173
|       #2: do_configure, autogenerated, line 2
ERROR: Task (/workdir/layers/meta-clang/recipes-extended/rocm/rocm-device-libs_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 824 tasks of which 797 didn't need to be rerun and 1 failed.
```

And a bunch of these:

```
| CMake Error at cmake/OCL.cmake:89 (add_custom_command):                                                                                                                                                                                                                                                                      
|   Error evaluating generator expression:                                                                                                                                                                                                                                                                                     
|                                                                                                                                                                                                                                                                                                                              
|     $<TARGET_FILE:clang>                                                                                                                                                                                                                                                                                                     
|                                                                                                                                                                                                                                                                                                                              
|   No target "clang"                                                                                                                                                                                                                                                                                                          
| Call Stack (most recent call first):                                                                                                                                                                                                                                                                                         
|   hc/CMakeLists.txt:21 (opencl_bc_lib)  
```

> With no target for `clang`, `llvm-link`...

Also, I've added `DEPENDS += "clang"` as I could see you are doing it for `castxml_git.bb` + `TOOLCHAIN = "clang"` but it didn't work.

Looking at what others are doing (e.g. https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=rocm-device-libs) they are only specifying the `LLVM_DIR` and the install prefix.

The [rocm-device-libs project](https://github.com/RadeonOpenCompute/ROCm-Device-Libs) specifies that it depends on [amd-llvm](https://github.com/RadeonOpenCompute/llvm-project) but I initially considered that shouldn't be the problem.

Another thing I've tried was to point `CMAKE_PREFIX_PATH` to the `recipe-sysroot/usr/lib/cmake/llvm` but somehow the recipe cannot fin the targets.

kraj, 2021-03-27:
hmmm does setting `export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config"` help

kraj, 2021-03-27:
I see that it might need some more work to make this cross compile friendly, maybe use a prebuilt llvm-native tree and see if it can build using that build tree via `-DLLVM_DIR`

dagmcr, 2021-03-27:
> hmmm does setting export YOCTO_ALTERNATE_EXE_PATH = "${STAGING_LIBDIR}/llvm-config" help with master or no ?

It didn't help.

> I wonder if it needs llvm sources to be available and not just installed artifacts.

Maybe that's what I'm missing as the documentation always points to the [llvm build directory](https://rocmdocs.amd.com/en/latest/ROCm_Compiler_SDK/ROCm-Compiler-SDK.html#building) (and [here](https://github.com/RadeonOpenCompute/ROCm-Device-Libs#building)).

> I see that it might need some more work to make this cross compile friendly, maybe use a prebuilt llvm-native tree and see if it can build using that build tree via -DLLVM_DIR

I was looking to use `llvm-project-source.bb` and maybe compile it as llvm external project but I think it wont be the best approach.

I'll give it a shot to the `llvm-native`.

Thank you for the help @kraj.