kraj/meta-clang #445: SIG ILL in systemd 247.3+

tewarid, 2021-03-23:
**Describe the bug**
I've setup a working yocto distro build for Raspberry Pi Zero W using Docker at https://github.com/tewarid/docker-meta-raspberrypi/tree/clang, that uses meta-clang b8ef5c875c789e1e704d22aca80ba05e9583b9bf, but it fails to boot due to SIG ILL in systemd. System boots when using `TOOLCHAIN_pn-systemd = "gcc"` to build systemd recipe.

**To Reproduce**
Please see instructions to build the system and write image at https://github.com/tewarid/docker-meta-raspberrypi/blob/clang/README.md.

**Expected behavior**
No crashes in systemd.

**Additional context**
System fails to boot, with log shown below captured after adding `debug ignore_loglevel` to kernel command line

```text
[    5.789416] systemd[1]: systemd 247.3+ running in system mode. (-PAM -AUDIT -SELINUX +IMA -APPARMOR -SMACK +SYSVINIT +UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -ZSTD -SECCOMP +BLKID -ELFUTILS +KMOD -IDN2 -IDN -PCRE2 default-hierarchy=hybrid)
[    5.827435] systemd[1]: No virtualization found in DMI
[    5.837248] systemd[1]: UML virtualization not found in /proc/cpuinfo.
[    5.848169] systemd[1]: No virtualization found in CPUID
[    5.857558] systemd[1]: Virtualization XEN not found, /proc/xen does not exist
[    5.869380] systemd[1]: No virtualization found in /proc/device-tree/*
[    5.879999] systemd[1]: This platform does not support /proc/sysinfo
[    5.890266] systemd[1]: Found VM virtualization none
[    5.898965] systemd[1]: Detected architecture arm.
[    5.908699] systemd[1]: Detected initialized system, this is not the first boot.
[    5.924465] systemd[1]: Mounting cgroup to /sys/fs/cgroup/pids of type cgroup with options pids.
[    5.941462] systemd[1]: Mounting cgroup to /sys/fs/cgroup/devices of type cgroup with options devices.
[    5.959013] systemd[1]: Mounting cgroup to /sys/fs/cgroup/freezer of type cgroup with options freezer.
[    5.976612] systemd[1]: Mounting cgroup to /sys/fs/cgroup/perf_event of type cgroup with options perf_event.
[    5.994971] systemd[1]: Mounting cgroup to /sys/fs/cgroup/blkio of type cgroup with options blkio.
[    6.012458] systemd[1]: Mounting cgroup to /sys/fs/cgroup/net_cls,net_prio of type cgroup with options net_cls,net_prio.
[    6.032158] systemd[1]: Mounting cgroup to /sys/fs/cgroup/cpu,cpuacct of type cgroup with options cpu,cpuacct.
[    6.073822] systemd[1]: Set hostname to <raspberrypi0-wifi>.
[    6.097428] random: systemd: uninitialized urandom read (16 bytes read)
[    6.108213] systemd[1]: Initializing machine ID from random generator.
[    6.119229] systemd[1]: Bind-mounting /run/machine-id on /etc/machine-id (MS_BIND "")...
[    6.135109] systemd[1]: Installed transient /etc/machine-id file.
[    6.145179] systemd[1]: Remounting /etc/machine-id (MS_RDONLY|MS_REMOUNT|MS_BIND "")...
[    6.167873] systemd[1]: Successfully added address 127.0.0.1 to loopback interface
[    6.183050] systemd[1]: Successfully added address ::1 to loopback interface
[    6.194191] systemd[1]: Successfully brought loopback interface up
[    6.205176] systemd[1]: Setting 'fs/file-max' to '2147483647'.
[    6.215278] systemd[1]: Setting 'fs/nr_open' to '2147483644'.
[    6.224937] systemd[1]: Couldn't write fs.nr_open as 2147483644, halving it.
[    6.235983] systemd[1]: Setting 'fs/nr_open' to '1073741820'.
[    6.245468] systemd[1]: Couldn't write fs.nr_open as 1073741820, halving it.
[    6.256475] systemd[1]: Setting 'fs/nr_open' to '536870908'.
[    6.265902] systemd[1]: Successfully bumped fs.nr_open to 536870908
[    6.276543] systemd[1]: Found cgroup2 on /sys/fs/cgroup/unified, unified hierarchy for systemd controller
[    6.293502] systemd[1]: Failed to read memory.limit_in_bytes cgroup attribute, ignoring cgroup memory limit: No such file or directory
[    6.314964] systemd[1]: Found cgroup2 on /sys/fs/cgroup/unified, unified hierarchy for systemd controller
[    6.331554] systemd[1]: Unified cgroup hierarchy is located at /sys/fs/cgroup/unified. Controllers are on legacy hierarchies.
[    6.369931] systemd[1]: Got EBADF when using BPF_F_ALLOW_MULTI, which indicates it is supported. Yay!
[    6.386886] systemd[1]: Controller 'cpu' supported: yes
[    6.404998] systemd[1]: Controller 'cpuacct' supported: yes
[    6.414144] systemd[1]: Controller 'cpuset' supported: no
[    6.423070] systemd[1]: Controller 'io' supported: no
[    6.431563] systemd[1]: Controller 'blkio' supported: yes
[    6.440317] systemd[1]: Controller 'memory' supported: no
[    6.449065] systemd[1]: Controller 'devices' supported: yes
[    6.457885] systemd[1]: Controller 'pids' supported: yes
[    6.466371] systemd[1]: Controller 'bpf-firewall' supported: yes
[    6.475524] systemd[1]: Controller 'bpf-devices' supported: yes
[    6.484752] systemd[1]: Set up TFD_TIMER_CANCEL_ON_SET timerfd.
[    6.493980] systemd[1]: Failed to stat /etc/localtime, ignoring: No such file or directory
[    6.508908] systemd[1]: /etc/localtime doesn't exist yet, watching /etc instead.
[    6.525171] systemd[1]: Enabling (yes) showing of status (commandline).
[    6.537596] systemd[1]: Successfully forked off '(sd-executor)' as PID 59.
[    6.555852] systemd[59]: Successfully forked off '(direxec)' as PID 60.
[    6.578027] systemd[59]: Successfully forked off '(direxec)' as PID 61.
[    6.618563] systemd[59]: Successfully forked off '(direxec)' as PID 62.
[    6.658659] systemd[59]: Successfully forked off '(direxec)' as PID 63.
[... 21 lines cut ...]
[    7.278177] systemd[59]: /lib/systemd/system-generators/systemd-sysv-generator terminated by signal ILL.
[    7.300213] systemd[59]: /lib/systemd/system-generators/systemd-fstab-generator succeeded.
[    7.317050] systemd[59]: /lib/systemd/system-generators/systemd-gpt-auto-generator terminated by signal ILL.
[    7.335462] systemd[59]: /lib/systemd/system-generators/systemd-run-generator succeeded.
[    7.352096] systemd[59]: /lib/systemd/system-generators/systemd-getty-generator succeeded.
[    7.370665] systemd[1]: (sd-executor) succeeded.
[    7.379975] systemd[1]: Looking for unit files in (higher priority first):
[    7.391150] systemd[1]: 	/etc/systemd/system.control
[    7.400205] systemd[1]: 	/run/systemd/system.control
[    7.409150] systemd[1]: 	/run/systemd/transient
[    7.417463] systemd[1]: 	/run/systemd/generator.early
[    7.426065] systemd[1]: 	/etc/systemd/system
[    7.433796] systemd[1]: 	/etc/systemd/system.attached
[    7.442169] systemd[1]: 	/run/systemd/system
[    7.449650] systemd[1]: 	/run/systemd/system.attached
[    7.457724] systemd[1]: 	/run/systemd/generator
[    7.465130] systemd[1]: 	/usr/local/lib/systemd/system
[    7.473069] systemd[1]: 	/lib/systemd/system
[    7.480056] systemd[1]: 	/usr/lib/systemd/system
[    7.487274] systemd[1]: 	/run/systemd/generator.late
[    7.513790] systemd[1]: Caught <ILL>, core dump failed (child 69, code=killed, status=4/ILL).
[    7.527276] systemd[1]: Freezing execution.
```

Thanks.

kraj, 2021-03-23:
pi0 is armv6 and it could fail in interesting ways where it might generate a v7 instructions perhaps. Can you post compiler commandline used while using clang to compile systemd ?

tewarid, 2021-03-23:
> pi0 is armv6 and it could fail in interesting ways where it might generate a v7 instructions perhaps. Can you post compiler commandline used while using clang to compile systemd ?

`[671/1452] arm-poky-linux-gnueabi-clang -target arm-poky-linux-gnueabi -march=armv6 -mfpu=vfp -mfloat-abi=hard -mtune=arm1176jzf-s -mfpu=vfp -rtlib=compiler-rt --unwindlib=libgcc --stdlib=libc++ -mlittle-endian -Qunused-arguments -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/pi/docker-meta-raspberrypi/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/systemd/1_247.3-r0/recipe-sysroot  -o systemd systemd.p/src_core_main.c.o -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-z,now -fstack-protector -Wl,--gc-sections -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,relro,-z,now '-Wl,-rpath,$ORIGIN/src/shared' -Wl,-rpath-link,/home/pi/docker-meta-raspberrypi/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/systemd/1_247.3-r0/build/src/shared -Wl,--start-group src/core/libcore.a src/core/libcore-shared.a src/shared/libsystemd-shared-247.so -pthread -lrt /home/pi/docker-meta-raspberrypi/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/systemd/1_247.3-r0/recipe-sysroot/usr/lib/libmount.so /home/pi/docker-meta-raspberrypi/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/systemd/1_247.3-r0/recipe-sysroot/usr/lib/libblkid.so /home/pi/docker-meta-raspberrypi/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/systemd/1_247.3-r0/recipe-sysroot/usr/lib/libkmod.so -lacl -Wl,--end-group`

Thanks.

kraj, 2021-03-24:
Can you try with the following patch applied to meta-raspberrypi ?

https://github.com/YoeDistro/meta-raspberrypi/commit/27d430f51de16456d4dc66ff61dc58febab87d06

kraj, 2021-03-24:
another test please try to build systemd with libgcc in local.conf use
```
COMPILER_RT_pn-systemd_toolchain-clang = "--rtlib=libgcc"
```

tewarid, 2021-03-24:
> another test please try to build systemd with libgcc in local.conf use
> 
> ```
> COMPILER_RT_pn-systemd_toolchain-clang = "--rtlib=libgcc"
> ```

This fixed the issue.

tewarid, 2021-03-24:
> Can you try with the top commit from the following branch applied to meta-raspberrypi ?
> https://github.com/YoeDistro/meta-raspberrypi/commits/kraj/pi0-tune
> 
> and apply the top patch from https://github.com/YoeDistro/meta-clang/commits/yoe/mut
> to meta-clang

These patches also fix the issue.

kraj, 2021-03-25:
this is now fixed. Please use latest master or both meta-clang and meta-raspberrypi

closed 2021-03-25 by kraj