kraj/meta-clang #448: ERROR: Function failed: do_configure

yong-ruilin, 2021-03-25:
hi,I encountered this problem when using the Sumo branch.

Log data follows:
| DEBUG: Executing shell function do_configure
| -- The C compiler identification is Clang 6.0.1
| -- The CXX compiler identification is Clang 6.0.1
| -- The ASM compiler identification is unknown
| -- Found assembler: /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/mips64el-loongos-linux/mips64el-loongos-linux-clang
| -- Check for working C compiler: /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/mips64el-loongos-linux/mips64el-loongos-linux-clang
| -- Check for working C compiler: /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/mips64el-loongos-linux/mips64el-loongos-linux-clang -- broken
| CMake Error at /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/share/cmake-3.10/Modules/CMakeTestCCompiler.cmake:52 (message):
|   The C compiler
| 
|     "/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/mips64el-loongos-linux/mips64el-loongos-linux-clang"
| 
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|     Change Dir: /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/build/CMakeFiles/CMakeTmp
| 
|     Run Build Command:"ninja" "cmTC_4864c"
|     [1/2] Building C object CMakeFiles/cmTC_4864c.dir/testCCompiler.c.o
|     FAILED: CMakeFiles/cmTC_4864c.dir/testCCompiler.c.o
|     /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/mips64el-loongos-linux/mips64el-loongos-linux-clang   -mabi=64 -mhard-float -march=mips64r2 -mtune=mips64r2 -mlittle-endian -D__extern_always_inline=inline -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0=/usr/src/debug/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0 -fdebug-prefix-map=/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot= -fdebug-prefix-map=/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native=  -mabi=64 -mhard-float -march=mips64r2 -mtune=mips64r2 -mlittle-endian -D__extern_always_inline=inline -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -o CMakeFiles/cmTC_4864c.dir/testCCompiler.c.o   -c testCCompiler.c
|     /tmp/testCCompiler-102867.s: Assembler messages:
|     /tmp/testCCompiler-102867.s:38: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:38: Warning: entity size for SHF_MERGE not specified
|     /tmp/testCCompiler-102867.s:38: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:55: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:55: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:67: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:67: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:161: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:161: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:214: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:214: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:215: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:215: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:218: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:218: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:228: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:228: Error: junk at end of line, first unrecognized character is `0'
|     /tmp/testCCompiler-102867.s:244: Warning: unrecognized section type
|     /tmp/testCCompiler-102867.s:244: Error: junk at end of line, first unrecognized character is `0'
|     clang-6.0: error: assembler command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.
| 
| 
| 
| 
|   CMake will not be able to correctly generate this project.
| Call Stack (most recent call first):
|   CMakeLists.txt:10 (project)
| 
| 
| -- Configuring incomplete, errors occurred!
| See also "/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/build/CMakeFiles/CMakeOutput.log".
| See also "/home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/build/CMakeFiles/CMakeError.log".
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/ubuntu/work/poky-test/poky/build/tmp-glibc/work/gs464e-loongos-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/temp/log.do_configure.101617)
ERROR: Task (/home/ubuntu/work/poky-test/poky/meta-openembedded/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 539 tasks of which 523 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/ubuntu/work/poky-test/poky/meta-openembedded/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_configure
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

kraj, 2021-03-25:
@yong-ruilin sumo is EOL for a long time and you are using mips64 port of compiler-rt, even today mips64 is not tested as much as other arches so I expect these kind of errors. Your best bet is to try one of the supported releases ( dunfell or newer ) of yocto and see if it still happens then we can do something

yong-ruilin, 2021-03-26:
Thanks, I will try to change the branch

closed 2021-03-26 by yong-ruilin