kraj/meta-clang #456: version conflict between oe-core llvm and meta-clang's clang

yifan19, 2021-04-15:
**Describe the bug**
```
/yow-lpggp32/yyu1/yocto/b_poky/build$ bitbake -s | grep llvm
WARNING: preferred version 12.0.0 of nativesdk-llvm not available
WARNING: versions of nativesdk-llvm available: 11.1.0
WARNING: preferred version 12.0.0 of llvm-native not available
WARNING: versions of llvm-native available: 11.1.0
WARNING: preferred version 12.0.0 of llvm not available
WARNING: versions of llvm available: 11.1.0
llvm                                               :11.1.0-r0
llvm-native                                        :11.1.0-r0
llvm-project-source-12.0.0                         :12.0.0-r0
nativesdk-llvm                                     :11.1.0-r0
nativesdk-spirv-llvm-translator                    :12.0.0-r0
spirv-llvm-translator                              :12.0.0-r0
spirv-llvm-translator-native                       :12.0.0-r0
```

**To Reproduce**
Steps to reproduce the behavior:
use poky+meta-clang
bitbake llvm

**Expected behavior**
I think there shouldn't be any warnings since clang provides llvm

**Additional context**
the current windriver fix. I find it really confusing.
```
diff --git a/conf/layer.conf b/conf/layer.conf
index 451bd6a..d427532 100644
--- a/conf/layer.conf
+++ b/conf/layer.conf
@@ -29,7 +29,11 @@ INHERIT += "clang"
 # Do not include clang in SDK unless user wants to
 CLANGSDK ??= "0"
 
-LLVMVERSION = "12.0.0"
+LLVMVERSION_clang-llvm = "11.0.1"
+
+PREFERRED_VERSION_llvm_clang-llvm = ""
+PREFERRED_VERSION_llvm-native_clang-llvm = ""
+PREFERRED_VERSION_nativesdk-llvm_clang-llvm = ""
 
 require conf/nonclangable.conf
 require conf/nonscanable.conf
```

kraj, 2021-05-28:
I dont get this conflict, so I am at a loss on how this can be re-creared

zboszor, 2021-09-29:
I have seen the same warning with oe-core/meta-clang master with using meta-security and building clamav from that layer which has
```
DEPENDS = "... llvm ..."
```
LLVM in oe-core is still version 12 and meta-clang master moved to LLVM 13.
`PREFERRED_PROVIDER_llvm[-native] = "clang[-native]` in meta-clang works so it's clang that gets built and not the llvm recipe from oe-core. The warning is purely an annoyance.