kraj/meta-clang #463: compiler-rt configure failing

jwinarske, 2021-05-07:
**Describe the bug**

configure fails

`
cmake -G 'Ninja' -DCMAKE_MAKE_PROGRAM=ninja $oecmake_sitefile /b/agl_workspace/build/tmp/work/corei7-64-agl-linux/compiler-rt/11.0.0-r0/git/compiler-rt -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_INSTALL_BINDIR:PATH=bin -DCMAKE_INSTALL_SBINDIR:PATH=sbin -DCMAKE_INSTALL_LIBEXECDIR:PATH=libexec -DCMAKE_INSTALL_SYSCONFDIR:PATH=/etc -DCMAKE_INSTALL_SHAREDSTATEDIR:PATH=../com -DCMAKE_INSTALL_LOCALSTATEDIR:PATH=/var -DCMAKE_INSTALL_LIBDIR:PATH=lib -DCMAKE_INSTALL_INCLUDEDIR:PATH=include -DCMAKE_INSTALL_DATAROOTDIR:PATH=share -DPYTHON_EXECUTABLE:PATH=/usr/bin/python3 -DPython_EXECUTABLE:PATH=/usr/bin/python3 -DPython3_EXECUTABLE:PATH=/usr/bin/python3 -DLIB_SUFFIX= -DCMAKE_INSTALL_SO_NO_EXE=0 -DCMAKE_TOOLCHAIN_FILE=/b/agl_workspace/build/tmp/work/corei7-64-agl-linux/compiler-rt/11.0.0-r0/toolchain.cmake -DCMAKE_NO_SYSTEM_FROM_IMPORTED=1 -DCOMPILER_RT_DEFAULT_TARGET_ONLY=ON -DCOMPILER_RT_BUILD_BUILTINS=ON -DCOMPILER_RT_BUILD_CRT=ON -DCOMPILER_RT_BUILD_SANITIZERS=OFF -DCOMPILER_RT_BUILD_XRAY=OFF -DCOMPILER_RT_BUILD_LIBFUZZER=OFF -DCOMPILER_RT_BUILD_PROFILE=OFF -Wno-dev
`

[Configure Log](https://gist.github.com/jwinarske/3be3d43cbec47e44d2bdaab946e0796c)

**To Reproduce**
build AGL jellyfish branch, machine target `qemux86-64`.

**Desktop (please complete the following information):**
 - OS: Fedora 33

kraj, 2021-05-07:
Thanks for report @jwinarske, which branch are you trying to build. Also check https://docs.yoctoproject.org/ref-manual/system-requirements.html#detailed-supported-distros for supported distros on given release.

kraj, 2021-05-07:
If you see this issue on master branch then please update here since then we have to fix it

kraj, 2021-05-28:
@jwinarske  if master is working ok then Can this be closed ?

jwinarske, 2021-05-29:
Absolutely.  Turns out it was my own recipe.  Sorry about that.  Problems was entirely resolved.

closed 2021-05-29 by jwinarske