kraj/meta-clang #466: bitbake -c scanview error on debian 10

dl9pf, 2021-05-23:
**Describe the bug**
When running bitbake -c scanview on a debian 10 host using 
Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.1"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "hardknott:05a8aad57ce250b124db16705acec557819905ae"
meta-clang           = "master:d797409435d3b0e9f2859992439989ff1d81e66d"

scanview does not start and errors out with:

Log data follows:
| DEBUG: Executing shell function do_scanview
| ================================================================
| Stating scan-view server at: http://192.168.2.6:8181
| Use Ctrl-C to exit
| ================================================================
| /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view:9: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
|   import imp
| Traceback (most recent call last):
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 150, in <module>
|     main()
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 147, in main
|     run(port, args, args.root)
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 74, in run
|     import ScanView
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/ScanView.py", line 29, in <module>
|     import Reporter
| ModuleNotFoundError: No module named 'Reporter'
| WARNING: Unable to detect that server started.
| WARNING: /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878:151 exit 1 from 'scan-view --host 192.168.2.6 --port 8181 --allow-all-hosts /home/dl9pf/YPDD_STATIC/build/tmp/static-scan/busybox/*/'
| WARNING: Backtrace (BB generated script):
|       #1: do_scanview, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 151
|       #2: main, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 164
| ERROR: Execution of '/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878' failed with exit code 1:
| /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view:9: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
|   import imp
| Traceback (most recent call last):
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 150, in <module>
|     main()
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 147, in main
|     run(port, args, args.root)
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 74, in run
|     import ScanView
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/ScanView.py", line 29, in <module>
|     import Reporter
| ModuleNotFoundError: No module named 'Reporter'
| WARNING: Unable to detect that server started.
| WARNING: /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878:151 exit 1 from 'scan-view --host 192.168.2.6 --port 8181 --allow-all-hosts /home/dl9pf/YPDD_STATIC/build/tmp/static-scan/busybox/*/'
| WARNING: Backtrace (BB generated script):
|       #1: do_scanview, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 151
|       #2: main, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 164
| 
| Backtrace (metadata-relative locations):
|       #1: do_scanview, /home/dl9pf/YPDD_STATIC/meta-clang/classes/scan-build.bbclass, line 32
ERROR: Task (/home/dl9pf/YPDD_STATIC/poky/meta/recipes-core/busybox/busybox_1.33.0.bb:do_scanview) failed with exit code '1'
NOTE: Tasks Summary: Attempted 614 tasks of which 613 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dl9pf/YPDD_STATIC/poky/meta/recipes-core/busybox/busybox_1.33.0.bb:do_scanview
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

[... 44 lines cut ...]
|     main()
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 147, in main
|     run(port, args, args.root)
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view", line 74, in run
|     import ScanView
|   File "/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/ScanView.py", line 29, in <module>
|     import Reporter
| ModuleNotFoundError: No module named 'Reporter'
| WARNING: Unable to detect that server started.
| WARNING: /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878:151 exit 1 from 'scan-view --host 192.168.2.6 --port 8181 --allow-all-hosts /home/dl9pf/YPDD_STATIC/build/tmp/static-scan/busybox/*/'
| WARNING: Backtrace (BB generated script):
|       #1: do_scanview, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 151
|       #2: main, /home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/temp/run.do_scanview.33878, line 164
| 
| Backtrace (metadata-relative locations):
|       #1: do_scanview, /home/dl9pf/YPDD_STATIC/meta-clang/classes/scan-build.bbclass, line 32
ERROR: Task (/home/dl9pf/YPDD_STATIC/poky/meta/recipes-core/busybox/busybox_1.33.0.bb:do_scanview) failed with exit code '1'
NOTE: Tasks Summary: Attempted 614 tasks of which 613 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dl9pf/YPDD_STATIC/poky/meta/recipes-core/busybox/busybox_1.33.0.bb:do_scanview
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

**Desktop (please complete the following information):**
 - OS: debian 10 buster

kraj, 2021-05-23:
I can not reproduce it on master branch. Can you post top 20 lines of 

/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view

dl9pf, 2021-05-23:
#!/usr/bin/env python3

from __future__ import print_function

"""The clang static analyzer results viewer.
"""
/home/dl9pf/YPDD_STATIC/build/tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/scan-view

import sys
import imp
import os
import posixpath
import threading
import time
try:
    from urllib.request import urlopen
except ImportError:
    from urllib2 import urlopen
import webbrowser

# How long to wait for server to start.

...Reporter.py seems missing:
gandalf:/home/dl9pf/YPDD_STATIC/build# ls -alh tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/
tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/:
total 52K
drwxr-xr-x  3 dl9pf dl9pf 4.0K May 19 12:23 .
drwxr-xr-x 23 dl9pf dl9pf 4.0K May 19 12:15 ..
-rw-r--r-- 84 dl9pf dl9pf  318 May 19 11:48 bugcatcher.ico
drwxr-xr-x  2 dl9pf dl9pf 4.0K May 19 12:23 __pycache__
-rw-r--r-- 84 dl9pf dl9pf  26K May 19 11:48 ScanView.py
-rw-r--r-- 84 dl9pf dl9pf 6.0K May 19 11:48 startfile.py

kraj, 2021-05-23:
For tests can you copy ./clang/tools/scan-view/share/Reporter.py from clang sources into tmp/work/core2-64-poky-linux/busybox/1.33.0-r0/recipe-sysroot-native/usr/bin/../share/scan-view/ and see if that fixes it ?

kraj, 2021-05-23:
I think the issue is that you need this patch applied to clang

https://github.com/llvm/llvm-project/commit/3263c81589eca689341ab5084723bdb7fe4a1286

Can you test it ?

if it works please submit a pull too.

kraj, 2021-05-28:
its merged now

closed 2021-05-28 by kraj