kraj/meta-clang #484: compiler-rt installed-vs-shipped

blithe2015, 2021-07-26:
**compiler-rt install path error**
   /usr/lib/linux/*
  Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
  compiler-rt: 41 installed and not shipped files. [installed-vs-shipped]

**bitbake compiler-rt **
Steps to reproduce the behavior:
1. aarch64 machine
2. bitbake compiler-rt

**install patch in lib64 not lib**

**Desktop (please complete the following information):**
 - OS: arm64
 - Browser: chrome
 - Version: 80.0.3987.132
 - 
[image]

kraj, 2021-07-26:
it seems you are on clang 7 which is really old. which branch of meta-clang are you using? I hope you are not on one of EOLed branches. Yuo can still fix it with by adding `FILES_${PN} += "${libdir}/linux"` or something like that

blithe2015, 2021-07-27:
thud of meta-clang,Can 64-bit machines use this method（FILES_${PN} += "${libdir}/linux"）?
Can I use the latest branch and make adaptation changes?

kraj, 2021-07-27:
I think using master with thud branch might not work but yes you can try to use master for everything and that will work

blithe2015, 2021-07-27:
I'll try to solve it. I also used path to change the installation path before, but the compilation chromium error ELF, generate SDK using clang compilation or find lib instead of lib64.

kraj, 2021-12-15:
@blithe2015 are you still needing some help or can we close this issue ?