kraj/meta-clang #491: llvm tool instead of gnu binutils

hizukiayaka, 2021-08-05:
https://developer.android.com/ndk/downloads/revision_history
GNU binutils is deprecated and will be removed in an upcoming NDK release. 

I am going to add support for it.
But those tools are different to gnu binutils, we can't build one for a target, cross, native. They would take the `--target ` option.
I have no idea how to overwrite those classes in OE.

hizukiayaka, 2021-08-20:
https://android.googlesource.com/platform/ndk/+/refs/heads/ndk-release-r22/docs/BuildSystemMaintainers.md#System-STL
Warning: Using LLD with GNU strip or objcopy breaks RelRO. LLVM strip and objcopy must be used with LLD. See Issue 843 and the Binutils section of this document for more information.

kraj, 2021-08-20:
we already use some LLVM tools by default in meta-clang when clang is used a default compiler. see https://github.com/kraj/meta-clang/blob/master/classes/clang.bbclass#L8-L10 it would be fine to add strip and objcopy there as well. Would you mind creating a patch ?

hizukiayaka, 2021-08-21:
Sure. The real problem for me is the objcopy and objdump that oe core would not use the environment variable.
I have begun to develop this since the last Friday.

hizukiayaka, 2021-08-21:
Besides, do I have to use that ${HOST_PREFIX}, as the llvm would take the triple archtecture string as the argument.
And in android ndk, it doesn’t offer the one with that.

kraj, 2021-08-21:
> Besides, do I have to use that ${HOST_PREFIX}, as the llvm would take the triple archtecture string as the argument.
> And in android ndk, it doesn’t offer the one with that.

OE is a cross build system and using cross-tools is preferred, and infact is the right way also. but I can see why android might be using objcopy without prefix. It would be good if it was not hardcoded and controlled by an env bar like OBJCOPY or somesuch does android have something like this ? if not then perhaps it will be a good change to make it and also upstream it to android.

hizukiayaka, 2021-08-31:
Currently, clang package would include those binary as llvm binutils, I am not sure whether I should separate them out as a package as the other distribution does.
I think the current problem is that clang would depend on gnu binutils to build toolchain, I am not sure I need to switch it to llvm itself.

kraj, 2021-12-15:
I think best option is to define them as cross tools on the lines of other tools see https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/clang_git.bb#L209