kraj/meta-clang #497: compiler-rt: broken dependencies on `hardknott`

ensc, 2021-08-12:
**Describe the bug**

Compiling `compiler-rt-native` fails on `hardknott` due to unsatisfied dependencies.  E.g.

```
Missing or unbuildable dependency chain was: [...  'chromium-ozone-wayland', 'gn-native', 'compiler-rt-native', 'clang-native-x86_64']
```

Other missing deps are:

- `virtual/libc-native`
- `virtual/compilerlibs-native`

```
$ bitbake    compiler-rt-native -e
n# $DEPENDS [10 operations]
#   set .../sources/poky/meta/conf/bitbake.conf:274
#     ""
#   set .../sources/poky/meta/conf/documentation.conf:137
#     [doc] "Lists a recipe's build-time dependencies (i.e. other recipe files)."
#   _prepend .../sources/poky/meta/classes/base.bbclass:80
#     "${BASEDEPENDS} "
#   _prepend .../sources/poky/meta/classes/cmake.bbclass:4
#     "cmake-native "
#   _prepend .../sources/poky/meta/classes/pkgconfig.bbclass:1
#     "pkgconfig-native "
#   _append .../sources/poky/meta/classes/python3native.bbclass:5
#     " python3-native "
#   append .../sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:22
#     "ninja-native clang-cross-${TARGET_ARCH} virtual/${MLPREFIX}libc virtual/${TARGET_PREFIX}compilerlibs"
#   _append[class-nativesdk] .../sources/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:23
#     " clang-native"
#   set native.bbclass:156 [map_dependencies]
#     "pkgconfig-native cmake-native ninja-native clang-native-x86_64 virtual/libc-native virtual/compilerlibs-native python3-native"
#   append cmake.bbclass:19 [__anon_24__..._sources_poky_meta_classes_cmake_bbclass]
#     " ninja-native"
# pre-expansion value:
#   "pkgconfig-native cmake-native ${BASEDEPENDS} pkgconfig-native cmake-native ninja-native clang-native-x86_64 virtual/libc-native virtual/compilerlibs-native python3-native python3-native  ninja-native"
DEPENDS="pkgconfig-native cmake-native  pkgconfig-native cmake-native ninja-native clang-native-x86_64 virtual/libc-native virtual/compilerlibs-native python3-native python3-native  ninja-native"
```

I guess, 95b2e77c088aacaae58aa9157fc7b7bde82527c8 and pehaps related patches are missing in `hardknott`.

Revisions:

- meta-clang: 7ef47d048267cf755c496a3962b34314f6f1f1e1
- poky: 3f9e8dfde19f62bfda2308437f18fca58d7363e9

kraj, 2021-12-15:
@ensc do you still need it on `hardknott` ? right now I take care of `master` mostly and when time permits some `dunfell`

ensc, 2021-12-16:
> @ensc do you still need it on `hardknott` ? 

no; switched to `master` (which works fine with `hardknott` atm)

closed 2021-12-16 by ensc