kraj/meta-clang #514: nativesdk-clang failed to compile: linker cannot find the libstdc++

jekhor, 2021-09-24:
When trying to build and populate SDK for our custom distro based on dunfell Yocto branch, the nativesdk-clang recipe failed to build with message about missing libstdc++:
```
| ninja: build stopped: subcommand failed.
| WARNING: /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/temp/run.do_compile.2287962:1 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} }VERBOSE=1 cmake --build '/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
| ERROR: Execution of '/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/temp/run.do_compile.2287962' failed with exit code 1:
| [1/4955] cd /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/build && /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E make_directory /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/build/NATIVE
| [2/4955] /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang++  -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Ilib/Support -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Support -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot/opt/poky/1.4.1-rc.3/sysroots/x86_64-pokysdk-linux/usr/include/libxml2 -Iinclude -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/include -target x86_64-pokysdk-linux   --sysroot=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot  -O2 -pipe -fmacro-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0=/usr/src/debug/nativesdk-clang/10.0.1-r0                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0=/usr/src/debug/nativesdk-clang/10.0.1-r0                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native=  -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -DNDEBUG   -std=c++14 -MD -MT lib/Support/CMakeFiles/LLVMSupport.dir/ABIBreak.cpp.o -MF lib/Support/CMakeFiles/LLVMSupport.dir/ABIBreak.cpp.o.d -o lib/Support/CMakeFiles/LLVMSupport.dir/ABIBreak.cpp.o -c /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Support/ABIBreak.cpp
| [3/4955] /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-clang++  -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Ilib/Demangle -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Demangle -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot/opt/poky/1.4.1-rc.3/sysroots/x86_64-pokysdk-linux/usr/include/libxml2 -Iinclude -I/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/include -target x86_64-pokysdk-linux   --sysroot=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot  -O2 -pipe -fmacro-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0=/usr/src/debug/nativesdk-clang/10.0.1-r0                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0=/usr/src/debug/nativesdk-clang/10.0.1-r0                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native=  -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -DNDEBUG     -std=c++14 -MD -MT lib/Demangle/CMakeFiles/LLVMDemangle.dir/Demangle.cpp.o -MF lib/Demangle/CMakeFiles/LLVMDemangle.dir/Demangle.cpp.o.d -o lib/Demangle/CMakeFiles/LLVMDemangle.dir/Demangle.cpp.o -c /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/lib/Demangle/Demangle.cpp
| [4/4955] cd /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/build/NATIVE && /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -G Ninja -DCMAKE_MAKE_PROGRAM="ninja" -DCMAKE_TOOLCHAIN_FILE=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/toolchain-native.cmake /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE -DLLVM_TARGETS_TO_BUILD="AMDGPU;AArch64;ARM;BPF;Mips;PowerPC;RISCV;X86" -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=";" -DLLVM_DEFAULT_TARGET_TRIPLE="x86_64-unknown-linux-gnu" -DLLVM_TARGET_ARCH="host" -DLLVM_ENABLE_PROJECTS="clang;clang-tools-extra;lld;lldb" -DLLVM_EXTERNAL_PROJECTS="" -DLLVM_TEMPORARILY_ALLOW_OLD_TOOLCHAIN="ON" -DPYTHON_EXECUTABLE="/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/python3-native/python3" -DCMAKE_BUILD_TYPE=Release -DLLVM_EXTERNAL_CLANG_SOURCE_DIR=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm/../clang
| FAILED: NATIVE/CMakeCache.txt
| cd /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/build/NATIVE && /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -G Ninja -DCMAKE_MAKE_PROGRAM="ninja" -DCMAKE_TOOLCHAIN_FILE=/home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/10.0.1-r0/toolchain-native.cmake /home/jek/homam/src/xarina-project-v2/xarina-build-dunfell/tmp/work-shared/llvm-project-source-10.0.1-r0/git/llvm -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE -DLLVM_TARGETS_TO_BUILD="AMDGPU;AArch64;ARM;BPF;Mips;PowerPC;RISCV;X86" -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD=";" -DLLVM_DEFAULT_TARGET_TRIPLE="x86_64-unkn [... cut ...]
This issue is reproduced in Debian 'unstable' distro at build host but cannot be reproduced in Ubuntu 18.04. Does anyone can point me way to resolve this? I still didn't try to check it with vanilla poky + meta-clang configuration but plan to do this.

**To Reproduce**
Steps to reproduce the behavior:
1. Create the Yocto buildsystem configuration based on the vanilla 'dunfell' poky and meta-clang layer
2. Run `bitbake core-image-minimal -c populate_sdk`

**Expected behavior**
SDK is built successfully.

kraj, 2021-09-24:
Not every distro is supported a build host, certainly, rolling distros are a problem since they change over time but dunfell does not. Here is list of supported distros for dunfell. Please use one of those

https://docs.yoctoproject.org/dunfell/ref-manual/ref-system-requirements.html#supported-linux-distributions

fwuehr, 2021-09-24:
@jekhor  I faced similar issues with other projects usinf cmake mostly with cross compilation, you can try to disable the compiler check with a patch in one of your layers, like it's described here: https://stackoverflow.com/questions/53633705/cmake-the-c-compiler-is-not-able-to-compile-a-simple-test-program/53635241

jekhor, 2021-09-27:
Hmm... Fixed by `apt dist-upgrade`, close the issue.

closed 2021-09-27 by jekhor