kraj/meta-clang #527: clang-native compilation error Assembler mesages

roystonvasey, 2021-11-08:
clang-native do_compile fails with error messages:

{standard input}: Assembler messages:
{standard input}:46751: Warning: end of file not at end of a line; newline inserted
{standard input}:47221: Error: bad register name `%rd'
{standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
The host build platform is x86_64-linux.
Using meta-clang master pulled yesterday.

kraj, 2021-11-08:
@roystonvasey your build machine is running out of memory, perhaps you need a better spec'ed machine with atleast 16GB of RAM

roystonvasey, 2021-11-09:
Many thanks your suggestion worked.

closed 2021-11-09 by roystonvasey