kraj/meta-clang #538: Segmentation fault (core dumped)

abogani, 2021-12-07:
**Describe the bug**
root@localhost:~# clang
Segmentation fault (core dumped)

**To Reproduce**
Using meta-clang (dunfell or dunfell-clang12 branches)

**Expected behavior**
clang should complains about missing input files.

**Additional context**
strace clang
[...]
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\220\327\240\0010\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=182888, ...}) = 0
mmap(NULL, 184032, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7ffff0902000
mmap(0x7ffff090f000, 53248, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7ffff090f000
mmap(0x7ffff091c000, 57344, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7ffff091c000
mmap(0x7ffff092a000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x27000) = 0x7ffff092a000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ffff0900000
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7ffff08fd000
arch_prctl(ARCH_SET_FS, 0x7ffff08fd780) = 0
mprotect(0x7ffff0e1e000, 12288, PROT_READ) = 0
mprotect(0x7ffff092a000, 16384, PROT_READ) = 0
mprotect(0x7ffff0954000, 4096, PROT_READ) = 0
mprotect(0x7ffff0c00000, 4096, PROT_READ) = 0
mprotect(0x7ffff0be5000, 4096, PROT_READ) = 0
mprotect(0x7ffff0be0000, 4096, PROT_READ) = 0
mprotect(0x7ffff0aa0000, 32768, PROT_READ) = 0
mprotect(0x7ffff0c35000, 8192, PROT_READ) = 0
mprotect(0x7ffff0c46000, 4096, PROT_READ) = 0
mprotect(0x7ffff0c65000, 4096, PROT_READ) = 0
mprotect(0x7ffff0e40000, 4096, PROT_READ) = 0
mprotect(0x7ffff100e000, 45056, PROT_READ) = 0
mprotect(0x7ffff4f2f000, 4001792, PROT_READ) = 0
mprotect(0x7ffff7dee000, 1740800, PROT_READ) = 0
mprotect(0x555555576000, 4096, PROT_READ) = 0
mprotect(0x7ffff7ffc000, 4096, PROT_READ) = 0
set_tid_address(0x7ffff08fda50)         = 5103
set_robust_list(0x7ffff08fda60, 24)     = 0
rt_sigaction(SIGRTMIN, {sa_handler=0x7ffff0c51670, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0x7ffff0c5dbb0}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0x7ffff0c51710, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0x7ffff0c5dbb0}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
brk(NULL)                               = 0x555555578000
brk(0x555555599000)                     = 0x555555599000
futex(0x7ffff5322b18, FUTEX_WAKE_PRIVATE, 2147483647) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x1ffffebf8} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

ldd /usr/bin/clang
	linux-vdso.so.1 (0x00007ffff7fd1000)
	libclang-cpp.so.10 => /usr/bin/../lib/libclang-cpp.so.10 (0x00007ffff534a000)
	libLLVM-10.so => /usr/bin/../lib/libLLVM-10.so (0x00007ffff0ffb000)
	libstdc++.so.6 => /usr/bin/../lib/libstdc++.so.6 (0x00007ffff0e1e000)
	libgcc_s.so.1 => /usr/bin/../lib/libgcc_s.so.1 (0x00007ffff0e04000)
	libc.so.6 => /usr/bin/../lib/libc.so.6 (0x00007ffff0c47000)
	libpthread.so.0 => /usr/bin/../lib/../lib/libpthread.so.0 (0x00007ffff0c26000)
	/usr/lib/ld-linux-x86-64.so.2 (0x00007ffff7fd2000)
	libffi.so.7 => /usr/bin/../lib/../lib/libffi.so.7 (0x00007ffff0c18000)
	libedit.so.0 => /usr/bin/../lib/../lib/libedit.so.0 (0x00007ffff0bde000)
	libz.so.1 => /usr/bin/../lib/../lib/libz.so.1 (0x00007ffff0bc3000)
	libdl.so.2 => /usr/bin/../lib/../lib/libdl.so.2 (0x00007ffff0bbe000)
	libm.so.6 => /usr/bin/../lib/../lib/libm.so.6 (0x00007ffff0a87000)
	libxml2.so.2 => /usr/bin/../lib/../lib/libxml2.so.2 (0x00007ffff0934000)
	libncurses.so.5 => /usr/bin/../lib/../lib/libncurses.so.5 (0x00007ffff090b000)
	libtinfo.so.5 => /usr/bin/../lib/../lib/libtinfo.so.5 (0x00007ffff08de000)

kraj, 2021-12-15:
@abogani  is it reproducible on `master` with clang-13 as well ?

abogani, 2021-12-17:
I used two boards with the same Yocto/OE project.

=== On a Cortex-A15 board ===
*) Using origin/dunfell the simple invocation, without any parameters, of clang(-10) works with "error: no input files".
*) Using origin/clang-12 the simple invocation, without any parameters, of clang(-12) works with "error: no input files".
*) Using origin/master the simple invocation, without any parameters, of clang(-13) works with "error: no input files".
=== On a Whiskey Lake board ===
*) Using origin/dunfell the simple invocation, without any parameters, of clang(-10) doesn't work with "Segmentation fault (core dumped)"
*) Using origin/clang-12 doesn't build:
[...]
ERROR: llvm-project-source-12.0.0-12.0.0-r0 do_fetch: Fetcher failure: Unable to find revision 67d3e271a28287b2c92ecef2f5e98c49134e5946 in branch llvm_release_100 even from upstream
ERROR: llvm-project-source-12.0.0-12.0.0-r0 do_fetch: Bitbake Fetcher Error: FetchError('Unable to fetch URL from any source.', 'git://github.com/KhronosGroup/SPIRV-LLVM-Translator.git;protocol=https;branch=llvm_release_100;destsuffix=git/llvm/projects/llvm-spirv;name=spirv')
ERROR: Logfile of failure stored in: /home/alessio.bogani/build-images/voltumna/build/tmp/work-shared/llvm-project-source-12.0.0-r0/temp/log.do_fetch.12876
ERROR: Task (/home/alessio.bogani/build-images/voltumna/meta-voltumna/openembedded-core/../meta-clang/recipes-devtools/clang/llvm-project-source.bb:do_fetch) failed with exit code '1'
*) Using origin/master doesn't build:
[...]
In file included from /home/alessio.bogani/build-images/voltumna/build/tmp/work-shared/llvm-project-source-13.0.1-r0/git/llvm/projects/llvm-spirv/lib/SPIRV/SPIRVRegularizeLLVM.cpp:48:0:
/home/alessio.bogani/build-images/voltumna/build/tmp/work-shared/llvm-project-source-13.0.1-r0/git/llvm/include/llvm/PassSupport.h:21:2: error: #error "Do not include <PassSupport.h>; include <Pass.h> instead"
 #error "Do not include <PassSupport.h>; include <Pass.h> instead"
  ^~~~~

kraj, 2021-12-17:
I think the issue is spirv fork of llvm perhaps you need to use `branch=llvm_release_120` when using clang12

abogani, 2021-12-18:
That seems related to meta-intel layer so I guess that the right place to address this bug is that layer.

Sorry for annoying @kraj !

closed 2021-12-23 by abogani

michaek, 2022-12-01:
@abogani Are you able to share how (or whether) you addressed this with the meta-intel layer?

abogani, 2022-12-01:
Sorry, I can't. I disabled clang generation and never enable it again since then.