kraj/meta-clang #557: clang-nativesdk Failing for SDKMACHINE aarch64

GabeDia, 2022-01-17:
**Describe the bug**
Normally I build Yocto with and deploy SDK's for x86_64 hosts. I'm trying to use my x86_64 machine to build an SDK deployed on ARM64 hosts (SDKMACHINE=aarch64) with cross-compilers for armv7 target.

nativesdk-clang fails with the following error:

|     /home/gabe/Code/yocto_imx6/build/tmp/hosttools/ld.gold: fatal error: /home/gabe/Code/yocto_imx6/build/tmp/work/aarch64-nativesdk-savantsdk-linux/nativesdk-clang/10.0.1-r0/recipe-sysroot/usr/lib//aarch64-savantsdk-linux/9.3.0/../../Scrt1.o: unsupported ELF machine number 183
|     clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.

The build/tmp/hosttools/ld.gold is x86_64 whereas the dynamic linker (and crt1.o per error above) are aarch64. So there's some sort of misconfiguration I just don't know how to resolve it.

Full run and log of configure task:

[run.do_configure.txt](https://github.com/kraj/meta-clang/files/7884146/run.do_configure.txt)
[log.do_configue.txt](https://github.com/kraj/meta-clang/files/7884147/log.do_configue.txt)

**To Reproduce**
Dunfell release, x86_64 ubuntu 20, armv7 tuned MACHINE
Bitbake nativesdk-clang

Thanks,
Gabe

kraj, 2022-01-28:
Can you try adding
```
LDFLAGS:append:class-nativesdk = "-fuse-ld=bfd"
```
to `recipes-devtools/clang/clang_git.bb`

and see if it helps ?

GabeDia, 2022-01-31:
Hi @kraj, thanks for taking time to help. 

This change seems to have worked - I'm guessing there's no arm64 gold ld available which is why bfd works. Is there any consequence from changing the linker to bfd, it should be transparent to developers right?

kraj, 2022-01-31:
@GabeDia it should work fine

GabeDia, 2022-01-31:
Ok, thanks again for the help - was able to install SDK and compile on arm64 host, will close.

closed 2022-01-31 by GabeDia