kraj/meta-clang #560: OpenMP - version node not found for symbol omp_get_num_places_@@VERSION

jwinarske, 2022-01-27:
**Describe the bug**
OpenMP fails to link for 10.0.1

```
/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-ld: runtime/src/libomp.so: version node not found for symbol omp_get_num_places_@@VERSION
/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-ld: failed to set dynamic section sizes: bad value
```

**To Reproduce**
Steps to reproduce the behavior:
1. branch dunfell-clang12 (old specifiers are required)
2. bitbake openmp

**Expected behavior**
Expect openmp to build without error

```
NOTE: VERBOSE=1 cmake --build /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build --target all --
[1/1] : && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-clang -fPIC -target aarch64-oe-linux -march=armv8-a -mcpu=cortex-a57 -mtune=cortex-a57   -mlittle-endian -Qunused-arguments  --sysroot=/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot  -O2 -fomit-frame-pointer    -Wa,--noexecstack -fexpensive-optimizations   -frename-registers -ftree-vectorize   -finline-functions -finline-limit=64   -Wno-error=maybe-uninitialized -Wno-error=unused-result     -target aarch64-oe-linux -march=armv8-a -mcpu=cortex-a57 -mtune=cortex-a57   -mlittle-endian -Qunused-arguments  --sysroot=/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot -DNDEBUG  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -shared -Wl,-soname,libomp.so -o runtime/src/libomp.so runtime/src/CMakeFiles/omp.dir/kmp_alloc.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_atomic.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_csupport.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_debug.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_itt.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_environment.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_error.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_global.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_i18n.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_io.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_runtime.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_settings.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_str.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_tasking.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_threadprivate.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_utility.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_barrier.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_wait_release.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_affinity.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_dispatch.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_lock.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_sched.cpp.o runtime/src/CMakeFiles/omp.dir/z_Linux_util.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_gsupport.cpp.o runtime/src/CMakeFiles/omp.dir/thirdparty/ittnotify/ittnotify_static.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_taskdeps.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_cancel.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_ftn_cdecl.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_ftn_extra.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_version.cpp.o runtime/src/CMakeFiles/omp.dir/ompt-general.cpp.o runtime/src/CMakeFiles/omp.dir/z_Linux_asm.S.o  -lpthread  -ldl && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E create_symlink libomp.so libgomp.so && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E create_symlink libomp.so libiomp5.so && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E make_directory /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/common.ompt.optional/include && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E copy omp.h /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/common.ompt.optional/include && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E copy omp-tools.h /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/common.ompt.optional/include && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E make_directory /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/lin_aarch64.ompt.optional/lib && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E copy /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src/libomp.so /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/lin_aarch64.ompt.optional/lib
FAILED: runtime/src/libomp.so 
: && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-clang -fPIC -target aarch64-oe-linux -march=armv8-a -mcpu=cortex-a57 -mtune=cortex-a57   -mlittle-endian -Qunused-arguments  --sysroot=/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot  -O2 -fomit-frame-pointer    -Wa,--noexecstack -fexpensive-optimizations   -frename-registers -ftree-vectorize   -finline-functions -finline-limit=64   -Wno-error=maybe-uninitialized -Wno-error=unused-result     -target aarch64-oe-linux -march=armv8-a -mcpu=cortex-a57 -mtune=cortex-a57   -mlittle-endian -Qunused-arguments  --sysroot=/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot -DNDEBUG  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -shared -Wl,-soname,libomp.so -o runtime/src/libomp.so runtime/src/CMakeFiles/omp.dir/kmp_alloc.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_atomic.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_csupport.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_debug.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_itt.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_environment.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_error.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_global.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_i18n.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_io.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_runtime.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_settings.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_str.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_tasking.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_threadprivate.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_utility.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_barrier.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_wait_release.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_affinity.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_dispatch.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_lock.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_sched.cpp.o runtime/src/CMakeFiles/omp.dir/z_Linux_util.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_gsupport.cpp.o runtime/src/CMakeFiles/omp.dir/thirdparty/ittnotify/ittnotify_static.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_taskdeps.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_cancel.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_ftn_cdecl.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_ftn_extra.cpp.o runtime/src/CMakeFiles/omp.dir/kmp_version.cpp.o runtime/src/CMakeFiles/omp.dir/ompt-general.cpp.o runtime/src/CMakeFiles/omp.dir/z_Linux_asm.S.o  -lpthread  -ldl && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E create_symlink libomp.so libgomp.so && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E create_symlink libomp.so libiomp5.so && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E make_directory /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/common.ompt.optional/include && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/cmake -E copy omp.h /build/tmp-glibc/work-shared/llvm-project-source-10.0.1-r0/git/openmp/runtime/exports/common.ompt.optional/include && cd /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/runtime/src && /build/tmp-glibc/work/aarch64-oe-linux/openmp/10. [... cut ...]
/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/recipe-sysroot-native/usr/bin/aarch64-oe-linux/aarch64-oe-linux-ld: failed to set dynamic section sizes: bad value
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
WARNING: /build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/temp/run.do_compile.6412:1 exit 1 from 'eval ${DESTDIR:+DESTDIR=${DESTDIR} }VERBOSE=1 cmake --build '/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build' "$@" -- ${EXTRA_OECMAKE_BUILD}'
```

jwinarske, 2022-01-27:
This had the clue:  https://reviews.llvm.org/D75001

This is a Qualcomm target, opposed to Nvidia.

```
[runner@fedora build]$ grep -r LIBOMP_HAVE_VERSION_SCRIPT_FLAG
CMakeFiles/CMakeError.log:Performing C Linker Flag test LIBOMP_HAVE_VERSION_SCRIPT_FLAG failed with the following output:
CMakeFiles/CMakeError.log:Change Dir: /__w/sax155p_yocto/sa81x5agl/src/apps/apps_proc/poky/build/tmp-glibc/work/aarch64-oe-linux/openmp/10.0.1-r0/build/CMakeFiles/CMakeTmp/link_flag_check_LIBOMP_HAVE_VERSION_SCRIPT_FLAG/build
CMakeCache.txt://Test LIBOMP_HAVE_VERSION_SCRIPT_FLAG
CMakeCache.txt:LIBOMP_HAVE_VERSION_SCRIPT_FLAG:INTERNAL=
[runner@fedora build]$ 
```
Which would indicate why it's failing.  About to try the suggested workaround 
```
-DCMAKE_CXX_FLAGS=-Wno-unknown-cuda-version -DCMAKE_C_FLAGS=-Wno-unknown-cuda-version
```

jwinarske, 2022-01-28:
I can get it to build if I do this in config-ix.cmake:

```
  libomp_check_linker_flag("-Wl,--version-script=${LIBOMP_SRC_DIR}/exports_so.txt" LIBOMP_HAVE_VERSION_SCRIPT_FLAG)
  set(LIBOMP_HAVE_VERSION_SCRIPT_FLAG true)
```

For whatever reason the linker flag is failing, and sets LIBOMP_HAVE_VERSION_SCRIPT_FLAG false.  Which causes it to fail at link time.

kraj, 2022-01-28:
@jwinarske  you should firstly set `LLVMVERSION = "12.0.0"` in `conf/layer.conf` this will atleast choose clang12
and perhaps that will fix it. If that does not fix it then try
```
TUNE_CCARGS_remove = "-Qunused-arguments"
```
in openmp recipe and see if that helps.

jwinarske, 2022-01-28:
Adding `LLVMVERSION = "12.0.0"` to `conf/layer.conf` did not resolve it.  I am trying bbappend with your second suggestion next.

jwinarske, 2022-01-28:
@kraj Adding `TUNE_CCARGS_remove = "-Qunused-arguments"` to `openmp_git.bb` did not resolve it.

kraj, 2022-02-09:
> LLVM bug: https://bugs.llvm.org/show_bug.cgi?id=39200
> 
> Referenced patch resolves build issue

OK, I can accept this patch, I wonder why its not yet applied upstream/master

jwinarske, 2022-02-09:
https://github.com/kraj/meta-clang/pull/563

Running final tests.  Will move from WIP when builds are confirmed good.