kraj/meta-clang #561: Yocto 3.1.13 missing recipes - dunfell-clang12

jwinarske, 2022-02-03:
**Describe the bug**
Branch: dunfell-clang12
Yocto: 3.1.13

ERROR: No recipes available for:
[363](https://github.com/meta-flutter/meta-flutter/runs/5055075725?check_suite_focus=true#step:7:363)
  /__w/meta-flutter/yocto-dunfell/meta-clang/recipes-multimedia/mediasdk/intel-mediasdk_%.bbappend
[364](https://github.com/meta-flutter/meta-flutter/runs/5055075725?check_suite_focus=true#step:7:364)
  /__w/meta-flutter/yocto-dunfell/meta-clang/recipes-security/optee/optee-os-stm32mp_%.bbappend
[365](https://github.com/meta-flutter/meta-flutter/runs/5055075725?check_suite_focus=true#step:7:365)

Workaround:
          rm -rf meta-clang/recipes-multimedia/mediasdk/intel-mediasdk_%.bbappend
          rm -rf meta-clang/recipes-security/optee/optee-os-stm32mp_%.bbappend

kraj, 2022-02-03:
https://github.com/kraj/meta-clang/commit/92abd44335b84fc895ddc166694fe8b2a2783ae6 should fix it, please try latest branch

jwinarske, 2022-02-03:
@kraj Builds are in flight.  Should know more after lunch.  Thanks!

jwinarske, 2022-02-03:
@kraj Two builds down.  So far so good

jwinarske, 2022-02-03:
@kraj all prior failures have been accounted for.  Fix is good.  Thanks again.

closed 2022-02-03 by jwinarske